machinekit/machinekit #412: GCode program fails with "Unexpected realtime delay on RT thread 1"

ftkalcevic, 2015-01-02:
As reported on the machinekit group (https://groups.google.com/d/msg/machinekit/NeZXwHa2FqU/BYlLLXJiRuEJ) linuxcnc fails repeatedly on the same line with...

> 14: Unexpected realtime delay on RT thread 1 
> emc/task/taskintf.cc 617: Error on axis 3, command number 248

when running the gcode...

```
G21        ;metric values
G90        ;absolute positioning
G64 P0.1    ; path mode

G92  A881.308240
;TYPE:FILL
G1 X-44.982000 Y16.959000 Z6.5 A881.308240 F4800
G1 X-45.503000 Y16.439000 A881.318030 
G0 X-45.012 Y16.364
G1 X-44.416000 Y16.959000 A881.329230 
G1 X-45.012000 Y16.364000 A881.340440 
G0 X-45.011 Y15.799
G1 X-43.851000 Y16.959000 A881.362260
G1 X-45.011000 Y15.799000 A881.384090 
G0 X-44.806 Y15.438
G1 X-43.699000 Y16.545000 A881.404920 
G1 X-44.806000 Y15.438000 A881.425740 
M2
```

This doesn't happen without the `G64 P0.1` command.

My machine is a lineardelta printer, using the lineardeltakins, but if I run with genkins, I get the same problem.  The hardware is a BeagleBoneBlack with CRAMPS interface.

I tried to turn debug up and got a few more extra lines...

```
Motion id 10 took 0.028377 seconds (current 11)
joint 0 -18.245 197.131 197.149
joint 1 15.171 204.743 204.728
joint 2 -11.522 157.982 157.993
joint 3 0.490 0.029 0.028
joint 0 -4.262 197.025 197.030
joint 1 3.522 204.831 204.827
joint 2 -2.698 157.915 157.918
joint 3 0.114 0.032 0.031
joint 0 -19.602 196.907 196.927
joint 1 14.513 204.919 204.904
joint 2 19.095 158.022 158.003
joint 3 0.000 0.032 0.032
Motion id 11 took 0.033792 seconds (current 12)
14: Unexpected realtime delay on RT thread 1
joint 0 0.000 198.630 198.630
joint 1 0.000 203.835 203.835
joint 2 0.000 159.385 159.385
joint 3 0.000 0.384 0.384
emc/task/taskintf.cc 626: Error on axis 3, command number 399
Motion id 12 took 0.151054 seconds (current 0)
NML_INTERP_LIST::append(nml_msg_ptr{size=12,type=EMC_TASK_PLAN_SYNCH}) : list_size=1, line_number=17
emcTaskPlanClose() called at emc/task/emctask.cc:206
```

The lines that start with "joint n..." are print statements in  taskintf.c:emcAxisUpdate showing: joint #, vel_cmd, pos_cmd, pos_fb.  That doesn't show much.  The only thing that looks suspicious is the line "Motion id 12 took 0.151054 seconds", (5 times longer than normal) but there is no error.

mhaberler, 2015-01-02:
I'll look into the motion debug issue

does this also happen when you use trivkins?

maybe this yields a clue (unfortunately the kins arent listed) - once you get a RT violation, run the following - this should yield the top CPU using thread function first:

mah@nwheezy:~/machinekit-prepare$ halcmd show param|grep '\.tmax$'|sort -n -r -k4
   269  s32   RW       44176327  motion-controller.tmax
   488  s32   RW       44150305  ddt_x.tmax
   269  s32   RW       42355912  motion-command-handler.tmax
   519  s32   RW       41415427  vel_xy.tmax
   488  s32   RW       40648788  ddt_xv.tmax
   576  s32   RW       36371046  near_speed.tmax
   488  s32   RW       29448400  ddt_zv.tmax
   550  s32   RW       17419105  sim_spindle.tmax
   488  s32   RW       11856476  ddt_z.tmax
   519  s32   RW        5054961  vel_xyz.tmax
   568  s32   RW        3798605  spindle_mass.tmax
   611  s32   RW        3242304  comp_y.tmax
   611  s32   RW        2721224  comp_z.tmax
   488  s32   RW        2695329  ddt_yv.tmax
   488  s32   RW        2537979  ddt_y.tmax
   558  s32   RW        2261615  limit_speed.tmax
   611  s32   RW        1693825  comp_x.tmax
   604  s32   RW        1541722  or2_0.tmax

also, post the output of 

mah@nwheezy:~/machinekit-prepare$ halcmd show thread
Realtime Threads (flavor: xenomai) :
     Period  FP     Name               (     Time, Max-Time )
    1000000  YES          servo-thread (    61221, 47371683 )
                  1 motion-command-handler
                  2 motion-controller
                  3 ddt_x
                  4 ddt_xv
                  5 ddt_y
                  6 ddt_yv
                  7 ddt_z
                  8 ddt_zv
                  9 vel_xy
                 10 vel_xyz
                 11 limit_speed
                 12 spindle_mass
                 13 near_speed
                 14 sim_spindle
                 15 comp_x
                 16 comp_y
                 17 comp_z
                 18 or2_0

(this examples are from a Vbox environment, so fairly random numbers)

mhaberler, 2015-01-02:
motion debug: just run linuxcnc like so:

$ DEBUG=5 linuxcnc 

this will yield command numbers in the log (CMD xxx), which you can correlate to the error:

mah@nwheezy:~/machinekit-prepare$ tail -f /var/log/linuxcnc.log 
Jan  2 02:08:50 nwheezy msgd:0: hal_lib:6261:rt FREE
Jan  2 02:08:50 nwheezy msgd:0: hal_lib:6261:rt 
Jan  2 02:08:50 nwheezy msgd:0: hal_lib:6270:user EMC_LUBE_ON
Jan  2 02:08:50 nwheezy msgd:0: hal_lib:6261:rt 3143: CMD 126, code  14 
Jan  2 02:08:50 nwheezy msgd:0: hal_lib:6261:rt FREE
Jan  2 02:08:50 nwheezy msgd:0: hal_lib:6261:rt 
Jan  2 02:08:52 nwheezy msgd:0: hal_lib:6261:rt 4616: CMD 127, code  24 
Jan  2 02:08:52 nwheezy msgd:0: hal_lib:6261:rt HOME
Jan  2 02:08:52 nwheezy msgd:0: hal_lib:6261:rt  -1
Jan  2 02:08:52 nwheezy msgd:0: hal_lib:6261:rt

ftkalcevic, 2015-01-02:
Tested trivkins and that gives the same issue.  Outputs below are using trivkins.

```
machinekit@beaglebone:~/linuxcnc$ halcmd show param|grep '.tmax$'|sort -n -r -k4
   285  s32   RW         146594  motion-controller.tmax
   285  s32   RW          23540  motion-command-handler.tmax
   582  s32   RW            760  hal_pru_generic.update.tmax
   582  s32   RW            421  hal_pru_generic.capture-position.tmax
   543  s32   RW            344  bb_gpio.write.tmax
   719  s32   RW            321  pid.0.do-pid-calcs.tmax
   543  s32   RW            290  bb_gpio.read.tmax
   719  s32   RW            225  pid.1.do-pid-calcs.tmax
   831  s32   RW            195  mux4.0.tmax
   793  s32   RW            184  at.0.tmax
   815  s32   RW            176  not.0.tmax
   778  s32   RW            173  limit1.0.tmax
   981  s32   RW            165  time.0.tmax
   793  s32   RW            162  at.1.tmax
   778  s32   RW            159  limit1.1.tmax
   989  s32   RW            148  and2.3.tmax
   806  s32   RW            148  and4.0.tmax
   815  s32   RW            147  not.1.tmax
   989  s32   RW            137  and2.2.tmax
   989  s32   RW            132  and2.0.tmax
   989  s32   RW             40  and2.1.tmax
   815  s32   RW             36  not.2.tmax
```

```
machinekit@beaglebone:~/linuxcnc$ halcmd show thread
Realtime Threads (flavor: xenomai) :
     Period  FP     Name               (     Time, Max-Time )
    1000000  YES          servo-thread (     2878,   148102 )
                  1 hal_pru_generic.capture-position
                  2 bb_gpio.read
                  3 motion-command-handler
                  4 motion-controller
                  5 pid.0.do-pid-calcs
                  6 pid.1.do-pid-calcs
                  7 limit1.0
                  8 limit1.1
                  9 hal_pru_generic.update
                 10 bb_gpio.write
                 11 at.0
                 12 at.1
                 13 not.0
                 14 and4.0
                 15 mux4.0
                 16 time.0
                 17 not.1
                 18 not.2
                 19 and2.0
                 20 and2.1
                 21 and2.2
                 22 and2.3
```

And this is the output from /var/log

```
Jan  2 17:09:25 beaglebone msgd:0: hal_lib:23124:rt 46989: CMD 179, code  29
Jan  2 17:09:25 beaglebone msgd:0: hal_lib:23124:rt SET_LINE
Jan  2 17:09:25 beaglebone msgd:0: hal_lib:23124:rt
Jan  2 17:09:25 beaglebone msgd:0: hal_lib:23124:rt 46997: CMD 180, code  29
Jan  2 17:09:25 beaglebone msgd:0: hal_lib:23124:rt SET_LINE
Jan  2 17:09:25 beaglebone msgd:0: hal_lib:23124:rt
Jan  2 17:09:25 beaglebone msgd:0: hal_lib:23124:rt 47014: CMD 181, code  29
Jan  2 17:09:25 beaglebone msgd:0: hal_lib:23124:rt SET_LINE
Jan  2 17:09:25 beaglebone msgd:0: hal_lib:23124:rt
Jan  2 17:09:25 beaglebone msgd:0: hal_lib:23124:rt 47023: CMD 182, code  29
[... 39 lines cut ...]
Outgoing motion id is 13.
Issuing EMC_TRAJ_LINEAR_MOVE --          (+220,+120,    +0,-43.851000,16.959000,6.500000,0.054020,0.000000,0.000000,0.000000,0.000000,0.000000,    +2,80.000000,636.396103,5656.854249,    +0,    -1,)
Outgoing motion id is 14.
Issuing EMC_TRAJ_LINEAR_MOVE --          (+220,+120,    +0,-45.011000,15.799000,6.500000,0.075850,0.000000,0.000000,0.000000,0.000000,0.000000,    +2,80.000000,636.396103,5656.854249,    +0,    -1,)
Outgoing motion id is 15.
Issuing EMC_TRAJ_LINEAR_MOVE --          (+220,+120,    +0,-44.806000,15.438000,6.500000,0.075850,0.000000,0.000000,0.000000,0.000000,0.000000,    +1,517.494712,517.494712,4599.952992,    +0,    -1,)
Outgoing motion id is 16.
Issuing EMC_TRAJ_LINEAR_MOVE --          (+220,+120,    +0,-43.699000,16.545000,6.500000,0.096680,0.000000,0.000000,0.000000,0.000000,0.000000,    +2,80.000000,636.396103,5656.854249,    +0,    -1,)
Outgoing motion id is 17.
Issuing EMC_TRAJ_LINEAR_MOVE --          (+220,+120,    +0,-44.806000,15.438000,6.500000,0.117500,0.000000,0.000000,0.000000,0.000000,0.000000,    +2,80.000000,636.396103,5656.854249,    +0,    -1,)
Motion id 7 took 9.288285 seconds (current 8)
Motion id 8 took 0.019658 seconds (current 9)
Motion id 9 took 0.019390 seconds (current 10)
Motion id 10 took 0.031414 seconds (current 11)
Motion id 11 took 0.018824 seconds (current 12)
14: Unexpected realtime delay on RT thread 1 
emc/task/taskintf.cc 626: Error on axis 3, command number 185
Motion id 12 took 0.171931 seconds (current 0)
NML_INTERP_LIST::append(nml_msg_ptr{size=12,type=EMC_TASK_PLAN_SYNCH}) : list_size=1, line_number=17
emcTaskPlanClose() called at emc/task/emctask.cc:206
NML_INTERP_LIST::append(nml_msg_ptr{size=12,type=EMC_TASK_PLAN_SYNCH}) : list_size=1, line_number=17
NML_INTERP_LIST::append(nml_msg_ptr{size=12,type=EMC_TASK_PLAN_SYNCH}) : list_size=1, line_number=17
joint 3 following error

```

mhaberler, 2015-01-02:
yes, that is useful - it's the motion lower half (RT trajectory planner) causing the RT violation (146uS maximum execution time which is outside the 10% window of 1mS motion cycle); also the kins module is not at fault because trivkins has negligible processing time requirements

the 'CMD xxx' numbers dont correlate with the line number, that's normal

since blending and just line segments are involved, I would guess it's the blending code exceeding the time window

@robEllenberg: any chance we can narrow this down if/which blending operation causes the RT budget to be exceeded?

robEllenberg, 2015-01-02:
I'll take a look on my Beaglebone setup in the next day or two. I don't think it's the blend arc code that's directly causing the delay, though. Currently the circular arc blending code only works with moves in the XYZ axes. Adding a move along the A axis causes it to fall back to parabolic blending. My initial suspicion is there there is a calculation that is choking on a numerical edge case somewhere (trig functions, sqrt, etc.). I'll know more once I can reproduce it.

ftkalcevic, 2015-01-02:
This is as small as I can get it.  This version fails on the CRAMPS example 3d printer configuration...

```
G21        ;metric values
G90        ;absolute positioning
G64 P0.25   ; path mode
G92  A881.340440

F4800
G1 X44.988 Y16.364000 A881.340440    
G0 X44.989 Y15.799
G1 X46.149 Y16.959000 A881.362260    
G1 X44.989 Y15.799000 A881.384090    
M2
```

mhaberler, 2015-01-03:
@kinsamanka: Hi Gemi - any chance you could run Frank's gcode sequence on a Raspberry?
I'm thinking - the Rpi has about half the CPU power; if it's a systematic error it should show on the Rpi too

kinsamanka, 2015-01-13:
There's no error when using ```axis_9axis.ini``` on RPi with Xenomai kernel (I don't have access to my PICnc hw atm).

@ftkalcevic: Does the error also happens when using the ```sim``` configs? The issue could be with the PRU code.

strahlex, 2015-01-15:
I will try it with my 3D printer setup on Monday.

strahlex, 2015-01-19:
I could reproduce your problem with another GCode file using the ´G64´ command.

Here is my log output
   already splitting on id 69479 with cycle time 0.001000
   14: Unexpected realtime delay on RT thread 1 
   emc/task/taskintf.cc 617: Error on axis 0, command number 69275
   joint 0 following error

robEllenberg, 2015-01-19:
I've been doing a bunch of small TP fixes that may impact 4th axis behavior
(rebased from linuxcnc 2.7):

https://github.com/robEllenberg/linuxcnc-mirror/tree/feature/mk-spiral-arc-handling-rebase

These changes may incidentally fix this issue, though I haven't been able
to test on a beaglebone / RT box yet.

Thanks,
Rob

robEllenberg, 2015-01-22:
Ok, good news, I think I may have figured out and fixed the cycle-splitting
error in this branch:

https://github.com/robEllenberg/linuxcnc-mirror/tree/feature/mk-spiral-arc-handling-rebase

The problem was that parabolic blending would occasionally blend over too
much of a segment. I added a check for this case when computing the blend
velocities, and now it seems to behave much better (at least in my sim
config).

-Rob

strahlex, 2015-01-22:
@robEllenberg Take a look at this print: https://wolke.effet.info/public.php?service=files&t=95ba87163fede0209b092fe9a839bcbf
the printing head seems to stop at every beginning and end of an arc. Could this be related to the problem you fixed?

robEllenberg, 2015-01-26:
Can you post a section of the G code for that part? I'd like to take a look
at a simulation run and see what's going on.

strahlex, 2015-01-27:
Should be exactly one layer: http://pastebin.com/skNm9ymW

robEllenberg, 2015-01-27:
What blend settings are you using on this program? I see a couple of tight
corners, like at line 107-110. If you have a large blend tolerance, then
the actual path taken will be much shorter in total length then the nominal
path (it's literally cutting the corner). This would mean that the corner
will have more plastic, so you would expect to see a bulge there. I'd try
running the same program with much tighter blend tolerances and see if that
bulge is still there.

strahlex, 2015-01-27:
The default settings. When I use e.g. `G64 P0.25` to set the tolerance I get the RT delay problem.

strahlex, 2015-01-28:
@robEllenberg If I understand the trajectory correctly the A axis is synchronized to the time of the movement (mentioned G2/3 documentation) meaning that the A axis is moved at a constant speed. The problem we see here is that for some reason the movement slowed down at these points which resulted in overextrusion. But why did it slow down if that should be a perfect circle?

strahlex, 2015-01-29:
I am experimenting with a velocity extrusion config. Seems like the A axis triggers the RT errors. With 3 axes only I get no such problems.

robEllenberg, 2015-01-29:
Hmm, so it's likely something to do with either parabolic blending, or
switching between arc blending and parabolic. I've been doing some work on
those issues in my branch off of linuxcnc 2.7, and I have a few fixes to
the TP I'm in the process of rebasing onto machinekit. Would you mind
testing them out once I'm done?

strahlex, 2015-02-02:
The TP patch fixes the RT delay problems (at least I never got the error after applying the patch).

robEllenberg, 2015-02-02:
Awesome! Let me know if you see any other hiccups.

strahlex, 2015-02-05:
@ftkalcevic Can you still reproduce the problem or can we close this issue?

ftkalcevic, 2015-02-06:
I'm still a git noob.  How do I get the update?  I have my own fork of machinekit with my own tweaks.  Do I need to merge the branch listed above?  Is that a merge?  Or pull?  Or fetch?  Or all?

Thanks.

mhaberler, 2015-02-06:
I guess the best way is to do this:

- commit all your changes
- fetch the current machinekit master
- rebase your branch ontop of machinekit/master

cdsteinkuehler, 2015-02-06:
@ftkalcevic: If you're using packages, just do a standard "apt-get update", "apt-get upgrade".  The TP patches referred to above were merged to master several days ago: https://github.com/machinekit/machinekit/pull/474

strahlex, 2015-02-06:
A colleague still encounters this problem with a 4 axis setup. Could this problem be caused by wrong or too low acceleration settings for the A axis?

strahlex, 2015-03-10:
I still encounter this error sometimes even with the 3 axes setup. Could this be related to the memory and CPU limitations of the BBB? Is there a relation between the TRAJ settings and CPU/memory usage?

robEllenberg, 2015-03-10:
It's possible, since the realtime side has to do a fair amount of work when
adding a new segment:

   - Build geometry from initial command
   - Run blending calculations / create blend segment
   - Run iterative optimization

One thing to try is reducing ARC_BLEND_OPTIMIZATION_DEPTH.

I'm not sure how floating point performance compares between the beaglebone
and x86 machines, but it may also be that some extreme cases (i.e. sqrt of
a very small number) perform acceptably on a desktop, but take just a
little bit too long on the beaglebone.

Does anyone else know of an easy way to log the motion thread's total
execution time? I'd really like to know how consistently it behaves, and
what our timing margins are. If we're well under the limit except for the
occasional spike, than I'd suspect numerical issues. If the average case is
near the limit, though, we may have to do something more drastic. For
example, this would be yet another argument to move as much of command.c
into userland.

-Rob

cdsteinkuehler, 2015-03-10:
On 3/10/2015 9:30 AM, Robert Ellenberg wrote:
> Does anyone else know of an easy way to log the motion thread's total
> execution time? I'd really like to know how consistently it behaves, and
> what our timing margins are. If we're well under the [...]

Doesn't each of the realtime functions have a HAL pin that represents execution time?  Can't that just be monitored via halscope or similar?

```
$ halcmd show pin motion-
Component Pins:
Owner   Type  Dir         Value  Name   Epsilon         Flags
    93  s32   OUT            65  motion-command-handler.time                    0
    93  s32   OUT          1305  motion-controller.time                 0
```

-- 
Charles Steinkuehler
charles@steinkuehler.net

strahlex, 2015-03-11:
I observed following through many failed prints:

- The error is more likely to happen when there are many GCodes without
a queue buster in between. Looks to my like something is accumulating
and slows down the execution.

- The error does usually happen on layers with infill. Could be related
to the previous point.

- The error is more likely to happen with smaller G64 tolerance values
(I use Q0.05 and P0.05 mostly). Without G64 explicitely set the error
does not happen at all (what is the default tolerance?).

Robert Ellenberg writes:

> It's possible, since the realtime side has to do a fair amount of work when
> adding a new segment:
>
>    - Build geometry from initial command
>    - Run blending calculations / create blend segment
>    - Run iterative optimization
>
> One thing to try is reducing ARC_BLEND_OPTIMIZATION_DEPTH.
>
> I'm not sure how floating point performance compares between the beaglebone
> and x86 machines, but it may also be that some extreme cases (i.e. sqrt of
> a very small number) perform acceptably on a desktop, but take just a
> little bit too long on the beaglebone.
>
> Does anyone else know of an easy way to log the motion thread's total
> execution time? I'd really like to know how consistently it behaves, and
> what our timing margins are. If we're well under the limit except for the
> occasional spike, than I'd suspect numerical issues. If the average case is
> near the limit, though, we may have to do something more drastic. For
> example, this would be yet another argument to move as much of command.c
> into userland.
>
> -Rob
>
>
>
>
>
>> I still encounter this error sometimes even with the 3 axes setup. Could
>> this be related to the memory and CPU limitations of the BBB? Is there a
>> relation between the TRAJ settings and CPU/memory usage?
>>
>>
>
>
> ---
> https://github.com/machinekit/machinekit/issues/412#issuecomment-78064667

strahlex, 2015-03-11:
How can I enable the TP debug output tp_debug_print

sam0737, 2015-05-02:
I am on a Apr 20 build and still hitting similar issue with
ARC_BLEND_ENABLE = 1
ARC_BLEND_FALLBACK_ENABLE = 0
ARC_BLEND_OPTIMIZATION_DEPTH = 25
ARC_BLEND_GAP_CYCLES = 4
ARC_BLEND_RAMP_FREQ = 20
and velocity extrusion configuration, and trivial kins.

What is queue buster? Is there any workaround by say - inserting a queue buster at every layer?

strahlex, 2015-05-02:
@sam0737 There is no fix at the moment that works. Playing with the parameters does not help.

However, you can workaround this bug by just not using the `G64 P0.1    ; path mode` command. That works fine for me until we find a solution.

mhaberler, 2015-05-02:
I recommend as a workaround to rework the code which causes the RT violation condition along the following algorithm: https://github.com/mhaberler/machinekit/blob/master/src/hal/components/rtmon.comp#L13-L31

make increment, decrement and limit pins, and default the values so the current behavior is replicated

then you can workaround by twisting those pins  - this is a more generally useful change

code to be fixed: https://github.com/mhaberler/machinekit/blob/master/src/emc/motion/control.c#L228-L297
(the rigidity of this existing motion code has never appealed to me anyway)

sam0737, 2015-05-02:
if there is a proposal of moving motion to userspace (implying non realtime?). (I forgot where did I see this idea)...does it mean the motion thread is not that time critical?
What about changing the motion thread to 5ms instead of 1ms? Does it mean it would gives 500us time for the thread to finish? If the record high time consumption is only at around ~150us...

mhaberler, 2015-05-02:
yes, it would give "time to finish". no, this is not an acceptable solution IMO. 

the lower half of motion (control) _is_ time critical, the upper (command) is not, but due to the way this is programmed it is hard to decouple - lots of shared variables which might have to be protected

strahlex, 2015-05-04:
The problem is the joint following error as a result of the RT delay. On a simulator config without feedback from the stepgens this is no problem at all. One solution could be to use an additinal PID loop and velocity control the stepgens -> a jump in the input position would most likely be no problem as the PID works a low pass filter to the position signal.

mhaberler, 2015-05-05:
The error @ftkalcevic reported actually comes from the xenomai layer and indicates the whole cycle budget (1mS) was exceeded (https://github.com/machinekit/machinekit/blob/master/src/rtapi/xenomai.c#L365-L374), not just the 10% or so window as motion calculates it to create a warning message; AFAICT this is caused by CPU usage, not a Xenomai domain switch which would relegate the thread to Posix scheduling; this would cause the `modesw=` counter to increase and it doesnt 

so in fact we have three event sources: the motion 10% indicator, the following error, and the '1mS exceeded' error reported by the xenomai layer, and to find the cause we have to look back in time at least once cycle, and measure the timing behavior of key code locations in the motion command handler

I think lttng is the tool of choice here as it is generally useful even if we have a bit of a learning curve now

this is how I would go about it:
- define two tracepoints and insert them here (my gut feeling): https://github.com/machinekit/machinekit/blob/master/src/emc/motion/control.c#L1465-L1467
- define a tracepoint on the event sources above (that actually can stop tracing)
- run the gcode reproducing the error
- pretty sure the last (or last few) times spent in tpRunCycle will show abnormal time use
- if that offender is identified we need to work up the food chain and instrument tpRunCycle in several places to narrow down the offending section

it might help to check the lttng docs/work with the mailing list if there is a way one can have a time-limited window back into history, together with a trigger which causes a snapshot of this window to be created, otherwise we might have an issue with the amount of stuff traced and writing it out

mhaberler, 2015-05-05:
it occurred to me only now the suggestion I made above (budget-based detection of time violations) will not help because that will not make the following error go away, rather just the motion message

sam0737, 2015-05-09:
I can reproduce the problem reliably with the gcode above. Strangly though it only occurs on the first program run, it does not occur afterwards, until the machinekit is restarted. I am not sure if it's the error message being suppressed, or it simply doesn't occur on the second one.

I hacked around the code, and think there might be something wrong in this segment:
https://github.com/machinekit/machinekit/blob/master/src/emc/tp/tp.c#L2043-L2049

This will repro the problem.
        long long x = *((long long*)&theta);
        rtapi_print_msg(RTAPI_MSG_ERR, "theta: %lld, %f, %f\n", x, theta, min_cos_theta);
        if (cos(theta) > min_cos_theta) {
            //tblend_vel = 2.0 * pmSqrt(acc_this * tc->tolerance / cos(theta));
            tblend_vel=10000;
            v_blend_this = fmin(v_blend_this, tblend_vel);
            v_blend_next = fmin(v_blend_next, tblend_vel);
        }

This will make the problem goes away.
        long long x = *((long long*)&theta);
        rtapi_print_msg(RTAPI_MSG_ERR, "theta: %lld, %f, %f\n", x, theta, min_cos_theta);
        if (0 && cos(theta) > min_cos_theta) {
            //tblend_vel = 2.0 * pmSqrt(acc_this * tc->tolerance / cos(theta));
            tblend_vel=10000;
            v_blend_this = fmin(v_blend_this, tblend_vel);
            v_blend_next = fmin(v_blend_next, tblend_vel);
        }

However, I see nothing wrong from the debug message.
All lines are "theta: -4701240790632769948, 0.000000, 0.001000"
I also don't understand why theta is not varying.

sam0737, 2015-05-09:
Now I do this instead:
```
rtapi_print_msg(RTAPI_MSG_ERR, "theta: #%d %lld, %f, %f\n", ++y, x, theta, min_cos_theta);
```
```
theta: #43 -4696772146649269816, 0.000000, 0.000000
theta: #44 -4696772146649269816, 0.000000, 0.000000
theta: #85 -4696772146649269816, 0.000000, 0.000000
theta: #86 -4696772146649269816, 0.000000, 0.000000
```
And found that some messages are missed.  What are the best way to log everything from the RT module?

ftkalcevic, 2015-05-09:
Unless something has changed, rtapi_prin_msg doesn't support printing floating point numbers (it has a limited subset of formatting - I don't know about 64bit numbers).  When I need to print floating point numbers I cast to the integer whole number and an integer decimal part

Eg (from memory)   "%d.%03d", (int)num, (int)(abs(num/1000))

mhaberler, 2015-05-09:
what has changed is - few people run kernel threads systems like RTAI any more since no good performance reasons left
the userland flavors (RT-PREEMPT, Xenomai, Posix) all support standard printf formats

sam0737, 2015-05-09:
I workaround the issue with something like this. It doesn't repro with the gcode above, and after 6 hours or so it's still happily printing. (usually it will hit something within 30 minutes of printing).

I still have no idea what's could be root cause.  The cos() calculation is bugged in kernel?

        double costheta = (theta < 0.001 || theta > -0.001) ? 1 : cos(theta);
        if (costheta > min_cos_theta) {
            tblend_vel = 2.0 * pmSqrt(acc_this * tc->tolerance / costheta);
            v_blend_this = fmin(v_blend_this, tblend_vel);
            v_blend_next = fmin(v_blend_next, tblend_vel);
        }

mhaberler, 2015-05-09:
@sam0737 - it makes zero sense to prop up the log subsystem until you can use printf-style debugging  like in some simple userland code at N times the servo rate from RT. The method is unsuitable  for this environment - you need to find a better one, like a better condition test _what_ and _when_ to log. And - what on earth do you do with N-thousand log entries a second, like actually read them?

It is exactly because printf and its rtapi_print* brethren are unsuitable we are bringing in the lttng.org userland tracing subsystem.

mhaberler, 2015-05-09:
@sam0737 - the code you isolate is an interesting smell - I read this as: one of the functions used there takes to much time, and it looks like its either cos() or fmin(). 

If one of those takes too long for the application we have that means it converges too slow, which is not necessarily a bug in terms of value returned but maybe the algorithm used in cos() or fmin() is unsuitable.

You might need a test program which evaluates the time behavior of this functions over a range of arguments. 

I would suggest to investigate this hypothesis either a) completely outside the motion code in a separate program or) instrumenting the code above with rtapi_get_clocks() and a trigger value (eg by a debugging hal pin) above which you trigger a log message.

sam0737, 2015-05-09:
it's probably the cos() but not fmin(). I tried leaving the ```if (cos(theta) > min_cos_theta)``` intact, but comment out the tblend_vel and fmin lines inside. It could reproduce the problem.

mhaberler, 2015-05-09:
well excellent, that reduces the search space

here's an idea: run the cos() function over a range of params in fine stepping and record/plot the execution times, maybe an aberration can be seen

probably each param should be run on a loop for a while; record the loop execution times with rtapi_get_clocks() and check if there is any spike

strahlex, 2015-05-10:
@mhaberler Maybe we should finally try http://gruntthepeon.free.fr/ssemath/neon_mathfun.html

strahlex, 2015-05-10:
Maybe more suitable for testing the behavior of a function depending on the input parameters is using a theorem checker e.g. frama-c. However, I am not sure it can find time problems.

strahlex, 2015-05-10:
This may not yield results but here is quick test: https://gist.github.com/strahlex/a94393bef7bc2b688fde

RunningLight, 2015-05-11:
@strahlex I played with your quick test. Unfortunately, `clock()` turns out to be a poor instrument. It increments by 10000 counts at each transition on both my Intel i5-4670K box and my Cubox-i4Pro ARM board. Hence, the resolution of `clock()` is only 0.01 sec even though its apparent precision is much higher: 1 count (e.g., a "tick") = 1 usec 

I used `gettimeofday()` to time how long it takes the output of `clock()` to change by 1000000; yup, it's consistent with the requirement that CLOCKS_PER_SEC is required by standards groups to be 1 million.

I know this is getting down in the weeds, but I'm trying to come up with another quick test to look for `cos()` corner cases.

mhaberler, 2015-05-11:
use rtapi_get_clocks(), this will be TSC resolution on x86 and 250Mhz res on a BBB

dlarue, 2015-05-11:
Is there anything I can do to help troubleshoot this? If it is the same issue, just tried another print of the Cyborg Beast prosthetic hand and it failed with a joint following error after just a few layers even though I'd printed for hours yesterday and the day before without issue.

strahlex, 2015-05-11:
@RunningLight What does the Nein mathfun stuff use for benchmarking? 

It could also work to use gprof to test the execution time of the cos function.

RunningLight, 2015-05-11:
@strahlex After unraveling the `#ifdef`'s  it looks like `neon_mathfun_test.c` also uses `clock()` on the BBB but uses it to loop for a relatively long time so the coarse resolution doesn't cause a major problem. After I enjoy a bit of real life this evening, I'll try @mhaberler's sensible suggestion to use `rtapi_get_clocks()`, first in your quick test to get my feet wet, then elsewhere.

strahlex, 2015-05-13:
Just tried to use the cephes_cos functions instead of the glib cos functions -> the rt-delay did not happen: here is the branch https://github.com/strahlex/machinekit/tree/rt-delay-hotfix

sam0737, 2015-05-25:
I'm on BBB + Replicape, and I have been printing with the rt-delay-hotfix for quite a few things (all kossel mini plastics parts) and everything is fine so far.

strahlex, 2015-05-26:
This PR here should fix this issue: https://github.com/machinekit/machinekit/pull/652

[1 more reply]

closed 2015-05-26 by zultron