machinekit/machinekit #413: cython v21 (jessie) does not work with MK

ArcEye, 2015-01-03:
Since commit 03a9ca125a200ba763e4b32fd1c9b837c65d0f46, forcing cython installation, noticed that v19 cython from wheezy-backports works, 
but v21 from jessie fails with errors relating to 
cimport statements being out of scope
(I am paraphrasing because I have cleared the original output, but essentially complaining that the
cimport specified was not in the PWD or immediate dirs)

Unsure if this requires an environment variable set, or it is plain broken

mhaberler, 2015-01-03:
hard to tell with the exact error message - could you make clean, rebuild and paste the error text? 

@zultron  and me are aware of some issues re cython versioning

I gave it a stab, and failed at the abysmal documentation of the cython change which caused the error to crop up; just couldnt make sense what actuall was wrong

ArcEye, 2015-01-03:
Knew you would say that, was not able to muck around with the repo until I had done the last PR.

Now removed v19 and installed v21 again
Relevant section of build output

<addr>
config.log
configure:2655: checking if cython is usable
configure:2661: checking for cython
configure:2679: found /usr/bin/cython
configure:2691: result: /usr/bin/cython
configure:2703: checking for Cython version
configure:2706: result: 0.21.1
configure:2753: Cython executable is '/usr/bin/cython'

build errors:
...............
Compiling hal/utils/pci_read.c
Linking pci_read
/usr/bin/cython -o hal/cython/machinekit/hal.c hal/cython/machinekit/hal.pyx

Error compiling Cython file:
------------------------------------------------------------
...
# made usable: Michael Haberler 6/2014

cimport cython
cimport hal_const
cimport ring_const
from .hal cimport *
^
------------------------------------------------------------

hal/cython/machinekit/hal.pyx:12:0: relative cimport beyond main package is not allowed

Error compiling Cython file:
------------------------------------------------------------
...

cimport cython
cimport hal_const
cimport ring_const
from .hal cimport *
from .rtapi cimport *
^
------------------------------------------------------------

hal/cython/machinekit/hal.pyx:13:0: relative cimport beyond main package is not allowed

Error compiling Cython file:
------------------------------------------------------------
...
cimport cython
cimport hal_const
cimport ring_const
from .hal cimport *
from .rtapi cimport *
from .hal_priv cimport *
^
------------------------------------------------------------

hal/cython/machinekit/hal.pyx:14:0: relative cimport beyond main package is not allowed

Error compiling Cython file:
------------------------------------------------------------
...
# hal_priv.h declarations

from .hal cimport *
^
[... 306 lines cut ...]
------------------------------------------------------------
...
from .hal_util cimport shmptr #hal2py, py2hal, shmptr, valid_dir, valid_type
from .hal_priv cimport MAX_EPSILON, hal_data
from .hal_group cimport *
^
------------------------------------------------------------

hal/cython/machinekit/hal_group.pyx:3:0: relative cimport beyond main package is not allowed

Error compiling Cython file:
------------------------------------------------------------
...
from .hal_util cimport shmptr #hal2py, py2hal, shmptr, valid_dir, valid_type
from .hal_priv cimport MAX_EPSILON, hal_data
from .hal_group cimport *
from .rtapi cimport  RTAPI_BIT_TEST
^
------------------------------------------------------------

hal/cython/machinekit/hal_group.pyx:4:0: relative cimport beyond main package is not allowed
make: *** [hal/cython/machinekit/hal.c] Error 1
make: Leaving directory `/usr/git/machinekit/src'

</addr>

mhaberler, 2015-01-03:
well that was exactly the error I was referring to, I couldnt make any sense/find docs/find a patch which gives a clue how to handle this

looks like tuning in on the cython devlist and waterboarding some devs is on order

mhaberler, 2015-01-03:
I've posted a question to https://groups.google.com/forum/#!forum/cython-users

mhaberler, 2015-01-05:
bad news - I think we're heading for troubles:

on Dec 26, wheezy-backports updated http://http.at.debian.org/debian-backports/pool/main/c/cython/cython_0.19.1+git34-gac3e3a2-1~bpo60+1_i386.deb 

that is bogus - despite the name, this yields cython 0.21.2:

root@nwheezy:/home/mah/machinekit-prepare/src# apt-get install -t wheezy-backports  cythonReading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  cython
0 upgraded, 1 newly installed, 0 to remove and 93 not upgraded.
Need to get 0 B/2,286 kB of archives.
After this operation, 7,877 kB of additional disk space will be used.
Selecting previously unselected package cython.
(Reading database ... 256187 files and directories currently installed.)
Unpacking cython (from .../cython_0.19.1+git34-gac3e3a2-1~bpo70+1_i386.deb) ...
Processing triggers for man-db ...
Setting up cython (0.19.1+git34-gac3e3a2-1~bpo70+1) ...

However: 
root@nwheezy:/home/mah/machinekit-prepare/src# /usr/bin/cython --version
Cython version 0.21.2

thanks to the above, and the fact we just made cython a build requirement, this means mk is currently not buildable from packages!

mhaberler, 2015-01-05:
currently the only way to get at cython 0.19 is like so:

 apt-get install python-pip
 pip install --upgrade git+git://github.com/cython/cython@0.19.x

mhaberler, 2015-01-05:
strike out the above regarding debian wheezy-backports cython being broken. It does report 0.19.2 - which is ok. My fault.

mhaberler, 2015-01-05:
well a minute ago a friendly cython user posted a suggestion: https://groups.google.com/forum/#!topic/cython-users/wWxLMuHc56Q%5B1-25-false%5D

the fix was:   touch hal/cython/machinekit/__init__.py

works fine now with Cython version 0.21.2 - I think we can close the cython saga for good soonish ;)

ArcEye, 2015-01-06:
Well done, now I know what all those empty __init__.py files are for:)

mhaberler, 2015-01-06:
at last.. after making a complete fool out of myselves ;)

closed 2015-01-06 by strahlex

cdsteinkuehler, 2015-01-06:
On 1/6/2015 1:40 AM, ArcEye wrote:
> Well done, now I know what all those empty __init__.py files are for:)

I saw a few of those and was confused by them as well.

Perhaps they should contain a comment indicating they are required to
identify the directory as a python module?

-- 
Charles Steinkuehler
charles@steinkuehler.net

commit 448b723b (2015-01-06): cython: fix #413

thanks to https://groups.google.com/forum/#!topic/cython-users/wWxLMuHc56Q%5B1-25-false%5D
commit 59f9bddb (2015-01-06): Merge pull request #417 from mhaberler/master

cython: fix #413