machinekit/machinekit #423: Change linear deltakins to allow joint offsets

jpmoniz, 2015-01-10:


luminize, 2015-01-12:
@cdsteinkuehler @jpmoniz 
I've taken the liberty to add the latest replies from this thread so we can discuss and continue, and if needed reference to this issue in the commits or coming (hopefully) PR.

>On 12 Jan 2015, at 15:31, Charles Steinkuehler  wrote:

>This should probably become an issue on github at some point...

>>On 1/12/2015 6:40 AM, Bas de Bruijn wrote:

>>>On 12 Jan 2015, at 13:15, Charles Steinkuehler wrote:

>>>> On 1/12/2015 12:35 AM, Bas de Bruijn wrote:
Hi, I think for a efficient way of calibrating we’d need both JP's
and Charles’ solution. (Only talking about manual calibration, let’s
keep it simple for now)

>>>>Charles, I have the feeling (while you made the HAL pins you
mentioned something of memory use cleanup) that there is still some
legacy stuff (old way). Could that not be the source of the move?
because the move to the “old” offset (INI file value) seems to be in
a commanded motion. No f-error despite the big change in value. So
perhaps it’s still being commanded to go to the old_copy-value of the
INI file.

>>>Someone needs to test this change on a system with homing switches to
see if it works as expected.  If so, I can easily clean up the code and
send a PR.  I was confused by the move after homing, but it actually
should move if HOME != HOME_OFFSET, even on my machine w/o switches.

>>Yes of course. I hadn't thought about HOME and so I didn't change HOME.
I ran on a physical machine with physical home switches yesterday evening FWIW.

>So if this is working, the question is:

>Should I export any other joint parameters while messing with the code?
https://github.com/cdsteinkuehler/linuxcnc/blob/home_offset/src/emc/motion/motion.c#L1009-L1056
...I think it might make sense to export:

>joint->max_pos_limit = 1.0;
joint->min_pos_limit = -1.0;
joint->vel_limit = 1.0;
joint->acc_limit = 1.0;
joint->min_ferror = 0.01;
joint->max_ferror = 1.0;
joint->home = 0.0;
joint->backlash = 0.0;

>...and it might make sense to export:

>joint->home_search_vel = 0.0;
joint->home_latch_vel = 0.0;
joint->home_final_vel = -1;
joint->home_offset = 0.0;
joint->home_sequence = -1;

>...anything else is probably asking for trouble.

>Charles Steinkuehler

and

>On 12 Jan 2015, at 07:35, Bas de Bruijn wrote:
Hi,
I think for a efficient way of calibrating we’d need both JP's and Chatles’ solution. (Only talking about manual calibration, let’s keep it simple for now)

>Charles,
I have the feeling (while you made the HAL pins you mentioned something of memory use cleanup) that there is still some legacy stuff (old way). Could that not be the source of the move? because the move to the “old” offset (INI file value) seems to be in a commanded motion. No f-error despite the big change in value. So perhaps it’s still being commanded to go to the old_copy-value of the INI file.

>JP
I will try to test your branch tonight (europe, morning while I write this).
I think the f-error issue is still there, because when I calibrate, I go to the tower position in world mode. go down to the bed in small steps. I’d do that with the “debugpins” lineardeltakins.J0off

>Andy,
I think about it this way: we need to be able to find the switch distance error (all cumulated error of the position during build and manufacturing). I’d use JP’s kinematics pins for this + charles’ method of changing the home_offset value.
find all 3 offset values (or 2 actually because the first one sets the pace, it’s the reference the second 2 should refer to).
adjust the delta radius in the same way.
when done, we know everything.
calculate the correct tower height (ini value + the J0off value),
set axis.0.home_offset with the resulting value from above
set J0off to zero again
change the INI values in the .ini file
re-home the machine
all without having to switch off the machine multiple times.

>Bas

>>On 12 Jan 2015, at 02:43, jmoniz wrote:

>>The f-error is not an issue anymore in my mind. When I was playing with it last night 2AM‎. My mind was not thinking well. There is no logical need to adjust the offset while in world. So what I suspect I will do is just create an Mcode that switches to joint and apply the offset then signal the user to re-home the machine. When applying the offset in joint mode, motion is not allowed for some reason (my guess is something to do with telop). Applying the offset in joint is not an issue as you have to be in joint anyways to home. After homing again the probing can be repeated to confirm.

>>JP

jpmoniz, 2015-01-12:
@luminize  @cdsteinkuehler 

Ok I have my probe setup working and have done some initial tests.  When I get home tonight I will be testing further.

I would export all of the pins mentioned. If nobody needs them thats ok.  "Better looking at them then for them"

jpmoniz, 2015-01-12:
Should I put a PR in for the changes to lineardeltakins?

luminize, 2015-01-12:
Hi Jean-Paul, I just built your branch, and when I change the J{n}off then I see the world x,y and z coordinates change. I would have expected that the joint being addressed would move, without affecting the x, y and z coordinates.

jpmoniz, 2015-01-12:
Ah crap.  That commit does not have the inverse code in it.

luminize, 2015-01-12:
if you'd push it, i'll pull it and I'll compile and test it, together with Charles' changes

jpmoniz, 2015-01-12:
Done

luminize, 2015-01-12:
On 12 Jan 2015, at 22:11, Jean-Paul Moniz <notifications@github.com> wrote:

> Done
> 
just pulled and is building

luminize, 2015-01-12:
built and joint offset moves as expected, without changing the cartesian value.

jpmoniz, 2015-01-13:
I have managed to test it tonight. Seems to work well.  Just writing the automatic calculation part now.

http://youtu.be/_9li_FC7pRQ

darenschwenke, 2015-01-13:
Before you get too far into it, you could probably just lift this entire section.  That was my plan before your last comment:
https://github.com/RichCattell/Marlin/blob/master/Marlin_main.cpp#L1507

closed 2015-01-17 by zultron

jpmoniz, 2015-01-17:
@zultron  Should I have put a PR in for the changes to lineardeltakins? or are we just going with Charles changes?

cdsteinkuehler, 2015-01-17:
On 1/16/2015 9:40 PM, Jean-Paul Moniz wrote:
> @zultron  Should I have put a PR in for the changes to
> lineardeltakins? or are we just going with Charles changes?

I'm generally in favor of extending linerdeltakins with pins to allow
"tweaking" as required by auto-calibration (adjustments to not only
delta radius and arm length, but tower offsets and similar.

I think these are needed regardless of the home offset changes I made
(and Michael Haberler tweaked).  There will likely be some overlap, but
one place Machinekit can clearly do better than the AVR controllers is
complex kinematics that don't require simplifying assumptions or
short-cuts to solve.  So let's go ahead and add the tuning knobs!

-- 
Charles Steinkuehler
charles@steinkuehler.net

jpmoniz, 2015-01-17:
Ok just got confused a bit.

darenschwenke, 2015-01-17:
The code I referenced is the multiple point bed calibration routine capable of not just radius and height correction, but also ideally detecting flaws in tower positioning.  It requires more variables to store these offsets.

luminize, 2015-01-17:
@jpmoniz I also think your pins are very welcome. Together with ease of use of the changeable offsets calibrating is a lot less hassle