machinekit/machinekit #428: Building for BBB also builds other flavours

luminize, 2015-01-11:
I recently noticed something strange when I compiled after running 
```
$ make clean
$ git clean -xfd
$ debian/configure -x
```
I specifically did not configure with the option --with-platform=beaglebone
I cd to src and `./configure` gives me this snip:
```
checking -lrt for RT_PREEMPT and POSIX... ok
checking whether to build POSIX threads... yes
checking whether to build RT_PREEMPT threads... yes
checking for xeno-config... /usr/bin/xeno-config
checking usability of xenomai utility, /usr/bin/xeno-config... yes
checking Xenomai runtime library... works
checking whether to build Xenomai userland threads... yes
checking whether to build Xenomai kernel threads... no
checking for rtai-config... no
checking whether to build RTAI threads... no
checking whether to build hardware drivers... yes
```

now I would not expect to tee a "yes" for POSIX and TR_PREEMPT

then later on when making:
```
ln -s . rtapi/posix
Compiling rtapi/posix/rtapi_app.cc
Compiling rtapi/posix/rtapi_compat.c
Compiling rtapi/posix/rtapi_hexdump.c
Compiling rtapi/posix/rtapi_support.c
Linking rtapi_app_posix
Preprocessing abs.comp
Preprocessing abs_s32.comp
...
Creating mesa_7i65.mak
Creating halcomp-srcs/hal/drivers/mesa_7i65.mak
Creating mesa_uart.mak
Creating halcomp-srcs/hal/drivers/mesa_uart.mak
...
more compiling for posix
Compiling realtime halcomp-srcs/hal/components/abs.c
Linking ../rtlib/posix/abs.so
Compiling realtime halcomp-srcs/hal/components/abs_s32.c
Linking ../rtlib/posix/abs_s32.so
...
and on and on the list goes
same for RT_PREEMPT
Compiling realtime halcomp-srcs/hal/components/abs.c
Linking ../rtlib/rt-preempt/abs.so
Compiling realtime halcomp-srcs/hal/components/abs_s32.c
Linking ../rtlib/rt-preempt/abs_s32.so
...
```

Regardless of drivers for mesa cards which do not belong on the BBB (but there maybe the option --with-platform=beaglebone would help), This seems live a waste of valuable time and resources on the BBB platform.

is this part of #336 ?

mhaberler, 2015-01-11:
no, posix builds are always built since the origins of ubcX - it is the 'simulator mode' which allows you to run on a stock kernel (bad path quality though). It is useful to have around for that purpose, and trying to isolate errors which are specific for a particular RT kernel.

rt-preempt is posix except for minor conditional code, and its the most widely available RT kernel on  x86; very few on arm though.

luminize, 2015-01-11:
Ok, configuring with posix means "posix only".
But why would I want to have rt-preempt built on a Beaglebone? I didn't ask rt-preempt to be built because I only did `debian/configure -x`

cdsteinkuehler, 2015-01-11:
On 1/11/2015 2:46 PM, Michael Haberler wrote:
> no, posix builds are always built since the origins of ubcX - it is
> the 'simulator mode' which allows you to run on a stock kernel (bad
> path quality though). It is useful to have around for that [...]

...and the drivers are being built by default at my request.  While the
BBB doesn't have a PCI or PCIe slot, there are *MANY* ARM boards that
do, so the default builds add user-mode PCI drivers by default.  Also,
there are non-real-time applications (like the LinuxCNC controlled
sprinkler system) that need hardware drivers on platforms other than
x86/RTAI.

I think there's a way to disable the user-mode PCI drivers, but I never
bother.  After the first build it doesn't take much in the way of extra
Makefile processing cycles.  And usually I'm building on a quad-core
iMX6 which goes much faster than the 'Bone regardless.  :)

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2015-01-11:
First, this is not a 'BB build', it is an ARM7 build with drivers for the BB platform enabled. While everybody runs Xenomai on the BB, this does not mean everybody runs _just_ Xenomai on an ARM; in fact there are (few and mediocre) RT-PREEMPT kernels for ARM (btw the primary reason for the UBC effort which landed in mk was to get rid of the tie-in of a build with a particular RT kernel, and that was considered progress)

I think you can disable a thread flavor by configuring --witout-rt-preempt or so.

luminize, 2015-01-11:
I now fully understand.
Close this issue

closed 2015-01-11 by luminize

zultron, 2015-01-12:
Sounds like you want

    ./configure --with-xenomai --disable-usermode-pci

That enables Xenomai userland threads *only*, and disables PCI.