machinekit/machinekit #432: Python: machinekit.config contains unsubstituted variables

strahlex, 2015-01-13:
The machinekit.config library provides configuration time paths to different Machinekit directories. It use the common @variable@ substition. However, the substituted variables contain bash style environment variable substitions like for example ${prefix}. Shouldn't these variables also be substituted for the use in the Python library?

e.g. in my config.py

    self.datarootdir = "${prefix}/share"
and

    self.prefix="/home/alexander/bin/machinekit-1"

Correctly substituted path would be

    self.datarootdir = "/home/alexander/bin/machinekit-1/share"

strahlex, 2015-01-14:
The problem seems to be the substition performed by the configure script just substitues paths without environment variable nesting. 

Fixing this for the Python module shouldn't be a problem, but isn't this a general problem with the substition? If so this should be fixed in the substitution script (which I cannot find)

strahlex, 2015-01-14:
This patch fixes the problem: https://github.com/strahlex/machinekit-1/commit/902546c4342016638398abf8a18ed1d9c4004320
Not sure it should be fixed in the configure script. This problem could also affect other files that use "@" type variable substitution.

zultron, 2015-01-14:
This is related to the unfortunate distinction between RIP and system installs.  IIRC the trouble is that evaluation only happens once, which is normally sufficient because `${prefix}` is not intended to be defined within `configure.in`.  Your solution is no worse than my ugly hack with `scripts/gen-rtapi.ini.sh.in.tmpl`.

strahlex, 2015-01-15:
Okay, I created the PR. Most users will not play around with @variables@ anyway. I think it's safe to fix this at configure time.

closed 2015-01-31 by strahlex