machinekit/machinekit #433: generating .asciidoc for pushing to machinekit-man

luminize, 2015-01-13:
We need to use the (generated) manpages for building .asciidoc files that represent homo-sapiens readable documentation of i.e. the components and drivers. Buildbot thing for pushing to machinekit-man

zultron, 2015-01-13:
Is there a `Makefile` target for generating those?  The rest of the infrastructure probably belongs in the Buildbot.

luminize, 2015-01-14:
@zultron 
maybe you could give me a nudge, because I'm breaking my head over making the Makefile. I'm trying to start simple here, and work outwards (in difficulty) from there. First goal is to generate `.asciidoc` files from some `.1` manfiles

the Makefile i'm making is in `man/man-generated`
the .1 files I'm trying to convert are in `man/man1`
the script is in `scripts/`

I can do what I want if I run from the terminal (`source=...` and `target=...` is terminal output from the script):
```
Bass-MacBook-Pro-3:man-generated basdebruijn$ python ../../scripts/troff-to-asciidoc.py ../man1/axis-remote.1 ../man1/axis-remote.1.asciidoc
source  = ../man1/axis-remote.1
target  = ../man1/axis-remote.1.asciidoc
```
But that needs to be automated :) so here's my try:
```
SOURCES = $(shell  find ../man1 -type f -iname "*.1")
TARGETS := $(SOURCES:.1=.1.asciidoc)

$(TARGETS):
%.asciidoc: %.1
	python ../../scripts/troff-to-asciidoc.py $< $@
```
After running `make` I'm getting the message:
`make: Nothing to be done for `../man1/axis-remote.1.asciidoc'.`
I have the feeling the solution is around the corner, but I can't find the corner :)

luminize, 2015-01-14:
I'm making progress. I now get a generated `.1.asciidoc` file but it's the only one so far. Like only the first one is used and not all in "TARGETS"
```
SOURCES := $(shell  find ../man1 -type f -iname "*.1")
TARGETS := $(SOURCES:.1=.1.asciidoc)

$(TARGETS)::
%.1.asciidoc: %.1
	$(shell python ../../scripts/troff-to-asciidoc.py $*.1 $*.1.asciidoc)
```
doing a `make -p` yields:
```
SOURCES := ../man1/axis-remote.1 ../man1/axis.1 ../man1/comp.1 ../man1/gladevcp.1 ../man1/gs2.1 ../man1/hal_input.1 ../man1/halcmd.1 ../man1/halmeter.1 ../man1/halrun.1 ../man1/halsampler.1 ../man1/halstreamer.1 ../man1/haltcl.1 ../man1/halui.1 ../man1/io.1 ../man1/iocontrol.1 ../man1/linuxcnc.1 ../man1/linuxcncrsh.1 ../man1/pyvcp.1 ../man1/shuttlexpress.1 ../man1/vfdb_vfd.1 ../man1/vfs11_vfd.1
TARGETS := ../man1/axis-remote.1.asciidoc ../man1/axis.1.asciidoc ../man1/comp.1.asciidoc ../man1/gladevcp.1.asciidoc ../man1/gs2.1.asciidoc ../man1/hal_input.1.asciidoc ../man1/halcmd.1.asciidoc ../man1/halmeter.1.asciidoc ../man1/halrun.1.asciidoc ../man1/halsampler.1.asciidoc ../man1/halstreamer.1.asciidoc ../man1/haltcl.1.asciidoc ../man1/halui.1.asciidoc ../man1/io.1.asciidoc ../man1/iocontrol.1.asciidoc ../man1/linuxcnc.1.asciidoc ../man1/linuxcncrsh.1.asciidoc ../man1/pyvcp.1.asciidoc ../man1/shuttlexpress.1.asciidoc ../man1/vfdb_vfd.1.asciidoc ../man1/vfs11_vfd.1.asciidoc
```

zultron, 2015-01-14:
First, `$(TARGETS)` is already used in `Makefile`.

This seems to work:
```
MAN_SOURCES = $(shell  find ../man/man1 -type f -iname "*.1")
MAN_TARGETS = $(patsubst ../man/man1/%,../man/man1/%.asciidoc,$(MAN_SOURCES))
$(MAN_TARGETS): %.asciidoc: %
        python ../scripts/troff-to-asciidoc.py \
            $< $@
manpages:  $(MAN_TARGETS)
```

zultron, 2015-01-14:
(That's meant for `src/Makefile`, where `$PWD` is the `src` subdirectory.)

luminize, 2015-01-14:
Thanks John, That's absolutely fabulous. I have to adapt my way of thinking with the Makefile. I now get a complete list of generated `.asciidoc` files. The next step will now be easy to put in the `man-generated/{insert directory}`.
I'm planning on filtering out `.3rtapi`, `.3hal` and `.3hm2` and putting them into the applicable directory. Something like `.3hal`, `.9` and `.9comp` in `man-generated/hal-components`, `.3hm2` in `man-generated/drivers` etcetera. Making it easy for making indexes and linking from `machinekit-docs` to `machinekit-man`.

luminize, 2015-01-15:
@zultron @mhaberler 
Here's the branch from which we could try to build the `.asciidoc` files and push to `machinekit-man`

I built the .asciidoc files manually with (from `src` directory): `make -f ../man/man-generated/Makefile manpages`
 
I pushed to the repo with from the root directory with: `git subtree push --prefix=man/man-generated machinekit-man master` where `machinekit-man` was at the time my repo.

Let me know if something's amiss. Especially if there are other thoughts about the directory structure or the way we could get overview for linking from `machinekit-docs` to `machinekit-man`

luminize, 2015-01-18:
any thoughts how I would get parse-able files from the links inside some manpages?
an example is `io.1` [here](https://github.com/machinekit/machinekit/blob/master/man/man1/io.1) where there's just 1 line containing `.so man1/iocontrol.1`

luminize, 2015-01-18:
aaah, this seems to be a link to `iocontrol.1`. so if I t type `man io` I get the manpage for `iocontrol`. @zultron, unless you object I will remove this cruft.

zultron, 2015-01-19:
Hi @luminize,

I don't know how to handle the `man io` case.  As far as I'm concerned, you're the expert!

Also see my [confused question][1] on the PR.

[1]: https://github.com/machinekit/machinekit/pull/445#issuecomment-70558795

luminize, 2015-01-19:
Just replied. I'll remove it since it seems some kind of "shortcut". If anybody complains I'll put it back.

PR #462 (2015-01-22): man/man-generated rework, this PR replaces PR #445
this branch replaces #445 by not having the asciidoc files added and committed.
fixes #433
fixed #434
and prepares has a Makefile in man/man-generated which can be used to generate `.asciidoc` files
closed 2015-01-22 by zultron