machinekit/machinekit #435: gantry_latching build error

luminize, 2015-01-13:
I'm building the latest on my 64 bit machine and noticed the following error
```
machinekit@debianXW6600:~/MK/machinekit/src$ git log
commit dbee84839df93e77a9729347de0a020171c1a283
Merge: b94e955 9dbec0a
Author: John Morris <john@zultron.com>
Date:   Mon Jan 12 23:53:46 2015 -0600

    Merge pull request #430 from strahlex/gantry-latching
```
below the build error:
```
  CC [M]  /home/machinekit/MK/machinekit/src/objects/xenomai-kernel/3.8-1-xenomai.x86-amd64/halcomp-srcs/hal/components/gantry.o
make[2]: Leaving directory `/usr/src/linux-headers-3.8-1-xenomai.x86-amd64'
make[5]: *** No rule to make target `/home/machinekit/MK/machinekit/src/objects/xenomai-kernel/3.8-1-xenomai.x86-amd64/halcomp-srcs/hal/components/gantry_latching.o', needed by `/home/machinekit/MK/machinekit/src/objects/xenomai-kernel/3.8-1-xenomai.x86-amd64/abs.o'.  Stop.
make[4]: *** [_module_/home/machinekit/MK/machinekit/src/objects/xenomai-kernel/3.8-1-xenomai.x86-amd64] Error 2
make[3]: *** [sub-make] Error 2
make[2]: *** [all] Error 2
make[1]: *** [modules] Error 2
make[1]: Leaving directory `/home/machinekit/MK/machinekit/src'
make: *** [modules] Error 2
make: Leaving directory `/home/machinekit/MK/machinekit/src'
```

zultron, 2015-01-14:
Curious.  The buildbot got through there with no problem, and I can't reproduce on my workstation (same commit) either.  Curiouser, I can't explain why `gantry_latching.o` would be a build-prereq of `abs.o`.

Can you do the ol' `git clean -xdf` (after being sure your work is committed!) and try again before we dig deeper?

luminize, 2015-01-14:
My mind must have melted after my experimenting in making a makefile.
I normally use `git clean -xfd` after a pull, but I seem to have forgotten last night. Compiles OK now.

closed 2015-01-14 by luminize