machinekit/machinekit #441: build: hostmot2 dependencies broken

mhaberler, 2015-01-15:
Michal reported this, I reproduced:

micges@debian:~/PROJECTS/mk/src$ touch hal/drivers/mesa-hostmot2/hostmot2.h
micges@debian:~/PROJECTS/mk/src$ make
Reading 276/277 dependency files
Done reading dependencies

Reading 0/0 realtime dependency files
Done reading realtime dependencies
make: Wejście do katalogu `/home/micges/PROJECTS/mk/src'
Making modules for flavor xenomai
Reading 284/285 dependency files
Done reading dependencies

Reading 0/236 realtime dependency files
Done reading realtime dependencies
make[1]: Wejście do katalogu `/home/micges/PROJECTS/mk/src'
make[1]: Opuszczenie katalogu `/home/micges/PROJECTS/mk/src'
make: Opuszczenie katalogu `/home/micges/PROJECTS/mk/src'
micges@debian:~/PROJECTS/mk/src$

cdsteinkuehler, 2015-01-15:
On 1/15/2015 1:21 PM, Michael Haberler wrote:
> Michal reported this, I reproduced:
>
> micges@debian:~/PROJECTS/mk/src$ touch hal/drivers/mesa-hostmot2/hostmot2.h
> micges@debian:~/PROJECTS/mk/src$ make
> Reading 276/277 dependency files
> Done reading [...]

I had a similar problem when updating src/emc/motion/motion_debug.h, it
didn't force anything to recompile, just the .c files I changed.

I figured it was something I was doing wrong, but perhaps the depend
system has some major recent breakage?

-- 
Charles Steinkuehler
charles@steinkuehler.net

zultron, 2015-01-15:
I'm pretty sure this has been going on for a while, quite possibly since my UB build system updates.

I'll try to figure it out tonight.

micges, 2015-10-15:
Any solutions for this yet?

closed 2018-08-04 by ArcEye