machinekit/machinekit #447: Post install script breaks Debian policy (and RCN build scripts)

cdsteinkuehler, 2015-01-17:
I am having problems generating a Machinekit image with the RCN build scripts because the postinst script violates Debian policy.  The postinst script is directly calling init scripts to restart logging when it should be using the invoke-rc.d abstraction layer:

https://github.com/machinekit/machinekit/blob/master/debian/machinekit.postinst#L22

https://www.debian.org/doc/debian-policy/ch-opersys.html#s9.3.3

This causes the host OS logging daemon to restart when running in a chroot, which then prevents the chroot from unmounting properly (open files) and chaos ensues.  If the proper abstractions were being used, the request to restart the logging daemon would be ignored when inside the chroot, and all would be well with (this tiny little part of) the world.

zultron, 2015-01-17:
Nice catch.  I was seeing this in the Buildbot, and used the sledgehammer approach to fix.

cdsteinkuehler, 2015-01-17:
You also need to craft a /usr/sbin/policy-rc.d script in your chroot.  For an example see: https://wiki.debian.org/chroot#Configuration

closed 2015-01-17 by strahlex