machinekit/machinekit #451: Motion has no real IO pins

strahlex, 2015-01-20:
The motion component exposes input as well as output pins to HAL. However, the component has no real IO pins supported by HAL. For some applications it is necessary to use IO pins instead of output pins as they do not force a certain value upon a signal.

For example:
[image]
This diagram shows a mechanism to set a signal from GCode as well as from a remote component. It also stores the values on change in some sort of storage (could be the config-service). The in2io component "converts" a output pin of the motion component to a IO pi. When the trigger is triggered the IO value gets updated. This whole in2io component would be reduntant if motion would support io pins natively.

**Possible solutions:**

Make all output pins of the motion component IO pins. They would ignore value changed when connected to signals, but update the signal when their value changes.

Add additional IO pins to the motion component.

Combine the input and ouput pins of motion to single IO pins.

strahlex, 2015-01-23:
Solution 1 is trivial to implement and simply works:
https://github.com/strahlex/machinekit-1/commit/6aa54913e8fdd6a2de9493b383552f3e80f477a1

strahlex, 2015-01-23:
Solution 4 is similar trivial:
https://github.com/strahlex/machinekit-1/commit/6fc6468abc022c3549f6c380e07a1d1a1f1bff63

strahlex, 2015-01-25:
Proposed solution 4. A similar method is used by the lincurve component.

strahlex, 2015-01-25:
Temporarily fixed by  #465

I leave this issue open since motion still does not support real I/O

mhaberler, 2015-01-27:
to the above solutions I'd add:

- make the tp code accept non-move commands and not have it trip over
- add motion command to set/query arbitrary pins by commands on the tp queue

closed 2018-08-04 by ArcEye