machinekit/machinekit #455: Configuration failed "configure: error: failed to find required module pyftpdlib.servers"

xknot, 2015-01-21:
```
# debian/configure -px
[...]
checking python module: pyftpdlib.servers... no
configure: error: failed to find required module pyftpdlib.servers
```
I'm using a qemu chroot environment configured according to http://code.google.com/p/picnc/wiki/QemuChrootCompiling
I've read https://github.com/machinekit/machinekit/issues/317 carefully, and add `none /dev/shm tmpfs rw,nosuid,nodev,noexec 0 0` to the `/etc/fstab` according to kinsamanka as saying, but I'm no lucky.
Any hint will be helpful.

strahlex, 2015-01-22:
As the error says the package python-pyftplib is missing or not new enough. Do you have the package installed and in which version?

xknot, 2015-01-22:
python-pyftpdlib has been installed:
```
root@linuxcnc:/etc/apt# dpkg -l python-pyftpdlib
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                  Version         Architecture    Description
+++-=====================-===============-===============-===============================================
ii  python-pyftpdlib      1.2.0-3~1da~whe all             Python FTP server library
```

strahlex, 2015-01-22:
1.2 should be fine. Can your run python in your chroot environment and execute `import dpkg -s python-pyftpdlib`

xknot, 2015-01-22:
In  the chroot:
```
mount /dev/shm
```
It works!

It seems like I met the problem described as:
http://stackoverflow.com/questions/6033599/oserror-38-errno-38-with-multiprocessing

Is there any way not  `mount /dev/shm` every time chroot? May be add something into the `/etc/fstab` of the host?

zultron, 2015-01-22:
I use a script to start my chroots.  It mounts `/dev/shm` and other stuff, too.

In use cases I've encountered, it wouldn't make sense to add chroot mounts to `/etc/fstab` unless there's only a single chroot in exactly the same place, and probably not even then.

xknot, 2015-01-23:
Using a script is a good idea, thanks for your help.

strahlex, 2015-01-25:
Can we close this issue?

zultron, 2015-01-25:
I'll close it.  Please reopen or submit new issue in case of further problems.

closed 2015-01-25 by zultron