machinekit/machinekit #461: Missing support for Xilinx Zynq processors/FPGA

claudiolorini, 2015-01-22:
I'd like to provide support code compilation for Zynq processors used for example in the Zedboard platform: http://zedboard.org/product/zedboard

I've already succesfully ported the Linuxcnc 2.5 base developed by MH and I'd like to share the results.

strahlex, 2015-01-22:
Sounds promising, go ahead.

mhaberler, 2015-01-27:
@claudiolorini : which kernel did you use? 

linuxcnc 2.5 is firmly stone-age as far as non-x86 support goes (read as: never)

who's MH?

claudiolorini, 2015-01-28:
I'm currently using this:
3.8.0-xilinx #6 SMP PREEMPT Tue Jan 27 10:19:40 CET 2015 armv7l
an 'almost vanilla' kernel from the Xilinx Git, patched with xenomai-2.6.4
but my plans are to try soon a xenomai-3.0 with a more recent kernel 
something like 3.17 or 3.18...
I'm running on a Linaro environment, one of the many Ubuntu flavours.

Yes, you're right it was a 2.6.0~pre something, from: git://git.mah.priv.at/emc2-dev.git, 
i suppose you are familiar with the place and M.H., aren't you? ;-)

In the meantime I managed to rebuild machinekit from sources, it was quite straightforward apart from 
some troubles from some 'python protobuf' stuff, but nothing that our friend google cannot solve...
The tests in the scripts dir are running smoothly, today i'll start to add some serious stuff like 
IO driver for the Zedboard and CAN bus drivers... then i can start testing on some real HW.

Btw, I wasn't able to follow  what happened between linuxcnc and machinekit but i suppose 
is something related to the extreme 'conservatorism' in keeping up with new developments...
As far as i'm concerned x86 platforms, parallel ports, RTAI are things of the past and i want to 
keep pace with the future (read ad ARM platforms, Xenomai)... so machinekit is very welcome!

C.

mhaberler, 2015-01-28:
ah ok. Good to see you caught up afterall.
your guess about the reasons for machinekit sounds just about right.. we're pleased to see some ideas finally find adoption over there - of course not without the obligatory start-from-scratch rewrite

medium term we might need to migrate (or maybe write new flavors for the xenomai 3 RT API, so any knowledgeable person is more than welcome!

claudiolorini, 2015-02-15:
...I finally managed to modify the config and makefile in order to add the hal drivers for GPIO and CAN bus for the Zedboard, something is starting to run but i have some behaviors i cannot fully understand.
for example in Axis, Machine -> Show Axis Configuration gives an error on stdout:
<pre><code>Error in startup script: couldn't load file "/home/linaro/machinekit/tcl/hal.so": /home/linaro/machinekit/tcl/hal.so: undefined symbol: zframe_send
    while executing
"load $::linuxcnc::TCL_LIB_DIR/hal.so"
    (procedure "hal" line 2)
    invoked from within
"hal list $node"
    (procedure "listHAL" line 5)
    invoked from within
"listHAL"
    (procedure "refreshHAL" line 12)
    invoked from within
"refreshHAL"
    (file "/home/linaro/machinekit/tcl/bin/halshow.tcl" line 578)
</code></pre>

suppose it's related to ZMQ, any pointer on which direction to look? any particular release of ZMQ is needed? I think i've installed last available (2.2.0)...

mhaberler, 2015-02-15:
actually these are the hal TCL bindings, and AFAICT there is no zeromq involved her (any other evidence?)

it looks the tcl hal.so is either not built at all or not accessible for some reason - check the configure and build logs

claudiolorini, 2015-02-15:
...i finished rebuild, I don't crosscompile so it takes quite a lot.
/home/linaro/machinekit/tcl/hal.so seem to be compiled but continues to complain:
 undefined symbol: zframe_send 
that's why i was thinking about (c)ZMQ...

i'll try to check the 'config.log', are build logs saved somewhere or should i redirect outputs?

mhaberler, 2015-02-15:
hm, right, that's from zmq
oh, I see where this is coming from, tcl/hal.so uses parts of halcmd to issue rt commands, and that needs zeromq

could you post the output of 
  ld --version
and 
  ldd tcl/hal.so

mine shows it's linked against libzmq and libczmq:

mah@nwheezy:~/machinekit-prepare/tcl$ ldd hal.so 
	linux-gate.so.1 =>  (0xb777b000)
	liblinuxcncini.so.0 => /home/mah/machinekit-prepare/lib/liblinuxcncini.so.0 (0xb775a000)
	liblinuxcnchal.so.0 => /home/mah/machinekit-prepare/lib/liblinuxcnchal.so.0 (0xb7732000)
	libX11.so.6 => /usr/lib/i386-linux-gnu/libX11.so.6 (0xb75e0000)
	libXss.so.1 => /usr/lib/i386-linux-gnu/libXss.so.1 (0xb75dd000)
	libXext.so.6 => /usr/lib/i386-linux-gnu/libXext.so.6 (0xb75cb000)
	libXft.so.2 => /usr/lib/i386-linux-gnu/libXft.so.2 (0xb75b6000)
	libfontconfig.so.1 => /usr/lib/i386-linux-gnu/libfontconfig.so.1 (0xb7580000)
	libpthread.so.0 => /lib/i386-linux-gnu/i686/cmov/libpthread.so.0 (0xb7566000)
	libdl.so.2 => /lib/i386-linux-gnu/i686/cmov/libdl.so.2 (0xb7562000)
	libm.so.6 => /lib/i386-linux-gnu/i686/cmov/libm.so.6 (0xb753c000)
	libtk8.5.so.0 => /usr/lib/libtk8.5.so.0 (0xb7404000)
	libGL.so.1 => /usr/lib/i386-linux-gnu/libGL.so.1 (0xb73aa000)
	libXinerama.so.1 => /usr/lib/i386-linux-gnu/libXinerama.so.1 (0xb73a6000)
	libgcc_s.so.1 => /lib/i386-linux-gnu/libgcc_s.so.1 (0xb7389000)
	libc.so.6 => /lib/i386-linux-gnu/i686/cmov/libc.so.6 (0xb7224000)
	libstdc++.so.6 => /usr/lib/i386-linux-gnu/libstdc++.so.6 (0xb7138000)
	liblinuxcnc-pb2++.so.0 => /home/mah/machinekit-prepare/lib/liblinuxcnc-pb2++.so.0 (0xb6f58000)
	librt.so.1 => /lib/i386-linux-gnu/i686/cmov/librt.so.1 (0xb6f4e000)
	libprotobuf.so.7 => /usr/lib/libprotobuf.so.7 (0xb6e52000)
	libz.so.1 => /lib/i386-linux-gnu/libz.so.1 (0xb6e39000)
	libczmq.so.1 => /usr/lib/i386-linux-gnu/libczmq.so.1 (0xb6ded000)
	libzmq.so.3 => /usr/lib/i386-linux-gnu/libzmq.so.3 (0xb6d44000)
	libnative.so.3 => /usr/lib/libnative.so.3 (0xb6d3a000)
	libxenomai.so.0 => /usr/lib/libxenomai.so.0 (0xb6d33000)
	libxcb.so.1 => /usr/lib/i386-linux-gnu/libxcb.so.1 (0xb6d10000)
	libfreetype.so.6 => /usr/lib/i386-linux-gnu/libfreetype.so.6 (0xb6c74000)
	libXrender.so.1 => /usr/lib/i386-linux-gnu/libXrender.so.1 (0xb6c6a000)
	libexpat.so.1 => /lib/i386-linux-gnu/libexpat.so.1 (0xb6c41000)
	/lib/ld-linux.so.2 (0xb777c000)
	libglapi.so.0 => /usr/lib/i386-linux-gnu/libglapi.so.0 (0xb6c2b000)
	libXdamage.so.1 => /usr/lib/i386-linux-gnu/libXdamage.so.1 (0xb6c28000)
	libXfixes.so.3 => /usr/lib/i386-linux-gnu/libXfixes.so.3 (0xb6c22000)
	libX11-xcb.so.1 => /usr/lib/i386-linux-gnu/libX11-xcb.so.1 (0xb6c20000)
	libxcb-glx.so.0 => /usr/lib/i386-linux-gnu/libxcb-glx.so.0 (0xb6c06000)
	libXxf86vm.so.1 => /usr/lib/i386-linux-gnu/libXxf86vm.so.1 (0xb6c00000)
	libdrm.so.2 => /usr/lib/i386-linux-gnu/libdrm.so.2 (0xb6bf2000)
	libsodium.so.10 => /usr/lib/libsodium.so.10 (0xb6b82000)
	libXau.so.6 => /usr/lib/i386-linux-gnu/libXau.so.6 (0xb6b7e000)
	libXdmcp.so.6 => /usr/lib/i386-linux-gnu/libXdmcp.so.6 (0xb6b78000)

claudiolorini, 2015-02-16:
linaro@linaro-nano:~/machinekit/tcl$ ld --version
GNU ld (GNU Binutils for Ubuntu) 2.23.52.20130913
up to here nothing to complain.... <code><pre> linaro@linaro-nano:~/machinekit/tcl$ ldd hal.so 
	liblinuxcncini.so.0 => /home/linaro/machinekit/lib/liblinuxcncini.so.0 (0xb6f51000)
	liblinuxcnchal.so.0 => /home/linaro/machinekit/lib/liblinuxcnchal.so.0 (0xb6f2d000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0xb6f01000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xb6e96000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0xb6e74000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb6d90000)
	/lib/ld-linux-armhf.so.3 (0xb6f7d000)
	libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0xb6ce4000)
	liblinuxcnc-pb2++.so.0 => /home/linaro/machinekit/lib/liblinuxcnc-pb2++.so.0 (0xb6c33000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb6c28000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0xb6c1a000)
	libprotobuf.so.7 => /usr/lib/libprotobuf.so.7 (0xb6b72000)
	libz.so.1 => /lib/arm-linux-gnueabihf/libz.so.1 (0xb6b56000)
</code></pre>
...ok seems that just a small part of the libraries is linked!
have to check the build process...

mhaberler, 2015-02-16:
yeah, that's suspicious

claudiolorini, 2015-03-14:
...ok guys, i think i've managed to have machinekit running on the Zedboard!
Mainly I made 2 hal drivers, one for the GPIOs and one for the CAN Bus communication and some 
small interventions in the Makefile/configure.in in order to compile.

- I've tested the GPIO with some HMI hardware i had handy (emergency stop button, limit switches and Jog encoder) and everything seems fine. I'm not sure if the GPIO is fast enough to use for step/dir but for HMI seems adequate.

- the CAN bus driver manages the communication protocol with the 2FOC motor control board that drives 2 brushless motors with encoder, the motor controller is running in 'speed loop' receiving from machinekit a continuous stream of speed setpoints and providing machinekit a continuous stream of position feedback, closing the position loop at around 300usec. 
The CAN bus interface is done with a CAN8 board that manages 8 separated CAN bus lines, the CAN bus IP is placed in the FPGA section of the Zync and interfaced to the CPU with a Axi bus.
At the moment I've tried the motion with a single z axe, but i plan to soon start with tests on a 6 axis setup...
 
I'd like to commit a pull-request in order to keep the code as aligned as possible with the developments of machinekit but i've never done it (i'm more of an Hardware guy...). I've pushed the development on my copy of mk, should I make a pull request on the makinekit repository? is that sufficient or should i do something else? 

I also have the .hal and .ini files that i've used for the tests, is it interesting to have them as example? if yes tell me where to put them and i'll upload that too. 

C.

strahlex, 2015-03-15:
Claudio Lorini writes:

> I'd like to commit a pull-request in order to keep the code as aligned
> as possible with the developments of machinekit but i've never done it
> (i'm more of an Hardware guy...). I've pushed the [...]
First make a git rebase origin master. Then create the PR.

>
> I also have the .hal and .ini files that i've used for the tests, is
> it interesting to have them as example? if yes tell me where to put
> them and i'll upload that too.
There is a configs directory in the Machinekit tree. You may see that
there is a ARM subfolder and then there is the BeagleBone
subfolder. Please create a new subfolder for Xiling Zynq and add you
example configs there.
https://github.com/machinekit/machinekit/tree/master/configs/ARM

>
> C.

>
> ---
> https://github.com/machinekit/machinekit/issues/461#issuecomment-80684999

claudiolorini, 2015-03-15:
...ok, done, hope i't all ok.
Please be gentle with a noob. ;-)
C.

closed 2015-03-17 by claudiolorini