machinekit/machinekit #504: Add Package Upgrade Warning

cdsteinkuehler, 2015-02-25:
Since the recent change to making TP a loadable HAL module breaks existing user configurations, it seems like there should at least be a warning to that effect when performing a package upgrade.

I know when I've upgraded various official Debian packages I will sometimes get notices about configuration file formats changing, locations moving, etc.  I'm not enough of a packaging wizard to know how this is accomplished, but it seems like a prudent thing to do for the user base.

I'm not real worried about having automatic configuration updates or maintaining full backwards compatibility, but I think the first time a user finds out they need to edit their configuration shouldn't be when the application fails to launch after an upgrade.

mhaberler, 2015-02-26:
there is a debian way to have a warning message displayed during an apt-get upgrade, it comes up once in a while eg with security issues

a quick google trail didnt turn anything up though?

RunningLight, 2015-03-19:
From http://www.fifi.org/doc/debconf-doc/tutorial.html

> Debconf should be used whenever your package needs to output something to the user, or ask a question.

but, from https://www.debian.org/doc/manuals/maint-guide/dother.en.html

> Upgrades to the new version should be silent and non-intrusive (existing users should not notice the upgrade except by discovering that old bugs have been fixed and perhaps that there are new features).
>
> When the upgrade is necessarily intrusive (eg., config files scattered through various home directories with totally different structure), you may consider as the last resort switching the package to a safe fallback state (e.g., disabling a service) and providing the proper documentation required by policy (README.Debian and NEWS.Debian). Don't bother the user with debconf notes invoked from these maintainer scripts for upgrades.

Despite the latter, I'm guessing artful use of `debconf` may be the answer?

RunningLight, 2015-03-20:
Follow-up.

In https://www.debian.org/doc/manuals/debian-faq/ch-pkg_basics.en.html, I came across this statement in Section **7.6 What is a Debian preinst, postinst, prerm, and postrm script?**

> postinst
> This script typically completes any required configuration of the package foo once foo has been unpacked from its Debian archive (".deb") file. Often, 'postinst' scripts ask the user for input, and/or warn the user that if he accepts default values, he should remember to go back and re-configure that package as the situation warrants. Many 'postinst' scripts then execute any commands necessary to start or restart a service once a new package has been installed or upgraded.

My mindset previously was that we warn before upgrading the package so the user could decide whether to proceed. This section reminds me that we could just warn after upgrading the package that changes have to be made (e.g., "you bought it, it's your problem now").

ArcEye, 2018-08-01:
Historical, tp work completed 3.5 yrs ago and all configs use it, unless someone tries to load a very old one

closed 2018-08-01 by ArcEye