machinekit/machinekit #513: installation of machinekit on debian based linux mint

luminize, 2015-03-01:
I've been dabbling with linux mint. Debian release with MATE edition.
Is this a path worth taking?

When running trying to install the build debs I get the following
```
bas@mbp-mint ~/machinekit $ debian/configure -p
debian/control:  copied base template
debian/rules:  copied base template
debian/control:  added POSIX threads package
debian/rules:  enabled posix threads
Unknown distro 'LinuxMint'
Usage:  debian/configure [ arg ... ]
   arg:		function:
   -p		build POSIX threads
   -r		build RT_PREEMPT threads
   -x		build Xenomai threads
   -X <kver>	build Xenomai-kernel threads ***
   -R <kver>	build RTAI-kernel threads ***
   -t <tclver>	set tcl/tk version
      *** Argument may be repeated for multiple kernels
bas@mbp-mint ~/machinekit $ sudo scripts/apt-installbuilddeps 
apt-get install
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 29 not upgraded.
dpkg-checkbuilddeps: warning: can't parse dependency @BUILD_DEPS@ @TCL_TK_BUILD_DEPS@
dpkg-checkbuilddeps: error: error occurred while parsing Build-Depends/Build-Depends-Arch/Build-Depends-Indep
```

strahlex, 2015-03-02:
Does it use the Debian repos? Then I see no reason why it shouldn't
work. Something like Ubuntu which is based on Debian but not compatible
at all might be harder. What is your reason for using a Debian based
Mint instead of Debian?

Bas de Bruijn writes:

> I've been dabbling with linux mint. Debian release with MATE edition.
> Is this a path worth taking?
>
> When running trying to install the build debs I get the following
> ```
> bas@mbp-mint ~/machinekit $ [...]

the-snowwhite, 2015-03-02:
Machinekit does build and run on Ubuntu 14.04 lts
here is an except from notes on my first compilation
cd machinekit
debian/configure -px
sudo scripts/apt-installbuilddeps
cd src
./autogen.sh

luminize, 2015-03-02:
well, the apt-installbuilddeps should be an easy way to add needed packages.

my `etc/apt/sources.list.d/` shows `machinekit.list`, `wheezy-backports.list` and `official-package-repository.list`.

the last one contains:
```
deb http://packages.linuxmint.com debian main upstream import 

deb http://debian.linuxmint.com/latest/ testing main contrib non-free
deb http://debian.linuxmint.com/latest/security testing/updates main contrib non-free
deb http://debian.linuxmint.com/latest/multimedia testing main non-free

deb http://extra.linuxmint.com debian main
```

as to why linux mint? I got the impression that linux-mint is more of a "rolling release". Furter more its less strict about non-free stuff. And since its built on debian I didn't expect troubles.

I think I'll start at looking at the "unknown distro" and the "dpkg-checkbuilddeps".

luminize, 2015-03-02:
@the-snowwhite 
if you don't have a xenomai patched kernel you should omit the `x` in `debian/configure -px` and use the option appropriate for you situation.

the-snowwhite, 2015-03-02:
forgot to paste the ./configure before
I tried re-compiling from a clean Machinekit clone but got the same setuid error as before
Can you specify which options are necessary to specify where
 to get logging to work in
this simulation environment ?

RunningLight, 2015-03-03:
Sorry to be late to the party.

Linux Mint is an obvious target for folks interested in general usage. I use it myself on all my non-development hosts. It's based on Ubuntu which is based on Debian, yada yada yada. Let's postpone to another day whether the MK build system should support every variant.

@luminize, you're tripping over the following routine in debian/configure

```
do_tcl_tk_version() {
    TCL_TK_VER="$1"
    if test -z "$TCL_TK_VER"; then
        # No -t option specified; to ensure a reproducible package,
        # use the latest version from the distro
        TCL_TK_VER=8.6
        case "$DISTRO_ID" in
            Debian)  test "${DISTRO_RELEASE/.*/}" -lt 8 && TCL_TK_VER=8.5 ;;
            Ubuntu)  test "${DISTRO_RELEASE/.*/}" -lt 14 && TCL_TK_VER=8.5 ;;
            *)  usage "Unknown distro '${DISTRO_ID}'" ;;
        esac
    fi
    TCL_TK_BUILD_DEPS="tcl${TCL_TK_VER}-dev, tk${TCL_TK_VER}-dev"
    TCL_TK_DEPS="tcl${TCL_TK_VER}, tk${TCL_TK_VER}"
    echo "debian/control:  Set tcl/tk build deps to version $TCL_TK_VER" >&2
}
```
where, early in the same script

```
# Distro info
#
# This should run on non-Debian distros; lsb_release is only used for
# TCL/TK version detection, which can be passed in through the '-t
# VERSION' switch
if test -x /usr/bin/lsb_release; then
    DISTRO_ID=$(lsb_release -is)  # Debian or Ubuntu
    DISTRO_RELEASE=$(lsb_release -rs)   # 8.1, 14.04, etc.
fi
```
So, you'll need another test in the case statement in `do_tcl_tk_version()`

```
LinuxMint)  test "${DISTRO_RELEASE/.*/}" -lt 16 && TCL_TK_VER=8.5 ;;
```

where I arbitrarily made the release version number 16 since I had LinuxMint version 16 running on the host I tried this on whereas 17.1 is the latest version.

When I made this adjustment, then I got as far as src/configure which tripped on

```
configure: error: failed to find required module google.protobuf.compiler.plugin_pb2
```

I haven't built from source for a while so I don't know if this is a more general issue or if it's strictly a LinuxMint issue. I'll look into it later today.

NOTE: I don't care for the way `debian/configure` tests for distros, etc. so please don't consider this response an endorsement of it!

RunningLight, 2015-03-03:
followup - I just created a clean LinuxMint-17.1-Mate virtual machine. Upping the version number to 17 in the above fix, I get all the way through the Machinekit build and the resulting simulator starts.

Don't know (don't care actually) whether my previous configure error was due to version 16 or my clumsiness.

luminize, 2015-03-03:
I'll give your solution a try!
Thx

RunningLight, 2015-03-03:
FYI, 

I should specify the two systems I was working with were 32-bit LinuxMint 16 Xfce with the 3.11.0-12-generic kernel and 64-bit LinuxMint 17.1 Mate with the 3.13.0-37-generic kernel.

The LM16 system has been in daily use and I can't easily say what has been added/modified. The LM17.1 system is a freshly installed virtual machine with no additions or mods other than what is specified on the machinekit.io/docs/building-from-sources page and the one-line mod to debian/configure which I describe above.

luminize, 2015-03-03:
@RunningLight your one-line addition was the key. Its building as we type.
I'll refrain from making a PR since I only have the posix thread on the VM. no RT patched kernel.

closed 2015-03-03 by luminize