machinekit/machinekit #531: UUID in machinekit.ini is always the same

strahlex, 2015-03-13:
The Machinekit Debian packages feature a default machinekit.ini file. There is an UUID values in this configuration file that is relevant to identify different Machinekit instances on the network. This UUID should be autogenerated after package install to ensure a different identification for every machine.

RunningLight, 2015-03-19:
Since some of us like to clone from one VM instance to another and otherwise reuse an installed Machinekit, it would be nice to provide a script which invokes `uuidgen` and then `sed`s the result into `machinekit.ini`

The script then could be used during in the postinst phase and again later.

mhaberler, 2015-03-19:
could you cook up a PR?

RunningLight, 2015-03-19:
I'll give it a shot.

mhaberler, 2015-03-20:
I think it boils down to intelligently deciding which file to fix:
- the global machinekit.ini if it's a package install or a 'make install' thing
- a build from source using rip-environment

on a packaged install there's no 'make something', so probably adding a script to fix up the global machinekit.ini is the way to do it

in a source build, a make target might be more appropriate, so the result is permanent if user decides she wants a 'make install' eventually.

it's a late fallout of our forebear's singletonitis - what do you mean, instances, we do no stinkin' instances around here ;)

RunningLight, 2015-03-20:
Agreed, where 'a build from source using rip-environment' implies fixing the `machinekit.ini` file in the build-directory's local `./etc` directory not the system-wide `/etc` directory.

As for 

> on a packaged install there's no 'make something', so probably adding a script to fix up the global machinekit.ini is the way to do it

I was thinking that the script I proposed would be executed auto-magically as a `postinst` procedure during the machinekit-package install, but still be available for use cases like I mentioned (cloning an installed system). Of course it could also be a make target for source builds.

mhaberler, 2015-03-20:
yes, but only if you find default UUID in the global machinekit.ini, so it's idempotent and doesnt change on every package update

RunningLight, 2015-03-24:
@mhaberler the value of MKUUID is generated every time Machinekit is built from source (see `./src/configure`) whether by the user at home or by the BuildBot.  Hence, I don't understand your use of "default UUID" and "idempotent" in this context.

The script itself can be pretty simple. Here's one possibility. Create a shell script containing the following lines and make it executable:

```
#!/bin/sh
newuuid=`uuidgen`
sed -i.old "s:\(MKUUID=\).*:\1$newuuid:" $1
```

I called mine `newmkuuid.sh`

Then, executing 
```
newmkuuid.sh machinekit.ini
```
saves the original file as `machinekit.ini.old` before overwriting the UUID in `machinekit.ini`. Belts and suspenders, as John would say.

This script can be used as needed. Obviously, it could be much more elaborate, e.g., test that it's been given a legitimate `machinekit.ini` file, save successive versions, comment out the old MKUUID= line and insert a new one, etc.

darenschwenke, 2015-03-24:
'ship' without it set.  Set it if not set on first run.

cdsteinkuehler, 2015-03-24:
On 3/24/2015 8:55 AM, Daren Schwenke wrote:
> 'ship' without it set.  Set it if not set on first run.

Perhaps even placing the UUID in it's own file (as with things like ssh
keys) so there's no potentially dangerous automated mangling of the ini
file required.

-- 
Charles Steinkuehler
charles@steinkuehler.net

closed 2018-08-04 by ArcEye