machinekit/machinekit #534: Cannot switch to world mode after jogging homed joints.

sliptonic, 2015-03-17:
Version: debian package 0.1.1073-1da.bbot.pr533.gitac38116~1wheezy.

Steps to reproduce:
Start MK, select the sim/axis/vismach/scara sample configuration.
Hit 'home all'
Select joint 0 and jog it a little bit.
Select view/world mode.

You'll get a message that all joints must be homed before switching to teleop mode even though all joints are already homed.  The view menu indicates that it is in world mode, but the preview still shows joint names and not cartesian axis names (XYZ) Re-homing has no effect.  

Same procedure but do not jog axis.  Switching to world mode works correctly.  User can switch back forth between the modes as long as joint is not jogged in joint mode.

mhaberler, 2015-03-18:
please add complete config

sliptonic, 2015-03-18:
Kernel 3.8-1-xenomai.x86-686-pae.
running on an older x86 dell pentium pc.  

Otherwise, this is a stock config running from packages with an unmodified sim machine config.
I'm looking for issue guidelines for what a "complete" config is but unsure what else you need.

closed 2018-08-04 by ArcEye