machinekit/machinekit #544: Did someone really use a filename with an embedded wildcard? Checkout fails on windoze.

modmaker, 2015-03-24:
Trying to clone the repo on windows for a quick glance. But the checkout failed on the file mentioned below. Of course everybody is using Linux, but should we make it impossible to use something else in case of emergencies? ;-)

suspected file: **machinekit/share/fdm/thermistor_tables/vishay_NTCS0603E3104*XT.txt**

-- Bas

strahlex, 2015-03-24:
Oh, thermistor name copy and paste error.

Bas Laarhoven writes:

> Trying to clone the repo on windows for a quick glance. But the checkout failed on the file mentioned below. Of course everybody is using Linux, but should we make it impossible to use something [...]

modmaker, 2015-03-24:
Real programmers don't cut and paste ;-)
Sorry. How are we going to fix this? Is there some dependency on this file or can it be renamed without consequences?

strahlex, 2015-03-25:
Can be renamed. But grep the name to make sure no config uses it and if so rename it there as well.

closed 2015-04-07 by mhaberler