machinekit/machinekit #554: Unsatisfactory RT performance in Xilinx Zynq processors/FPGA GPIO driver

claudiolorini, 2015-04-06:
Current GPIO driver for the Zedboard has a poor RT behavior due to the presence in the code of 
standard linux syscalls, i propose to modify it in order to guarantee proper RT performance using
direct memory address access to the Zynq GPIO peripheral.

mhaberler, 2015-04-06:
poor.. I am rather sure your RT threads get relaxed and run as posix

you have to options - rough: direct mmap I/O as in hal_bb_gpio
or the "proper xenomai way" - an RTDM driver - unfortunately "proper" translates into "Xenomai only"

if you google Christophe Blaess gpio rtdm you should find some hints

claudiolorini, 2015-04-06:
Yeah, as long as you don't touch anything it's work...ish, but if you do anything... BANG!.
I've been testing the modifications as in hal_bb_gpio, it seems that a HAL generated PWM 
jitter stays within 20usecs under any load condition, so maybe can be of use also for motion 
with step/dir drivers...
As soon as i've polished the code and HAL examples i'll issue a PR.

mhaberler, 2015-04-06:
thats pretty good, bb jitter is around 40uS

claudiolorini, 2015-04-06:
...the zynq has a dual processor and i've done some IRQ balancing and kept the system overhead at minimum, maybe it's due to that. I've also noted that with the simple PWM generation and running 
simple G-Codes the load stays around 30% on both CPUs, that's encouraging...

closed 2015-04-06 by claudiolorini

claudiolorini, 2015-04-07:
...my memory is fading, i remembered 20usecs but... 
i've checked the xscope in the lab and got it almost right ;-) (zeros doesn't count, does it?).
Make it more like 200usecs...

mhaberler, 2015-04-07:
hm, what do you get on latency-test or latency-histogram on this kernel? pastebin a nice image ;)

on the bb latency-test is in the 40uS ballpark

claudiolorini, 2015-04-07:
...here there are, the tails seems a bit high, consistent with the xcope readings.
[image]

[image]

claudiolorini, 2015-04-07:
moving irqs around:
MMC card IRQ to CPU2:
[image]

Eth0 IRQ to CPU2:
[image]

things changes slightly but xcope measures more or less remains the same...

claudiolorini, 2015-04-13:
Some developments:
kernel 3.14.17 Xenomai patched, some advancements but still far from 40uS jitter!

[image]

[image]