machinekit/machinekit #563: Python 2.7 vs Python 3

RunningLight, 2015-04-16:
This isn't a "sky is falling" issue for now, but we have about 300 python files in our source repo and many are inconsistent with Python 3. Some conflicts can be resolved with minor syntactic changes, e.g., to modify the argument lists of old-style print statements; some require actual thought.

Why do I mention this, especially since I still write Python the way I learned it before the turn of the century? Because I was just reading a posting by a Debian team member (re)confirming their commitment to eliminate Python 2 from Debian by 2020.  Five years isn't such a long time that we shouldn't be thinking about this problem now. 

At a minimum, I believe we should commit to support both Python 2 and 3 in any new modules we write and to begin to port the modules we already have. We might even build pylint or something like it into our code testing.

To quote from https://docs.python.org/3/howto/pyporting.html

> A key point about supporting Python 2 & 3 simultaneously is that you can start today! Even if your dependencies are not supporting Python 3 yet that does not mean you can’t modernize your code now to support Python 3. Most changes required to support Python 3 lead to cleaner code using newer practices even in Python 2.

> Another key point is that modernizing your Python 2 code to also support Python 3 is largely automated for you. While you might have to make some API decisions thanks to Python 3 clarifying text data versus binary data, the lower-level work is now mostly done for you and thus can at least benefit from the automated changes immediately.

Several automation tools which aid in porting and testing are mentioned in the above-referenced page.

mhaberler, 2015-04-17:
what about doing an example migration of a single Python file for a start, and document the steps to be done (stuff installed, changes) as a PR?

also a command to be inserted into the 'Python checking' make rule(s) might help, at least for generating warnings so we're aware of what needs to change?

unfortunately these checking commands are littered over dozens of Submakefiles, example here:

https://github.com/machinekit/machinekit/blob/master/src/hal/user_comps/Submakefile#L10-L14

RunningLight, 2015-04-17:
Makes sense to me. I'll tackle it next week.

Do we have a Machinekit Code-writing Style Sheet anywhere? If I knew, I've forgotten.

strahlex, 2015-04-17:
Here is my Emacs setup with pylint, pyflakes and pep8 as well as whitespace highlighting enabled
https://gist.github.com/strahlex/44e77b43d67d1d0db180

strahlex, 2015-04-17:
pep8 also has a Python3 conversion mode, with the tools above enabled only a few incompatibilites reside, mostly related to renamed modules (e.g. ConfigParser and configparser)

RunningLight, 2015-04-18:
Cool, thanks.

strahlex, 2015-04-19:
The ConfigParser issue can be easily workout around using:
    
    if sys.version_info >= (3, 0):
        import configparser
    else:
        import ConfigParser as configparser

strahlex, 2015-04-19:
Turns out Python3 chokes on the Cython libraries e.g. `from machinekit import hal`

    ImportError: /home/alexander/bin/machinekit-1/lib/python/machinekit/hal.so: undefined symbol: _Py_ZeroStruct

A quick Google search returned that the speciic python library need to be linked during compilation as well meaning that linking the Python3 library could help.

machinekoder, 2018-03-13:
May I point out that Python 2 is EOL 2020. We should finally consider moving to Python 3.

There are good tools and modules available to make the code Python 3 compatible such a `six` for example.

The Machinetalk stuff is already Python 3 compatible.

machinekoder, 2018-03-22:
The clock is ticking: [Legacy Python Death Clock](https://pythonclock.org/)

[Futurize](http://python-future.org/futurize.html) might be another option to move the code to Python3.

The hard part will be the EMC/Linuxcnc CPython bindings and other Python2 only dependencies.

closed 2018-08-04 by ArcEye