machinekit/machinekit #565: required cleanups of the instantiation features

mhaberler, 2015-04-18:
catch-all issue for instantiation stuff. @strahlex  mentioned two:

- functs emitted by instcomp are called '<instance>.funct'; to be in line with comp it should just be '<instance>', no need to tack on a type indicator which isnt used in comp
- rt.newinst() does not autoload the comp if not yet loaded, other than halcmd which does

mhaberler, 2015-04-18:
the rt.newinst() issue is a bit harder to fix than it looks, because rt is a level below hal and does not see HAL components as such; newinst is conceptually a HAL-level operation even if it needs to go through through the RT API to have it executed

one solution would be to add a hal.newinst() function which just wraps the rt.newinst() method and checks the component is loaded; I'll go for that and see how it works out

strahlex, 2015-04-19:
The hal.newinst solution would be trivial
    if not compType in hal.components:
        rt.loadrt(compType)

strahlex, 2015-04-19:
wrapping loadrt into the HAL library would make using the Python HAL library easier anyway as the user does mostly need the hal module functions

mhaberler, 2015-04-19:
not at all, rather by explicitly calling into the  rtapi layer, for instance here: https://github.com/machinekit/machinekit/blob/master/nosetests/unittest_icomp.py#L27

strahlex, 2015-04-19:
newinst could also automatically prefix loaded component types with i_ e.g. `newinst add2` would check if component `i_add2` is loaded and then execute `newinst i_add2` and the rtapi layer

strahlex, 2015-04-19:
This PR changes the HAL rtapi API to use loadrt: https://github.com/machinekit/machinekit/pull/575

mhaberler, 2015-04-19:
@strahlex  re "automatically prefix loaded component types with i_.."

I dont think this is a good idea, because there is no reason to keep two sets of components forever

@ArcEye - I'd say we watch for a month or two how this works out, and work out the kinks; then we just remove the singleton components and migrate everything to instantiable comps

it's a bit of work in the second migration, but the duplicity does not make sense; switching now is a bit early

strahlex, 2015-04-20:
@mhaberler Thats why I would iike to add the i_ prefix internally so we can remove it when the legacy components are gone. Furthermore, there is no need for users to differentiate between instantiable components and legacy components. The i_ prefix just adds complexity and confusion. Hiding this internally makes sense to me.

mhaberler, 2015-04-20:
I see. Makes sense. So I guess the logic would be:

- usercommand 'newinst foo <args>'
- assuming i_foo exists, loadrt i_foo but as HAL component name foo, and then call 'newinst foo <args>'
- if i_foo does not exist, loadrt foo as foo, and proceed as above

@ArcEye: d'accord with this?

ArcEye, 2015-04-20:
The more I looked at converting existing hal files, the more I wondered if the solution lay in halcmd.
If halcmd took the command 
loadrt and2 count=3  

and actually did

loadrt i_and2
newinst i_and2 and2.0
newinst i_and2 and2.1
newinst i_and2 and2.2

The user would get what they expected and existing files will still work
(the i_ prefix is simply to prevent name collisions in the sources and compiled modules at present, so yes it can go)

Would only work for count= and names= because personality / cfg are gone, so actually quite simple.

If the component is a legacy one which is unconverted/unconvertable, halcmd just does what it used to do.

This might be the basis of a longer term solution?

mhaberler, 2015-04-20:
yes, I think this could be permanent, no second migration

ArcEye, 2015-04-20:
The jump off place for the change might be about line 1112 of halcmd_commands.c

The first consideration appears to be the method for deciding if the component is an instantiated one or not.

The simplest would appear to be to search the module for a keyword which will not be in a legacy module
'instantiate' springs to mind

You could use grep -cim to get a return value, but as getting a return value from an external process is fraught, may be simpler just to open the module for reading and scan for an occurence of 'instantiate'.

mhaberler, 2015-04-20:
yes, I think some sort of symbol-table-tagging is the way to go, so one can check object file properties without actually dlopen()/insmod() the module

I'll look into that, it might make sense to have an API_VERSION macro or so which does export a detectable property, it just needs to work across kmods and shared libs

btw we already do some ELF fiddling here: https://github.com/machinekit/machinekit/blob/master/src/rtapi/rtapi_compat.c#L496-L512

mhaberler, 2015-07-06:
merged instance branch. Done.

closed 2015-07-06 by mhaberler