machinekit/machinekit #566: hal/drivers/hal_zed_gpio.c:272:5: error: unknown type name ‘FILE’ - Can't Compile

dlktdr, 2015-04-18:
I found machinekit wouldn't compile due to hal_zed_gpio.c missing #include<stdio.h>

The error I got,
hal/drivers/hal_zed_gpio.c:272:5: error: unknown type name ‘FILE’
hal/drivers/hal_zed_gpio.c:272:5: warning: implicit declaration of function ‘fopen’ [-Wimplicit-function-declaration]

I added the missing header and it compiles fine. Not sure how to go about modifying the code tho.

mhaberler, 2015-04-18:
"machinekit wouldn't compile" is not an answerable question

please post the full detail: configure.log, and your platform details.

claudiolorini, 2015-04-18:
...are trying to compile for the Zedboard platform?
can you post the command sequence to compile you used? should be something like:
./autogen.sh
./configure
make
sudo make setuid

dlktdr, 2015-04-19:
No I wasn't trying to compile for the Zedboard platform, compiling on Beagle Bone Black

Commands I used
```
sudo apt-get install git dpkg-dev
git clone https://github.com/machinekit/machinekit.git 
cd machinekit/
debian/configure -px
sudo scripts/apt-installbuilddeps
cd src/
./autogen.sh
./configure && make
```

Did get it working by adding the header, just thought I would post in case anyone else ran into it too.

mhaberler, 2015-04-19:
you didnt give a --with-platform-beaglebone spec

strahlex, 2015-04-19:
Had the same problem a few days ago. #include <stdio.h> is missing in the zedboard driver

strahlex, 2015-04-19:
#include <studio.h>

strahlex, 2015-04-19:
The zedboard platform should not be compiled by default without speciifying additional configure parameters. Optionally just including stdio.h is a good idea anyway and does no additional harm.

claudiolorini, 2015-04-19:
I agree, I'll make a PR for the include, but the reason why it happens is not clear to me... 
touching the Makefile is a bit 'dangerous', i can try to modify the default building but i's better
someone check the modifications...

closed 2015-04-19 by strahlex