machinekit/machinekit #572: reserved identifier violation

elfring, 2015-04-19:
I would like to point out that identifiers like "[`_Pb__ContainerType`](https://github.com/machinekit/machinekit/blob/d4277a5aaa7fb7c4816b692fb8568678d7376b0a/src/machinetalk/include/container.h#L29)" and "[`_PBUTIL_HH_INCLUDED`](https://github.com/machinekit/machinekit/blob/fde1283509d2f363e3410f34871afe1c72cc928c/src/machinetalk/include/pbutil.hh#L19)" [do not fit](https://www.securecoding.cert.org/confluence/display/cplusplus/DCL51-CPP.+Do+not+declare+or+define+a+reserved+identifier "Do not use identifiers that are reserved for the compiler implementation.") to the expected naming convention of the C++ language standard.
Would you like to adjust your selection for unique names?

mhaberler, 2015-04-19:
If you think that "Each name that contains a double underscore __ or begins with an underscore followed by an uppercase letter is reserved to the implementation for any use." applies, I suggest you file a bug at https://code.google.com/p/nanopb/issues/list which is the underlying package

elfring, 2015-04-19:
Are any components also [affected](http://stackoverflow.com/questions/228783/what-are-the-rules-about-using-an-underscore-in-a-c-identifier#answer-228797 "What are the rules about using an underscore in a C++ identifier?") from your own software?

mhaberler, 2015-04-19:
sorry, I dont understand what the error/risk scenario could be, can you explain?

As I said: this comes from an upstream package, and if you want to amend this, the best place is upstream, i.e. nanopb - there is no point in doing a patch here just to create merge issues on next upgrade

I therefore suggest you consult the nanopb mailing list

elfring, 2015-04-19:
It seems that there are [more update candidates](https://github.com/machinekit/machinekit/blob/cb40ab0da23c59f580d0412534f54d2c0f372bc9/src/machinetalk/include/inihelp.hh#L4 "Another update candidate") besides a library for the handling of protocol buffers, aren't there?

How do you think about to avoid that this software [depends on undefined behaviour](https://www.securecoding.cert.org/confluence/display/c/CC.+Undefined+Behavior#CC.UndefinedBehavior-ub_106 "Avoid undefined behaviour?")?

mhaberler, 2015-04-19:
a PR would be most welcome!

elfring, 2015-04-19:
How do you distinguish own software components from bundled libraries?

mhaberler, 2015-04-19:
I'm a bit unsure about the usefulness of this thread

if you find a bug, please file a bug
if you have a fix, please file a PR

closed 2015-04-19 by mhaberler