machinekit/machinekit #573: Using 'debian/rules build' gives an error not present in normal build

ArcEye, 2015-04-19:
Trying to build a package to test package install rebranding, but the failure in the boost
code does not occur in a normal source build.

Any ideas?

```
Compiling emc/usr_intf/emcsh.cc
Compiling emc/usr_intf/shcom.cc
Linking linuxcnc.so
Compiling emc/usr_intf/emcrsh.cc
Linking linuxcncrsh
Compiling emc/usr_intf/schedrmt.cc
Compiling emc/usr_intf/emcsched.cc
Linking schedrmt
Compiling emc/usr_intf/emclcd.cc
Compiling emc/usr_intf/sockets.c
Linking linuxcnclcd
Compiling emc/usr_intf/halui.cc
Linking halui
Compiling emc/usr_intf/keystick.cc
Linking keystick
Compiling emc/usr_intf/xemc.cc
Linking xlinuxcnc
Compiling emc/usr_intf/emccontroller.cc
In file included from /usr/include/boost/thread/pthread/mutex.hpp:14:0,
                 from /usr/include/boost/thread/mutex.hpp:16,
                 from /usr/include/boost/thread/pthread/thread_data.hpp:12,
                 from /usr/include/boost/thread/thread.hpp:17,
                 from /usr/include/boost/thread.hpp:13,
                 from emc/usr_intf/emccontroller.hh:27,
                 from emc/usr_intf/emccontroller.cc:34:
/usr/include/boost/thread/xtime.hpp:23:5: error: expected identifier before numeric constant
/usr/include/boost/thread/xtime.hpp:23:5: error: expected ‘}’ before numeric constant
/usr/include/boost/thread/xtime.hpp:23:5: error: expected unqualified-id before numeric constant
/usr/include/boost/thread/xtime.hpp:46:14: error: expected type-specifier before ‘system_time’
In file included from /usr/include/boost/thread/pthread/mutex.hpp:14:0,
                 from /usr/include/boost/thread/mutex.hpp:16,
                 from /usr/include/boost/thread/pthread/thread_data.hpp:12,
                 from /usr/include/boost/thread/thread.hpp:17,
                 from /usr/include/boost/thread.hpp:13,
                 from emc/usr_intf/emccontroller.hh:27,
                 from emc/usr_intf/emccontroller.cc:34:
/usr/include/boost/thread/xtime.hpp: In function ‘int xtime_get(xtime*, int)’:
/usr/include/boost/thread/xtime.hpp:73:40: error: ‘get_system_time’ was not declared in this scope
/usr/include/boost/thread/xtime.hpp:73:40: note: suggested alternative:
In file included from /usr/include/boost/thread/locks.hpp:12:0,
                 from /usr/include/boost/thread/pthread/mutex.hpp:12,
                 from /usr/include/boost/thread/mutex.hpp:16,
                 from /usr/include/boost/thread/pthread/thread_data.hpp:12,
                 from /usr/include/boost/thread/thread.hpp:17,
                 from /usr/include/boost/thread.hpp:13,
                 from emc/usr_intf/emccontroller.hh:27,
                 from emc/usr_intf/emccontroller.cc:34:
/usr/include/boost/thread/thread_time.hpp:19:24: note:   ‘boost::get_system_time’
In file included from /usr/include/boost/thread/pthread/mutex.hpp:14:0,
                 from /usr/include/boost/thread/mutex.hpp:16,
                 from /usr/include/boost/thread/pthread/thread_data.hpp:12,
                 from /usr/include/boost/thread/thread.hpp:17,
                 from /usr/include/boost/thread.hpp:13,
                 from emc/usr_intf/emccontroller.hh:27,
                 from emc/usr_intf/emccontroller.cc:34:
/usr/include/boost/thread/xtime.hpp: At global scope:
/usr/include/boost/thread/xtime.hpp:88:1: error: expected declaration before ‘}’ token
make[1]: *** [objects/emc/usr_intf/emccontroller.o] Error 1
make[1]: Leaving directory `/usr/src/machinekit-orig/src'
make: *** [build-stamp] Error 2
```

mhaberler, 2015-04-19:
this looks like its tied into --enable-emcweb and your boost version, where are you at?

ArcEye, 2015-04-19:
Disabled  --enable-emcweb, which I never use normally and it builds OK.

libboost is ver 1.49 (standard version for wheezy AFAIK)

mhaberler, 2015-04-20:
I cannot reproduce on wheezy

look at the error line, xtime.hpp
enum xtime_clock_types
{
    TIME_UTC=1  //<------ "error: expected identifier before numeric constant"

could it be TIME_UTC is #defined as some other symbol?

the way to nail such issues is to do a gcc -E (save preprocessed output) and look at that

ArcEye, 2015-04-20:
I suspect it is connected with how I set up the time zone on this machine.

I did not link to UTC, because it causes all sorts of problems when we go in and out of summer time and I was sick of my system clock always being an hour out

I will look at it again on another occassion, it was just a testing niggle really.

Thanks

closed 2015-04-20 by ArcEye