machinekit/machinekit #581: HAL Python API using operators for netting

strahlex, 2015-04-24:
I have played around with the HAL Python API in the last weeks to implement my Halanduino configuration. So far I like the Python API as I am finally able to create functions and classes. There is just one thing I would ilke to address: netting or connecting pins to signals

Take a look at this peace of code for example. It is not very visual in my opionion I can not tell which directions the pins have just by looking at the code.

    sigFeedback = name + '-feedback'
    hal.net(sigFeedback, pid + '.feedback')
    hal.net(sigFeedback, 'kalman.angle')

So I have come up with the idea to use Pythons operators to address this issue:

    testSig = hal.Signal('test2')
    outPin = hal.Pin('and2.test.out')
    testSig <<= outPin  # link out pin to signal
    testSig >>= hal.Pin('and2.test.in0')  # link in pin to signal
    testSig += hal.Pin('and2.test.in1')  # link any pin to signal
    testSig -= outPin  # unlink pin from signal

What do you think about this idea? Does it make sense?

strahlex, 2015-04-24:
Maybe also adding multiple pins at a time to a signal might be nice to have:

```python
signal >>= inPin1 + inPin2 + inPin3  # links signal to all in pins
signal += outPin + inPin1 + inPin2  # links signal to all pin
signal -= inPin1 + inPin3  # unlinks all pins from signal

strahlex, 2015-04-24:
Quickly tested it > automatically connecting pins is definitely possible and a 10 liner in Python. The only downside I see is that component and signal names need to be generated. Easiest way would be to use a continous number or alphanumeric sequence. This makes following possible:
```python
testSig <<= outPin & outPin2  # outPin and outPin2 are automatically connecto to and2 component and the out pin of this component is connected to testSig
```

luminize, 2015-04-24:
I also think the “netting” in HAL configuration is non-intuitive.
I don’t know how I would do it, but I think that it’s more valuable to get a clear way of netting pins and signals.
My way would be something generic like
```
hal.net(source, target)
```
then you might be able to use that to do
```
hal.net(button.out0, signal1)
hal.net(signal1, light0.in0)
hal.net(signal1, light1.in0)
```
and you could argue also for
```
hal.net(button.out0, signal1)
hal.net(signal1, light0.in0, light1.in0)
```

that would make it understandable for me.

mhaberler, 2015-04-24:
in a sense the latter exists - on the name string level: https://github.com/mhaberler/machinekit/blob/master/nosetests/test_netcmd.py#L117:

hal.net("noexiste", "c1.s32out")

I do get the idea behind the hal.net(button.out0, signal1) example, but this is the issue:

HAL pins have 'object structured' names (like "button.out0"), BUT that is just a nameing convention and at the library level those are just strings; the HAL library per se has no notion of class, instance etc of a pin. So it would be hard to represent a structured name as a Python object with nested attributes.

That's why I went for string option

mhaberler, 2015-04-24:
I think the Python operators idea is great!

I would hope we arrive at an intuitive set of operators and a clear semantics, and that I dont see just yet

for instance, I simply dont understand what this means:

````
 # outPin and outPin2 are automatically connecto to and2 component and 
 # the out pin of this component is connected to testSig
 testSig <<= outPin & outPin2
```
can you unroll this into halcmd steps for the rest of us ;-?

luminize, 2015-04-24:
Ok, i understand. the dot messes things up. Would `hal.net(“button.out0”, “signal1”)` work?

strahlex, 2015-04-24:
@luminize No, net works the other way around. `hal.net("signal1", "button.out0")` would work, but as mentioned in my first post this becomes unreadable pretty fast. You simply can tell which direction a pin has and what it does to the signal just by looking at it. Furthermore when you want to set a signal you also need a `hal.Signal("signal").set(True)` command. So using the Python Signal objects is easier than using strings.

strahlex, 2015-04-24:
@mhaberler This is the code to what I posted earlier:

```python
class MyPin(hal.Pin):
    def __init__(self, *args, **kwargs):
        super(MyPin, self).__init__(*args, **kwargs)

    def __and__(self, pin):
        compName = 'and2.' + createName()
        rt.newinst('i_and2', compName)
        sigOne = hal.new_sig(self.name + '-sig', self.type)
        sigOne.link(self)
        sigOne.link(hal.Pin(compName + '.in0'))
        sigTwo = hal.new_sig(pin.name + '-sig', pin.type)
        sigTwo.link(pin)
        sigTwo.link(hal.Pin(compName + '.in1'))
        hal.addf(compName, 'base-thread')
        return hal.Pin(compName + '.out')
```

mhaberler, 2015-04-24:
assuming an RT thread in a library function implies the pins are from an RT comp

I dont think that kind of mixup is a good idea - simplifying linking: fine, throwing in threads/addf: I suggest not

luminize, 2015-04-24:
well, I just commented on readability and the non-obviousness of the “net signal source1 target0 target1 target2” mantra which makes things hard to read. Guess we have different views on readability.
If you could make a clear distinction between connect an output to a signal, and in another line link targets to the signal that would make it understandable in my view.

other than above I don’t have a real background to give a valued opinion

strahlex, 2015-04-24:
Then summarizing my linking proposal:
```python
hal.net("signal", "pin")  # works, similar to halcmd net
signal = hal.new_sig("signal", hal.HAL_BIT)
inPin1 = hal.Pin("and2.test.in0")
inPin2 = hal.Pin("and2.test.in1")
outPin = hal.Pin("and2.test.out")
signal <<= outPin  # driving a singal by a pin
signal >>= [inPin1, inPin2]  # linking to multiple pins 
signal += [outPin, inPin1, inPin2]  # mixed signal linking
signal -= inPin1  # unlinking a pin

inPin1.set(True)  # works, similar to setp in halcmd
```

So the idea behind this is making the use of the Python language to simplify linking.

luminize, 2015-04-24:
I can follow what’s going on, so that makes it fairly monkey proof :)
can in a later stadium an `inPin3` still be netted to `signal`?

strahlex, 2015-04-24:
@luminize Yes, could be done like so for example:
```python
hal.Signal("signal") >>= inPin3
```

RunningLight, 2015-04-24:
On 04/24/2015 07:02 AM, Alexander Rössler wrote:
> Then summarizing my linking proposal:
>
> hal.net("signal","pin")# works, similar to halcmd net
> signal=  hal.new_sig("signal", hal.HAL_BIT)
> inPin1=  hal.Pin("and2.test.in0")
> inPin2= [...]

I did something similar to this when I wrote my hal2html program years 
ago. No need to resurrect my code. It's out of date and you guys are 
better Python programmers anyway, but my experiments suggested to me 
this is a reasonable approach. Party On, Duds.

mhaberler, 2015-04-24:
@strahlex 

If we can fix the operator irregularity here (why 2 different operators?)

```
signal <<= outPin  # driving a singal by a pin
signal >>= [inPin1, inPin2]  # linking to multiple pins 
`````
I'd be all for it, especially _either_ using names _or_ object handles 

we _should_ be able to write operators with polymorphic arguments in Cython too in case we'd want to

strahlex, 2015-04-24:
@mhaberler 
This irregularity has a purpose `<<=` throws an error if someone tries to link a Pin with direction `in` on the other hand `>>=` throws an error when linking to a `out` Pin. So the operators itself represent the linking direction. The generic operand is `+=` linking any Pin. Or should we use `**=` instead of `+=` representing a "wildcard" in the direction?

mhaberler, 2015-04-24:
well, the  `<=` and `=>`  operators in halcmd are strictly syntactic sugar and actually removed during parsing, semantics is applied in do_net_cmd() and hal_link() respectively and that checks the pin compatibility - that's where error messages com from:

this is the foundation function, it works a pin at a time (hal/lib/hal_signal.c):
 
`````
int hal_link(const char *pin_name, const char *sig_name)
````
the halcmd code drapes `int do_net_cmd(char *signal, char *pins[])` ontop of this (hal/utils/halcmd_commands.c)

so the code doesnt distinguish as it doesnt see any operator, which is why I think it does not make sense to distinguish at the syntax level either

as far as I'm concerned `signal += pin ` and `signal  += <iterable which yields pins>` is sufficient and least confusing

RunningLight, 2015-04-24:
Indeed, those damned arrows cause no end of confusion for newbies who 
think they have significance and sometimes mislead veterans who are 
scanning other people's files. I agree with Michael's "as far as I'm 
concerned" statement.

closed 2015-04-26 by strahlex