machinekit/machinekit #585: missing build dependency

mhaberler, 2015-04-25:
it seems we have a build dependency missing, see this thread: https://groups.google.com/forum/#!topic/machinekit/pMSccTKovpY

it would point to a missing dep between rtapi_global.h and rtapi_app.cc in the minimum, I have observed similar issues with components as well

strahlex, 2015-05-14:
Is the issue still present?

mhaberler, 2015-05-14:
yes, I think so

zultron, 2015-05-14:
If we could get someone to make a tarball of the built tree at the point this error crops up, I could try to solve it.  Otherwise, I'm hoping we can start spinning out portions of the code base (e.g. UIs; Tcl junk, if possible) into separate repos and then refactor the build system to use `automake` or `cmake`.

mhaberler, 2015-07-13:
suspicion from working on the HAL library: the RT comps are missing a dep on src/hal/lib/*.h

mhaberler, 2015-09-02:
well simply no .d dependency files are generated for userland threads RT comps, and at least since Dec 2014 (3f5d79dd3b7e9ffc864be474b377543f8e7117a7 does not either)

mhaberler, 2015-09-02:
reading src/Makefile and results of a build, I think it is safe to say anything referring to DEPDIR in the Makefile is baloney - that directory is never created and hence not populated with any files.

one can `-include  $(READ_RTDEPS)` , but  if these files do not exist - no error.

mhaberler, 2015-09-02:
I think I have a workaround: https://github.com/mhaberler/machinekit/commit/074584ddffbc1c062b91797bc9d9f9f350de14dc

opinions or better solutions?

man, do I hope we can put this build system to rest soonish

zultron, 2015-09-03:
@mhaberler, your patch makes sense to me, short of testing.

mhaberler, 2015-09-03:
well it certainly does not hurt, so I'll do a PR

PR #746 (2015-09-03): build: make RT objects depend on their .d (depends) file
fixes #585
closed 2015-09-03 by zultron