machinekit/machinekit #589: HAL driver for CAN

luminize, 2015-04-25:
As a subissue of https://github.com/machinekit/machinekit/issues/586 I'll try to get history, questions, contemplations, answers, tips and all what passes collected into this issue.

During the process I want to write down "the stuff" so when we release it into the wild we have the docs also ready to go. I'll do that in a dedicated branch can-in-machinekit here: https://github.com/luminize/machinekit-docs/tree/can-in-machinekit

mhaberler, 2015-04-27:
yes, let's make this thread the focus of 'all things CAN/CANopen and machinekit', possibly forking off sub-issues as needed.

Let's start with taking stock of resources:

Linux basic CAN I/O support:
- any modern kernel has support for [SocketCan](http://en.wikipedia.org/wiki/SocketCAN) which would be appropriate for non-RT/posix flavor ( [some useful hints](http://elinux.org/CAN_Bus))
- [must read on SocketCAN API] (https://www.kernel.org/doc/Documentation/networking/can.txt)
- basic [intro to SocketCAN](www.can-cia.org/fileadmin/cia/files/icc/13/kleine-budde.pdf)
- Xenomai has an [RT-CAN stack](https://xenomai.org/documentation/trunk/html/api/group__rtcan.html), supporting several flavors of hardware
- the current -r72 Xenomai kernel from Robert Nelson's beaglebone debian package stream supports the AM335x D_CAN peripheral for RT-CAN out of the box

Linux CAN - software tools:
- the [can-utils package] (https://github.com/linux-can/can-utils) has commandline tools for SocketCAN configuration and use. NB: in jessie, wheezy needs building from source.
- the [python-can package] (https://python-can.readthedocs.org/en/latest/) has Python bindings for basic CAN message I/O

basic SocketCAN programming examples:

- [here]  (https://lnguin.wordpress.com/tag/socketcan-example/) or [here] (http://socket-can.996257.n3.nabble.com/Netdev-example-tp273p274.html)

CANopen-related tools ( the stack ontop of raw CAN messages):
- [CanFestival](http://www.canfestival.org/), suitable for master+slave, vanilla + RT linux kernels, some embedded support
- [canopen-node] (https://sourceforge.net/projects/canopennode/?source=navbar) Janez' embedded CANopen stack
- [libcanopen](https://github.com/rscada/libcanopen) - status unknown

mhaberler, 2015-04-27:
In terms of results, I would see the following (number=priority)

- (1) a CANOpen master HAL module, supporting several devices with different profiles
- (1) a simple CANopen slave module for linux, must be able to connect to the above (eg over a virtual can interface). Might export HAL pins too. Primary use case: intra-HAL regression testing. 
- (2) a CANOpen slave HAL module (maybe the same as the previous item). This would enable wrapping machinekit as a CANopen-controlled device (servo drive, pin I/O etc)
- (2) an embedded CANopen client on a popular platform like Arduino, as an example to start from.

Here's a [note on how a master might be structured in terms of the HAL API] (https://github.com/mhaberler/asciidoc-sandbox/wiki/Using-HAL-instantiation-for-HAL-device-drivers)

mhaberler, 2015-04-27:
Testing fieldbus components: here's a sketch how this could be done: https://github.com/mhaberler/asciidoc-sandbox/wiki/A-testing-strategy-for-fieldbus-HAL-components

jpmoniz, 2015-04-27:
This might look like it may lessen the workload for an arduino slave.

http://www.google.ca/url?q=https://github.com/jgeisler0303/CANFestivino&sa=U&ei=R7c-VcT6A8qcygTln4HoBQ&ved=0CCEQFjAE&usg=AFQjCNF29W9CgE8bEQZQUWbUQxZztAEDlQ

http://www.google.ca/url?q=https://github.com/jgeisler0303/AGCON&sa=U&ei=R7c-VcT6A8qcygTln4HoBQ&ved=0CCQQFjAF&usg=AFQjCNEAEp5uChbBRWEbvhkjHnB6J8sKjQ

mhaberler, 2015-04-28:
maybe we should invite https://github.com/jgeisler0303 over? I see mutual benefits

pJanez, 2015-04-29:
Hello from CANopenNode.

mhaberler, 2015-04-29:
oh hi! welcome to the caravan!

pJanez, 2015-05-02:
May I suggest Arduino Due. It has SAM3X8E ARM Cortex-M3 CPU. It is much stronger than 8-bit AVR on arduino uno. And it already has CAN. No problem to connect a motor, do some path calculation and run CANopen smoothly. And it seems, somebody wrote a driver for that processor for CANopenNode already.
I didn't investigate any further, but I think, there is no sense to use Arduino uno. I think, It is not strong enough to drive a decent motor.

mhaberler, 2015-05-02:
you're right, at these prices it doesnt make much sense to use a uno, and interfacing will actually be easier since no mcp2515 is needed, just a mcp2551 or equivalent

I'll see what can be had off fleabay ;)

mhaberler, 2015-05-03:
setup tips:

For development I use virtualbox (under OSX, but shouldnt matter)
for playing with socketCAN it's useful to have at least a virtual can interface (vcanX) 
here's a gist for vcan setup: https://gist.github.com/mhaberler/a9af7ef9f885113795cc

in case you have an [USBtin](http://www.fischl.de/usbtin/) it is also easy to use from a VM:

on OSX, I have a USB device filter 'USBtin' with Vendor 04d8 and Product 0000a so the USBtin is attached to the VM as soon as I plug it in, it comes up as /dev/ttyACM0

here's a gist how I configured wheezy so USBtin at /dev/ttyACM0 comes up as can0 at 1Mbit:
https://gist.github.com/mhaberler/eef30dd73aa7b326b8b3

for using wireshark, dont forget to add your userid to the netdev group - otherwise permission problems

luminize, 2015-05-03:
One of my goals is to keep the docs up to date with my efforts.
For those interested in setting up development for socketcan, wireshark and the stuff @mhaberler mentions above:
https://github.com/luminize/machinekit-docs/blob/can-in-machinekit/machinekit-documentation/developing/CAN-developing.asciidoc
and
https://github.com/luminize/machinekit-docs/blob/can-in-machinekit/machinekit-documentation/setting-up/CAN-developing-setup.asciidoc

documents are work in progress

amitgoradia, 2015-05-04:
This is also a nice resource for a canopen ds402 implementation.
https://github.com/golems/socanmatic/

They have an EDS parser written in python: https://github.com/golems/socanmatic/blob/master/canmatc

regards,
-amit

luminize, 2015-05-07:
I'm setting up my USBtins with socketcan.
http://www.fischl.de/usbtin/linux_can_socketcan/

Where there's also mention of can for Rpi
http://elinux.org/RPi_CANBus

mhaberler, 2015-05-07:
this is essentially SPI to SPI-Canbus-Controller-to-Canbus which every board/SoC with an SPI interface can do, pretty much like the AVR Canbus shield

so it's the 'ok, CAN not natively available, let's fallback onto SPI' solution which would work with just about every board, at the cost/inflexibility introduced by the CAN controller chip (typically this one: http://www.microchip.com/wwwproducts/devices.aspx?dDocName=en010406)

pJanez, 2015-05-08:
Is there any other SPI CAN controller with more than six filters? Comfortable CANopen device needs more than six input filters. If CAN module does not have enough filters, then microcontroller must receive all traffic via SPI. It may be quite a heavy load for 8-bit device.

A long ago I was using 8-bit PIC18F458. I tested it at 1Mbps with full CAN bus load and it worked fine. But microcontroller has integrated CAN and there is some inline assembly in receive interrupt.
See http://sourceforge.net/projects/canopennode/files/canopennode/ , version 0.9, CANopDriver.c

mhaberler, 2015-05-08:
which side are you talking about?

For the HAL/mk master side we can assume there is a given controller, either on-chip like with the Sitara's, or as a card, like a PCI card for a PC, so I dont see much flexibility here - I would prefer to stay out of custom hardware as much as possible, and too early - that makes adoption so much more difficult

for the client side - where would the need for larger filter sets come from ? I thought a client just filters on its node id and that is it?

pJanez, 2015-05-08:
Master side should not be a problem. It's fast enough.

CANopen CAN message ID's:

```
typedef enum{
     CO_CAN_ID_NMT_SERVICE       = 0x000,   /**< 0x000, Network management */
     CO_CAN_ID_SYNC              = 0x080,   /**< 0x080, Synchronous message */
     CO_CAN_ID_EMERGENCY         = 0x080,   /**< 0x080, Emergency messages (+nodeID) */
     CO_CAN_ID_TIME_STAMP        = 0x100,   /**< 0x100, Time stamp message */
     CO_CAN_ID_TPDO_1            = 0x180,   /**< 0x180, Default TPDO1 (+nodeID) */
     CO_CAN_ID_RPDO_1            = 0x200,   /**< 0x200, Default RPDO1 (+nodeID) */
     CO_CAN_ID_TPDO_2            = 0x280,   /**< 0x280, Default TPDO2 (+nodeID) */
     CO_CAN_ID_RPDO_2            = 0x300,   /**< 0x300, Default RPDO2 (+nodeID) */
     CO_CAN_ID_TPDO_3            = 0x380,   /**< 0x380, Default TPDO3 (+nodeID) */
     CO_CAN_ID_RPDO_3            = 0x400,   /**< 0x400, Default RPDO3 (+nodeID) */
     CO_CAN_ID_TPDO_4            = 0x480,   /**< 0x480, Default TPDO4 (+nodeID) */
     CO_CAN_ID_RPDO_4            = 0x500,   /**< 0x500, Default RPDO5 (+nodeID) */
     CO_CAN_ID_TSDO              = 0x580,   /**< 0x580, SDO response from server (+nodeID) */
     CO_CAN_ID_RSDO              = 0x600,   /**< 0x600, SDO request from client (+nodeID) */
     CO_CAN_ID_HEARTBEAT         = 0x700    /**< 0x700, Heartbeat message */
}CO_Default_CAN_ID_t;
```

Average CANopen client should accept:
NMT, SYNC, 4xRPDO, 1xRSDO, 4xHeartbeat (for monitoring)

6 receive messages should go too:
NMT, SYNC, 2xRPDO, 1xRSDO, 1xHeartbeat (in case, master has problems, client should shut-down actuators)

luminize, 2015-05-09:
The more I read about CANopen, the more I'm realising that this fieldbus bus is highly configurable. Master-slave, multi-master, syncing, hearbeats, event change driven... It's going to be all about the messaging.

The TPDO and RPDO's are in my laymen terms the registers that will determine for a node which message to send, and for a node to which message on the bus to listen to.

Nodes can listen to the master, or to other nodes. Extrapolating: it's a subscriber model. having one foot in userland, the other foot in realtime or userland

The way I see it, is that Machinekit is not limited to a realtime timing. So messages can come from nodes at undetermined times. Polling will give a lot of overhead and is discouraged.

Reading the data from the bus: the HAL driver should always listen to the bus, and put the messages from the bus in a ring buffer. Components can than act on the messages
Writing to the bus could be done in 2 ways:
* There should be a second ringbuffer acting as master/slave that can put messages gathered from hal-canopen-components onto the physical bus. This will make **one** "HAL CANopen node" which has to act as would any other node with respect to collisions of data etc (master/slave does not matter yet).
* Each hal-canopen-component should be a full fledged node, message triggered by a thread. It should have access to the bus, acting in the way as would any other CANopen node wrt collision and conformity.

I think the 2nd option would be quite "heavy" and since every hal-canopen-component becomes a node, this will go at the cost of the available nodes on the bus.

reading on, so opinion might change.
Thoughts?

mhaberler, 2015-05-10:
those communication patterns are certainly possible, but maybe on average just very simple patterns are used

the right abstraction will likely depend on the use case, so maybe two hypothetic devices at both ends of the spectrum are a good starting point - like a dumb I/O expander to wiggle/read pins (not time critical), and at the other end some super-duper motion subsystem which can move several motors in a coordinated fashion, maybe even segments (likely time critical)

the primary object to deal with IMO is the device dictionary, which is a collection of variables with name, id, type, and purpose (_very_ similar to pins). The messaging is mostly subordinate to that in the sense that it helps establish a comms path, and more or less peek and poke values from/to this dictionary on a per-device basis.

now a particular HAL component/application might use just a small fraction of the dictionary, so an obvious task of a comp startup would be to select dict entries to talk to - which would wind up in pins, and the messages going back and forth serve to synchronize the device values and HAL pins

that simple view probably is enough for the dumb I/O expander case and then some (NB Fernandos canopen.comp use this model).

in the intelligent drive use case you'd likely have more work than just peek/poke type operations, like downloading segments, synchronizing their execution, and monitoring their execution status. Even there, I'm unsure a generalized messaging API to the CANbus will be helpful outside the handling component. What might be needed is a segment-level API between motion and the CANopen component for segments - no way to do this through pins

A ringbuffer or queue for that matter helps storing values until receiving components can act upon them, in case they cannot do so right away; but at least in the dumb I/O expander model, and likely the drive use case the HAL-side component could very well just do synchronous I/O

so overall my idea would be: keep CAN comms internal to the component for now, and strive for doing the dumb I/O expander case first (NB this can used with a motor drive - where commanded position values are poked to, and actual position values are peeked from)

once we get this right, we can look into such a segment-level API.

before that, I guess we need to get session management (NMT), heartbeating, comms errors (emergency?) and the like right; looks to me this layer will be pretty much the same for all use cases

I wonder what @pJanez @amitgoradia and Uwe's take is on this?

luminize, 2015-05-10:
My first component will be a component, attached to the CAN bus, and when i send a can message I want to see a HAL pin wiggle.

Regardless of the way to implement things we should ponder whether a HAL-CANopen-slave or HAL-CANopen-master are on the "normal" RT servo-thread. Meaning
* that either all HAL components listen directly on the bus and thus are not depending on the servo thread
* or the ring buffer listens independent on servo thread, and is the Schnittstelle between RT and non-RT (I see the CAN bus itself as non-RT, might be wrong :) )

all other stuff will come later, but the nature of the bus itself is in a very important thing to understand for me.

githubuser0xFFFF, 2015-05-10:
Before I give an advice, I would like to ask a question. If I use a BeagleBone Black with PRU for control of 3 stepper motors for a coordinated movement plus 2 output ports to control of 2 heaters  - how does the machinekit HAL talk to the 3 stepper motors and how does it talk to the I/O ports?

If I replace the BeagleBone Black with 3 CANopen DS402 Motion Controllers and 1 CANopen DS401 I/O device for the 2 outputs, does something change for the machinekit HAL? I do not have in depth knowledge of the machinekit HAL yet, to have an idea, what is different now. I would assume nothing. CANopen creates a local image of the remote CANopen process data and you talk to the CANopen drives, as if they are locally attached to your control unit, that runs machinekit. Normally it is not necessary to care about any CANopen communication details, if you have a proper abstraction.

In our CANopen software we have for example a CDS402DriveDevice class. This class has an API to execute movements and read drive status, position... If you use this class, you do not need to care about any CANopen communication details, because it all runs behind the scenes.

So at the moment, I'm not sure, what is the difference between a locally attached stepper drive and a remotely attached CANopen device?

mhaberler, 2015-05-10:
Uwe - the way things stand now the interface between motion and hardware of any shape or form is like so (completely platform-independent)

- joint commanded position values are generated each servo cycle, and output in the axis.N.motor-pos-cmd pins
- the actual positions are read back via the axis.N.motor-pos-fb pins
- the following error of each joint is reflected in axis.N.f-error

now what is connected to this depends on the hardware, but usually it's a stepgen which pretty much works like a servo loop (it gets a commanded position and yields an actual position), or it is a PID loop

you will find examples for either under configs/ - in the 'simulator' configs pid loops are used and the 'motors' are simulated by ddt components

now for the difference between pid/stepgen and CANopen attachment:
- in Fernando's canopen.comp - CANopen over RTnet: none. He just pumps out cmd-pos to the CANopen device each servo cycle.
- in Bas' future component: that remains to be seen - if we can retain the current 'one commanded position value per cycle' OR we need to turn down the servo rate to not overwhelm the bus or device (NOT good) OR we find some other API (like segment-level into the tp) to reduce the data rate

fmederos, 2015-05-10:
Bas,
Writing a component that directly access a CANOpen bus is one very big
task, if that is what you meant. Anyway you should not need to do so!

I'm not sure about canopennode, but CANfestival does all the bus
arbitration so a CANOpen HAL component just needs to pass data to and from
CANfestival.

CANfestival uses its own real-time for bus timings, this runs independent
from LCNC/MKIT's real-time. Sure the routines accessing the bus uses their
own buffers as you say.

Maybe canopennode provides this functionality or more, I did not look at it
long enough...

amitgoradia, 2015-05-11:
Michael,
First we need to analyse the bus bandwidth available in order to decide on a strategy for dumping motor position every millisecond or adopt a segments queue approach. We also need to identify the maximum drives that can be addressed from one CANBus channel to decide the approach.

An interesting test done by someone on CANBus bandwidth : http://electronics.stackexchange.com/questions/121329/whats-the-maximum-can-bus-frame-message-rate-at-125-kbit-s
It is mentioned that they are able to reach the theoretical maximum bus bandwidth!! Which is about 954 frames/sec (8byte frames). But with arbitrary data they can see worst case times of 850 fps. This is done at 125kBps bus speed.

Extrapolating at 1Mbps, we can transmit 6.8 frames/msec. Lowering the payload byte count is not going to affect this theoretical maximum by a great amount (maybe 1.5x higher can be achieved by reducing the payload bytes).

Consider regular mill with 3 axes. Consider Cyclic Servo Positioning (CSP) mode, i.e., the one where we dump millisecond level position points to the drives. The following data needs to be exchanged at the servo rate (every msec):
1. position for axis 1 (controller to drive)
2. encoder for axis 1 (drive to controller)
3. position for axis 2
4. encoder for axis2 
5. position for axis3
6. encoder for axis3
7. Sync pulse for applying the axis position sent by the bus. (as a part of the CANOpen standard).

So, we need 7 frames/msec minimum to keep things going which is theoretically possible (which shorter payload frames as SYNC may not need any payload and position and encoder are 4 byte values). Also, to my knowledge CANOpen uses 11 but identifier frames as opposed to 29 bit frames so that will also give us a little more bandwidth. We also need to account for heartbeat message and NMT messages which are a part of DS 301.

Add in one more axis or a discrete IO device to the mix and we hit the boundary of what is possible with a single CANBus in CSP mode. Move the non-synchronized devices to another bus and we are back the feasible realm.

CANOpen is a very lightweight and configurable protocol for industrial bus communications. No wonder it forms the basis of COE (CANOpen over Ethercat).

Part 2. Segments API:
This approach requires a fundamental change to the way things are done in LinuxCNC where the Linux PC is considered supreme and needs to do all the work (in some cases generate the high speed pulse train too!! ). As we see the limitation of the bus is for the controller to be able to send points every msec to the drives. In a segments approach, we model the output of the motion planner as a polynomial equation with coefficients.
A linear move can be defined as follows:
startpoint: X0, Y0, Z0, A0, B0, C0, U0, V0 ,W0
endpoint: X1, Y1, Z1, A1, B1, C1, U1, V1, W1
a polynomial: p = at^3 + b*t^2 + c*t + d where p is the position along the path as a function of time. 
t_final- assume the polynomial holds from t=0 to t= t_final.

t varies from 0 to t_final. p varies from 0 to 1 but need not cover the entire range. when p=0 the path is at the startpoint provided, and p=1 path is at endpoint provided in the command.

Position of the axis is calculated using the equations:
p = at^3 + b*t^2 + c*t + d
x = X0*p +X1*(1-p)

Now  by specifying 23 parameters, we can basically cover a large line segment spanning 100's of milliseconds. All we need to now specify is the time tick on a per msec basis which is done by the SYNC command in CANOpen. 
With this approach, each drive will be sent its startpoint (X0), its endpoint (X1), polynomial coefficients (a,b,c,d) and the final time(t_final). Further, the parameters (a,b,c,d,t_final) are common to all the drives and can be sent out as a broadcast to the Axes. Then sending a SYNC command will just cycle the trajectory.

NOTE: I still have to readup on the segment level API offered by CANOpen which should be a part of DS402.

My initial thoughts: First try CSP with limited number of axes. This will avoid fiddling with the trajectory planner and allow us to test the CANBus implementation first.

-amit

mhaberler, 2015-05-11:
Amit - that's exactly what I was thinking when Andreas came up with the idea first time around: we might swamp the bus. That's why Claudio uses several CAN interfaces in his zedboard setup.

There is one fundamental question which I dont have an answer to, assuming a segment-level API; maybe you folks with more experience with fieldbus drives can answer this - it is: where is the kinematics handled, and how does that impact what is being sent to the drive?

With servo-rate quantisation host-side, we're in infinitesimal territory - it is safe to assume the 1mS increments can be approached through a line. Plus the drive being a low-pass: we get a curve.

BUT: if we lower the rate, this assumption might not hold anymore in connection with a non-trivial kinematics:

let us assume a non-trivial kins, like a scissor kinematics, and the interpreter sending a cartesian straight move. This will translate into a fairly arbitrary cuve per joint over time, and the way things are done right now this curve is just handled by sampling into very short lines.

Now sending such a curve to the drives - I see two approaches:

- how it is currently done - sample at high rate, meaning bus load problem
- something you hinted at: transforming this curve into a polynomial equation and sending the parameters  to the drive for execution.

My question: how is this usually done? what is the flow? does one go down all the way to samples, then generate the polynomial params from that sample set? does one push the issue uplevel and only send polynomial equations from the interpreter downwards? if so - the parameters would not survive a nontrivial kinematics unchanged.

the drives obviously would operate in joint space (short of a DeltaTau I've seen none which can do kins on-drive).

Or is it just that industrial applications of such drives are all trivkins anyway and I'm pushing a non-issue?

jpmoniz, 2015-05-11:
@mhaberler  @amitgoradia 

Guys,

Keep in mind that just because a feature may be apart of DS402  does not mean OEM's implement them.  Some choose to implement basic features.  If i look at the drive I have. They have only implemented basic torque and velocity .  Others choose to implement things like interpolated position mode.  It may be worthwhile surveying the most popular DS402 operation modes supported. Then decide on what work may be required.

pJanez, 2015-05-11:
There is no need to handle CAN messages directly. Application should use variables from object dictionary (controlWord, statusWord, TargetPosition, PositionPoints, ActualPosition, etc.). It is then configurable by CANopen, what are the rules for exchanging the values. CANopenNode and CANfestival (as explained @fmenderos) use this approach.

I think, CANopen drives must be autonomous and they only accept points and controlWord from master(trajectory planner, Machinekit.). Drive must read its encoder and handle PID loop or count steps on its own. So there is no need for real-time encoder information on the bus. Drive sends back only status word (one bit gives information, if following error is inside window or not). It usually sends actual position, but it does not need to be real-time, it can be sent after each tenth sync message only. It is only for information.
That way there is no problem to drive three or more axis complex joint in one millisecond interval. CAN FD will later enable even more data to transfer.
Another way is to use polynomial, which could further reduce traffic. In trapezoidal movement, there are three points necessary to get exact movement, in jerk-free trapezoidal, there are seven points. In complex joint system there could be some smart function, which splits trajectory into segments of proper length, I guess.

I think, Machinekit should be first capable to use first approach (send points each millisecond), because it is simpler and some commercial drives use this approach (like [this](http://en.nanotec.com/products/226-pd2-o41-plug-drive-stepper-motor-with-integrated-controller-and-rs485canopen-interface/) ).
EPOS 2, on the other side, uses polynomial approach, as I know.
However, it is not standardised in CiA402, how points are used.
We may think polynomial approach as an upgrade in next phase, where we could be capable to implement some CANopen drives too.

The first approach is also compatible with "Cyclic position mode". I think, most users support that mode. Michael, please verify that.

pJanez, 2015-05-11:
If CiA402 drive implements only torque and velocity mode (frequency inverter), then it is probably not suitable for servo positioning.

luminize, 2015-05-11:
to all: thanks for the discussion/answers/questions. For me at this moment this is still a few steps away regarding implementation, but investigating further I need to understand the boundaries for my future blunder prevention.

Last week I got into some discussion with @mhaberler and @alueger and what struck me regarding the DS402 PVT profile was that points on a curve can be defined for different time intervals. We looked at section 8 (page 54) of this document http://www.a-m-c.com/download/sw/dw300_3-0-3/CAN_Manual300_3-0-3.pdf

At the risk of derailing the discussion...
When we get to segments (i know this is still too early :) ) and more difficult applications maybe we need to temporarily forget the fixed time approach, and consider the allowable path deviation instead.
Example:
Making an arc with trivial kinematics, it makes no sense to generate a lot of data for motion for the y-axis motor when we're at the (X=0, Y=R) position. Because there we can let the node of the y-axis do the interpolation. 
When the arc progresses the frequency of points to interpolate of the y-axis will increase, and at the same time for the x-axis will decrease.

I'm only guessing here, but that would mean that we can use less bandwidth as long as we know when _not_ to send a new position at a specified point in time, simply because it is not needed from the point of accuracy. But that could be a future planning thing.

@fmederos @pJanez , first thing for me now is that I will explore the available functionality wrt communication, NMT, messages etc. First thing is to just activate and read and write date to and from a device. Probably not even with machinekit yet.

githubuser0xFFFF, 2015-05-12:
We use CANopen drives from Maxon, ELMO, Trinamic, Nanotec and Kocomotion. The following table shows a small matrix of the supported operation modes that are relevant (Profile Velocity Mode PVM, Velocity Mode VM, Interpolated Position Mode IPM. Cyclic Synchronous Position Mode CSVM):

| Drive      | PVM | VM | IPM | CSVM |
|------------|-----|----|-----|------|
| EPOS       | 1   | 1  | 1   | 0    |
| Trinamic   | 1   | 1  | 0   | 0    |
| Nanotec    | 0   | 1  | 1   | 0    |
| Kocomotion | 1   | 0  | 0   | 0    |
| ELMO       | 1   | 0  | 1   | 1    |

If we use the CANopen drives as dumb drives and simply pump out speeds in a 1ms intervall, than we could use the PVM or VM for setting the Velocity. For this mode a controlword (16 Bit), the velocity (32 bit) and the statusword (16Bit) is required. The controlword + velocity is mapped into an RPDO and the statusword is mapped into a TPDO. Additionally we need an global 8Bit sync message so that the new velocity is executed synchronously by all drives. So we have:

RPDO (controlword + velocity) = 48 Bit Data + 47 Bit Frame = 95 Bit
TPDO (statusword) = 16 Bit Data + 47 Bit Frame = 63 Bit

Sync-Msg: = 8 Bit Data + 47 Bit Frame = 55 Bit

If we running the bus with 1MBit and a update intervall of 1 ms, then we can theoretically control:

(1000 Bit/s - 55 Bit Sync) / 158 Bit per Drive = 5,98 = 5 Drives (ca. 80% bus load)

If the update intervall is increased to 1.5 ms or 2 ms, control of more drives would be possible.

mhaberler, 2015-05-12:
semi-related/testing devices: I found this [CAN Monitor]( http://www.ta.co.at/de/produkte/uvr1611/can-mt.html) in my cellar - it's part of an unused heater control. It is a CANopen device, and I might use it for testing non-drive interaction. The docs are here http://static.mah.priv.at/public/CAN-Protokoll.zip and are in fact pretty good; only downside is it seems fixed at 50Kbit/s.

pJanez, 2015-05-12:
Thanks to Uwe for information about different drives. Situation may be similar with positioning modes. However, smart implementation of interpolated position mode should quite reduce the traffic. So, here we could compete with Ethercat and similar Ethernet protocols.

Interpolated position mode is not standardised in detail. Mode only provides a memory block for interpolation points and not specifies, how points are structured. 
- In simple implementation (Nanotec, Kollmorgen), there is only array of 32-bit points with constant delta-time.
- EPOS has position-velocity pair for each point (I think, it is constant delta-t). This actually means, that points describe spline (third order curve).
- AMC seems quite interesting, I didn't know it before. It is similar (but not same) as EPOS. It's point contains 24-bit position(absolute), 24-bit-velocity, 8-bit time and 8-bit counter(for control). This goes into one PDO message.
- If 24-bit position is not enough, then we could use 32-bit position, 24-bit velocity, 7-bit time and one toggle bit for control.

Machinekit will have to be configurable in this part.

pJanez, 2015-05-12:
For testing and getting taste of CANopen I suggest http://www.port.de/en/products/canopen/tools/canopen-device-monitor.html
It is commercial product but runs in demo mode (limited to one slave) in Linux. Then load eds file from your favourite motor and enjoy exploring. I didn't test it on Linux, but I think, it should run with CAN4linux.

mhaberler, 2015-05-18:
I got delivery of a [nanotec.de SMCI36 stepper/bldc drive with CANopen control](http://de.nanotec.com/produkte/1041-smci36-schritt-und-bldc-motorsteuerung/) for testing, and also a Maxon EPOS2 Positioning controller.

hekav, 2015-05-29:
Hello,

let me in introduce myself. My name is Helmut and I'm from the southern part of Germany. I'm a retired software architect and in my spare time ( I have a lot of that now) I'm building a model train layout in my basement. I also have tree milling machine doing some milling for my model train and others. For the milling I use LinuxCNC and Mach3. Right now I'm building the hardware to control my model train. The hardware is uses mergCBUS. This is a implementation of the CAN protocol to control model train. I works very well and I happy with the decision to change to the mergCBUS hardware and software.

A few weeks ago I was taking to my son about my implementation of mergCBUS. He is working with Festo AG and develops electronic components. He also uses Festo motion controller for his projects using CANOpen protocol. We came up with the idea to use a Festo controller and a Servo motor for my milling machine. 

So we borrowed a controller and a motor from Festo and build up the experimental hardware.

Looking at the LinuxCNC we were  very disappointed, there is no support for CAN. So I was looking for any CANOpen implementation and found Canfestival. The Festo controller (CMMP-A5) has many operating modes, so we decided the easiest mode to use with LinuxCNC is interpolated position mode. 

I modified the Canfestival masterdemo to configure the controller and turn on IPM. I was using a peak-usb interface. The IPM of the Festo controller requires a 8ms sync massage and between the sync messages we have to send the new positions. It turned out the controller kept the IPM only for a few seconds and then went into the fault mode. We measured the jitter of the sync massages coming from the PC. The jitter were about 200 us, way too much for the controller. The controller accepts no more then 20 us jitter.

So I decided implement my Canfestival program as a real time application. Canfestival has support for RTAI and XENOMAI real time system.  Unfortunately the paek-usb interface is not capable for real time. I purchased a paek-pci card witch has a real time driver. I installed RTAI and XENOMAI on different Linux partitions but never was able to compile the Canfestival program in an real time environment.

Looking for a solution for my various  problems I found this board and realized other persons are also looking for a solution for our idea.

So my question is, has anybody a working implementation in a real time environment talking via CAN to a external controller ?  If there is a implementation I can offer to test it on my hardware and also doing some implementation (coding) if necessary.

pJanez, 2015-05-29:
Maybe setting a PC as SYNC producer is not best idea. In general, any device may be set as SYNC producer. Maybe it could be possible to set Festo drive as SYNC producer.

Index 1005, bit 30: 1(0) - node generates (does NOT generate) SYNC object

pJanez, 2015-05-29:
Back to CANopen implementation. I must appologize, I published nothing yet, except few chats here. I have a "bad" habit, I must learn everything first, before I started to program. Linux has quite a lot of functionality.

First my plans:
- Make a command line program canopen_dump, which will decode and nicely display CANopen messages.
- Make a command line program "canopen_od <read or write> <node_Id> <index> <subindex> <data> <datalen>", which will read or write object dictionary variables on CANopen devices.
- Make a clear demo for driving motor from Linux, using command line with above two commands and maybe cansend command from can-utils.

With SDO communication it is possible to command positioning motor to target position. No PDO communication is necessary for the first time. For using interpolated position mode, there will be more work to do.

mhaberler, 2015-05-29:
@hekav: good to hear you're thinking in the same overall direction, we really need to do something about this fieldbus theme

to summarize where we are -

we can use all hardware Xenomai supports on PC-type platforms, and we have a working RT-CAN driver in the current -r72 xenomai kernel for the beaglebone

I did explore this so far as to do raw CAN I/O from a RT thread, and the rtcansend/rtcandump utiliities inlcuded in Xenomai work on both platforms. I have a Peak PCIe card and the BB does include two CAN interfaces on-chip, only bus drivers needed.

We're right now collecting various devices to test against, so your efforts both with CBUS and the Festo drive fits into the general direction

In terms of timing - if external devices rely on jitter well below 20uS I agree with Janez we will have to rethink how sync messages are produced. However that is done will have to interwork with position/velocity sample generation. One option I could think of is to do CAN I/O through a the AM335x's (BB CPU) PRU - the PRU's have excellent timing behavior and that would take care of jitter

upper layers: There is some work by Fernando Mederos which can be recycled, see [above]
(https://github.com/machinekit/machinekit/issues/589#issuecomment-96533283)

mhaberler, 2015-05-29:
related: this sync timing issue touches upon the question "who does the timing"

so far this has been machinekit host-side - there is a periodic thread and that drives everything else

that is simple but does have some disadvantages - jitter comes into play, and there is currently no obvious way to use an external timing source with machinekit

I did some work on [triggered function chains](https://github.com/mhaberler/machinekit/tree/hal-triggers) which I want to evolve towards HAL funct-chain style execution driven by external sources, but that code is not there yet (the driver interface is missing)

There are other setups possible like Peter Wallace DPLL scheme where the hardware essentially locks onto host timing with a PLL but delivers samples in more exact timing and the host side does calculations on these more accurate samples

mhaberler, 2015-05-29:
@pJanez 

- dumping CANopen messages: I suggest to install wireshark - it has a great CANopen dissector. I think it would be a waste of time to replicate what wireshark already does

as for your other examples - give [libcanopen](https://github.com/rscada/libcanopen) a try, it has some of what you suggest

hekav, 2015-05-30:
@pJanez

sorry at least on my Festo controller this bit is not writeable ( I tried it). In the documentation of other controllers for example Beckhoff is says its writeable, but I cannot verify it. It was a good idea.

pJanez, 2015-05-30:
Michael,
CANopen is partly implemented in Wireshark. It is still hard to further decode messages. There is still nothing nice, useful and free in Linux.

mhaberler, 2015-05-30:
if this is the case, then the best use of our time is to improve the CANopen dissector in wireshark.

That is a well-documented process and would expose any features to a huge user base pretty much immediately, opening up any work to improvement by others. This is unlikely to be the case with  a standalone utility, which is just a legacy in the making.

NB I am a violent opponent of reinvention. I still have to find a case where reinvention of a basic sanitary feature makes sense.

pJanez, 2015-05-30:
I agree with you.
However, simple command line utility for decoding CANopen messages is not much reinvention. I will take candump from can-utils as a base and will try to add a little CANopen decode. This could also be run from the BB directly.
Improving the CANopen dissector in wireshark is not a bad idea also. We can do both.
But I just got some Linux C programming knowledge, I want to make a command line first. I won't complicate much.

mhaberler, 2015-05-31:
@hekav - I would be very interested in a manual reference on this jitter limit of 20uS as you mentioned, and if that limit is the true source of the problem. Is there any error code or so to verify it is jitter which is at fault?

this is why: if this is in fact the case, then I think there is no realistic option to control such a drive from even the most reactive Linux RT kernel. All those have latencies in that range or well beyond that. Add to that I/O and bus latency and we're above that limit for any realistic host setup to create sync messages.

Now clearly somebody must have been before (ie operate this drive in IPM from a Linux host). So I ask myself the question: what was the magic trick? I just cannot imagine sync generation was moved to external hardware because that particular drive is a bit touchy.

fmederos, 2015-06-01:
Michael,
I did not do any numeric analisys on this but right now am looking at a
pretty long capture taken with Wireshark monitoring bus activity of
linuxcnc controlling one slave device thru our CANopen/RTnet
implementation. All Sync frames that I see fall inside a 4uS jitter window,
this is wireshark's timestamp which we did not test against any lab
instrument though.

Note that it is RTnet which gets access to the physical Ethernet port here.
RTnet takes away from the equation interface and medium latencies which as
you mention would easily add-up beyond 20uS, it is common practice among
fieldbuses to include tools to compensate for latencies employing
handshaking mechanisms during remote devices registration at bus startup.

As I wrote above, I did not test this properly but maybe 20uS jitter limit
is not unrealistic with Xenomai/linuxcnc/CANfestival.

Wireshark did prove very useful at debugging this bus implementation, we
considered developing a proper CANopen/RTnet dissector but we did not have
the time...

Regards,
Fernando

mhaberler, 2015-06-01:
@fmederos - that is a pretty good timing result for a linux-driven affair ;)

Andreas pointed me to this related and interesting read: [High Precision Drive Synchronisation with CANopen](http://www.can-cia.org/fileadmin/cia/files/icc/8/rostan.pdf) - the way I read it though the DPLL must be on the drive's side to cancel out sync message jitter, and I guess this assumes _all_ drives to the same thing so their DPLL's arrive at the same timing behavior

mhaberler, 2015-06-01:
@pJanez - re SYNC produced be some other device than the host providing position updates:

I would think in that case you have a stream of position updates, and a stream of SYNC messages on the bus, but I do not see how those could be phase-locked - I guess a SYNC can be posted only after the position update is sent

I guess there's potential for some weird results in that case - like a position latched twice or not at all within a cycle

githubuser0xFFFF, 2015-06-01:
I'm not an expert for synchronous operation of drives, but from my theoretical understanding of the CAN bus, I think there will be always a jitter of at least 120 us. This jitter is caused by bus arbitration. The sync message can be sent, if the transmission of the last CAN message is finished. In the worst case, the transmission of an other CAN message with 8 byte of date has just been startet. 8 byte of data means 111 bit = 111 µs. Now you can add a little additional jitter caused by your Sync producer and you have a minimum of 120 µs.

I don't think, that this should cause an error in an attached drive. The jitter simply leads to a deviation in the path if multiple drives are synchronized. The higher the speed of the drives, the higher the deviation. So if you have an application that does not have very high speed demands, than a certain jitter is no problem, because if you driver slower, the deviation becomes smaller. At least in the specifications I have not found any information, that a jitter in a sync message should cause an error. So this looks like a manufacturer specific implementation.

I think, the SYNC message is not meant for a perfect time synchronisation. It simply ensures, that all connected devices process their input and output data synchronously at the same time. For a perfect synchronisation, it is necessary to synchroniue the clocks of the connected drives via the "High resolution Time Stamp" protocol.

pJanez, 2015-06-01:
Synchronous PDOs are send just after the reception of SYNC message. (It doesnt matter, who generates it.) Data becomes valid after reception of the next SYNC message. That is general rule.

Internal timer thread, which controls the drive, should be synchronized with SYNC message.

It is possible to further control jitter, caused by bus arbitration. I don't remeber, where I read it, but it is possible to synchronise internal control threads in each drive to very accurate level. Suppose, we have a very accurate drives and master. In that case SYNC producer must send additional time information in the same sync message. This time information is the exact delay of previous sync message(it is possible to get information from the CAN module, when exactly previous sync message was sent). Each receiver of sync message can use this information to synchronize its own internal control thread. But that would be almost catching the speed of light :)

You can read here more about the SYNC and other protocols in DS301 or here:
http://www.can-cia.org/index.php?id=206

hekav, 2015-06-03:
Sorry for my late reply but I was on the way to my Vacation and had for two days no Internet access. 
When we started the project with a peak-usb interface. There you can get a little Windows program to send messages to the can bus. I initialized the controller and tried to put it in the IPM (interpolated positioning mode). I saw the goes into the IPM, but after 5 to 15 seconds it falls into fault state. So I measured the jitter with my Rigol DSO with the CAN analyzer directly on the wire. We got about 40 uS jitter. So my son asked a developer of the controller. And he said on this controller the maximum they can allow is 20 us. This is because the complete control chain depends on the accuracy of the sync interval. He said every nano second counts. This applies to Festo controller other vendors my be different. In many industrial applications all the master logic is done by a Siemens SPS and they have also problems with the jitter. In the newer Festo Controller they increased the jitter tolerance, but I don't know how much. 

So we had the 40 us jitter on Windows and I asked my what's about Linux? Because of the lag of a similar test application I modified Canfestival. And here I got the 200 us jitter.

mhaberler, 2015-06-03:
@hekav - I have certain doubts to the factuality of these claims

but if this jitter restriction is the case, it would be very interesting how they propose to achieve that kind of timing on a bus with contention (see Uwe's comment above)

They certainly have a reference setup which is supposed to work. Now how are they going to achieve that level of jitter? I mean _normal bus traffic_ will cause this to jump to _6 times_ this value. Doesnt add up for me.

So my counter question is: please show a working setup.

hekav, 2015-06-03:
Continue:
This was on the same hardware. Shouldn't a real time system by able to generate a signal every 8ms within a tolerance of 20 us. I newer application they change from SPS to VxWorks running on a PC.

mhaberler, 2015-06-03:
@hekav : "Shouldnt.." : No I do not think so - in particular not with a bus which has non-deterministic timing like CAN; maybe with EtherCAT but then this is pushing the envelope.

back of the envelope estimate:

RT thread jitter: anywhere in the range of 10-80uS
I/O delay added by RT CAN driver: some more - unknown
Bus access jitter: 0.. if idle... 111uS if bus busy (see @githubuser0xFFFF 's comment)

adds up to (gut feeling) anywere from 20 to > 150uS, of which most is outside of host control (transmit delay if bus busy - host cannot correct for that)

frankly I dont see how you can generate SYNC's with <20uS jitter except on an idle bus

hekav, 2015-06-03:
Michael,

No the sync message is only 3 bytes long and has the highest priority 0x00. It makes the canopen protocol capable of doing real timle. 
  At least on Festo controller the complete time calculations for the control chain is based on the time interval of the sync signal. Maybe that's a bad design, but that's how it is. 
I am on  vacation and don't have accessto my setup. When I am back I can give it to you. I am back in 4 weeks.

mhaberler, 2015-06-03:
@hekav: the length of the sync message is not the problem.

The problem is the time when the transmission of the sync message can begin. As explained above, this is determined by bus traffic and hence not something the host can do about.

Certainly any device which does path interpretation needs a timing reference, but the restriction they specify is (sorry to say) nonsense IMO - the standard method to do this is to implement a digital phase locked loop in  the drive to cancel out sync message jitter and peg timing to the PLL clock.

See the Rostan paper on "High Precision Drive Synchronisation with CANopen" or the DPLL scheme as implemented by the Mesanet FPGA's.

hekav, 2015-06-03:
@michael
There are thousands of setup ( SPS as the master and a motor controller) in the industrthey. Go to a assembly line and you will see them. 

When a master sends a sync message the hardware takes over the bus and all other nodes have to wait. That's why only one node should provide the sync message.

mhaberler, 2015-06-03:
@hekav - I have no doubts drives generally work with SPS's and in large numbers, but then I was trying to understand a different problem

I might not understand CAN bus access good enough, so please clarify:

does "taking over the bus" translate into "any ongoing transmission is preempted and automatically retried by hardware if some node is about to send a sync message"?

If the answer is yes, then this not quite add up with Uwe's comment above - at least the way I understood it

If the answer is no, how do you address the jitter introduced by ongoing transmissions? In that case I do not see a way how the node can influence jitter

githubuser0xFFFF, 2015-06-03:
Recarding jitter of sync message:

https://www.kuebler.com/PDFs/Feldbus_Multiturn/CANopen/busprofil_CAN.pdf

Page 47:

"There can be a time jitter in transmission by the Sync-Producer
corresponding approximately to the latency due to another message
being transmitted just before the Sync Object. You also have to
consider the case, that the Sync is transmitted twice."

So the lenght of the message sent just before the sync message is the problem - not the sync message itself.

mhaberler, 2015-06-03:
@githubuser0xFFFF - bingo! Now let's hear how the Festo firmware tries to get around that one ;)

[126 more replies]