machinekit/machinekit #592: 'Show HAL configuration' fails due to missing 'logpath'

modmaker, 2015-04-26:
MK was built from source. Axis is working, motors are moving. But when I open the hal configuration window that fails with the following messages:

```
emcTaskInit: using builtin interpreter
Error in startup script: couldn't load file "/home/bas/work/machinekit/tcl/hal.so": /home/bas/work/machinekit/tcl/hal.so: undefined symbol: logpath
    while executing
"load $::linuxcnc::TCL_LIB_DIR/hal.so"
    (procedure "hal" line 2)
    invoked from within
"hal list $node"
    (procedure "listHAL" line 5)
    invoked from within
"listHAL"
    (procedure "refreshHAL" line 12)
    invoked from within
"refreshHAL"
    (file "/home/bas/work/machinekit/tcl/bin/halshow.tcl" line 578)
```

Since 'logpath' is defined in hal/utils/halcmd_main.c, it looks like the that code is not linked to the library. Can someone with a better understanding of the build system shed some light on this?

Thanks,
-- Bas

mhaberler, 2015-04-26:
yes, that was me. Blush. Fix coming.

mhaberler, 2015-04-26:
Bas, @ArcEye - I merged a fix (out of C4 process..) - please pull and retry

ArcEye, 2015-04-26:
What is the chance of 2 people finding the same error at pretty much the same time?!

Have done the pull and it builds, but will have to reboot, had problem with the mutex auto-release not working whilst doing the mods to halcmd and now realtime is jammed:(

Let you know soon

ArcEye, 2015-04-26:
Yes, that works again

modmaker, 2015-04-26:
Thanks, pulled changes and building right now.

modmaker, 2015-04-26:
Fixed!

closed 2015-04-26 by modmaker