machinekit/machinekit #597: Xenomai test failures

zultron, 2015-04-29:
I can't point to a specific commit where this intermittent issue began.  Looks like the first PR was #580, which contains nothing that would cause test failures (esp. on x86).  First the x86-64 Xenomai slave, now the i386 Xenomai slave began failing with two problems.

x86_64 `xenomai-kernel` threads build, [many tests failed][1]:
[1]: http://buildbot.dovetail-automata.com/builders/d7-64-x-k-tst/builds/659/steps/runtests/logs/stdio

    MSGD:0 ERROR: cannot create global segment key=0x154711 No such file or directory

And then a leftover process causes a failure in the ['cleanup' phase][2]:
[2]: http://buildbot.dovetail-automata.com/builders/d7-64-x-k-tst/builds/659/steps/clean-buildroot/logs/stdio

    umount: /var/lib/buildslave/d7-64-bldtest/build/source: device is busy.

By the time I found this, the process had already exited.

Once the problem occurs once, it seems to recur consistently until the build slave is rebooted.

x86_64 failed PR builds:  #580, #582, #584, #587 (twice)
i386 failed PR builds:  #588, #596

I'll track the problem in this issue, or else close it out if it turns out to be spurious.

mhaberler, 2015-04-29:
Weird. trying an x-k on an amd64 now.

mhaberler, 2015-04-29:
well I did, three runs of runtests with FLAVOR=xenomai-kernel on an amd64 and all pass, so here I cant observe any kernel wedging here

any chance to retrieve a lsmod  + dmesg output after failure?

mhaberler, 2015-04-29:
hm, suddenly green. Rattling the bag of bones ;-?

zultron, 2015-04-29:
Thanks for trying that, Michael.  Yeah, a reboot fixes the problem; this issue is just to track the problem in case it resurfaces.

zultron, 2015-05-28:
I haven't seen this since.  Closing issue.

closed 2015-05-28 by zultron