machinekit/machinekit #598: machinekit not ready for Xenomai3, jessie, beagleboard X15

mhaberler, 2015-04-29:
we have several major release changes ahead in supporting components and platforms:

- debian jessie will become stable later this spring
- the [Xenomai project is readying a major release] (https://xenomai.org/introducing-xenomai-3/)
- the [Beagleboard X15](http://elinux.org/Beagleboard:BeagleBoard-X15) high-end successor to the Beaglebone will appear sooner or later but is not supported by the current kernel

there were some offlist discussions on the topic. To summarize the key issues:

- the jessie transition will bring some relief on most of the custom packages required, but not all of them
- the Xenomai 2->3 transition requires a new kernel, an new runtime support package, and review/likely some changes in the RTAPI xenomai code
- the Xenomai2->3 transition affects Xenomai kernels for all platforms (x86, amd64, arm) but there is a less stringent version lock-in with x86 and amd64 as most of required support code is mainlined
- there is a 3.14 xenomai kernel based on a TI branch which supports both the BBB and the X15, however it does *not* support capemgr (dynamic loading of device tree overlays) -- all device tree overlays needed must be compiled into the kernel. capemgr will appear mainline hopefully but unclear when, and when a Xenomai patch for the matching baseline version is available.
- for the TI 3.14 based kernels, also an RT-PREEMPT kernel is available, which might be interesting for 7i80 ethernet users and other applications.
- Charles' config-pin utility is most what is needed; some tweaks might be required.

resulting feature matrix and options:
````
kernel    RT        hardware    Xenomai   capemgr   config-pin

3.8       xeno      BBB         2 and 3    yes     yes
3.14-ti   xeno      BBB+X15     2 and 3    no      yes
3.14-ti   rtpreempt BBB+X15     N/A        no      yes

3.14      xeno      x86,amd64   2 and 3    N/A     N/A
3.18(20?) rtpreempt x86,amd64   N/A        N/A     N/A
````

Based on the above, for a major transition there was some preference for:

- including Xenomai 3 in the transition
- seek to have appropriate kernels for all platforms, which would suggest 3.14-xenomai for  BBB/X15 and at least a Xenomai2->3 iteration on the x86/amd64 kernels
- ideally have SGX drivers and accelerated Qt based desktops on the TI platforms

the missing capemgr seems not to be a showstopper, but we must assure all required device trees are compiled in. We might need a bit of soul searching so we have all candidates rounded up.

so much for the _wouldnt it be nice if.._ feature list.

the intent of this issue: collect all aspects on how to get there

RunningLight, 2015-04-29:
On 04/29/2015 05:56 AM, Michael Haberler wrote:
> debian jessie will become stable later this spring

Excuse me. As of 25 April 2015, Debian 8 is the "latest stable release." 
That's why I'm exercising John's fixes.

mhaberler, 2015-04-29:
oops, updated

zultron, 2015-04-29:
Packaging-related differences:
- MK deps absent from Jessie that I'm preparing to ship:
  - Xenomai and RTAI runtime and kernels
  - `czmq`
  - `libwebsockets`: in Jessie; remind me why we have our own package?
- `zeromq` CURVE support missing in Jessie pkgs:
  - `zeromq3` is in Jessie, as is `libsodium`, but `zeromq3` not compiled with CURVE support
  - AFAIK, MK doesn't actually implement CURVE support, so I dropped those packages
  - Therefore my `czmq` package also drops CURVE support
- Jessie dropped the RT_PREEMPT kernel

Kernels:
- At some point I'll update the ipipe-based kernels.
  - Will use Xenomai 3
  - Looks like there's a 3.16.7 ipipe patch in both Xenomai 3 and RTAI
    - Since RCN is shipping our BBB kernels, I don't need to target 3.14, and can skip to 3.16, yes?
- Debian dropped RT_PREEMPT
  - Fuckers
  - No 3.16 patch, so the featureset won't be in my combined ipipe kernel package
  - A quick fix is to rebuild the Wheezy package for Jessie
  - Need to investigate alternate/3rd-party package sources

mhaberler, 2015-04-29:
libwebsockets - that was an API change we needed, I need to check if the jessie version already has it

re CURVE: oops - this is a a really important todo item IMO, we have no authentication on the zeromq sockets right now (!): #204 - we really should have the option in case we find somebody to work on it, that issue is not going away (I didnt know until now they are not included anymore? maybe I missed something)

re BB kernel: I think 3.16 is ok, yes: http://download.gna.org/adeos/patches/v3.x/x86/

re RT_PREEMPT: Xenomai3 can run over RT_PREEMPT as well, I have no experience with that yet

mhaberler, 2015-04-29:
libwebsockets: just checked - unfortunately the jessie version is still useless for us, https://github.com/warmcat/libwebsockets/commits/v1.21-chrome26-firefox18

zultron, 2015-04-29:
Buildbot:
- New Docker-based build scripts looking pretty good
  - Build Wheezy and Jessie (and Trusty and part of RPi) MK and build-dep packages
  - Build x86_64, i386 and armhf packages (with below Jessie/ARM exception)
- The Jessie `qemu` regression threw a wrench into my plan for retiring the native ARM builder
  - Options:
    - Temporarily acquire additional ARM builder to develop Buildbot config (@cdsteinkuehler: nudge, wink ;)
      - Build scripts need to be updated for ARM-arch Docker container
    - Issue is with `gettext` utilities; teach build system to only build that for binary-indep pkg builds
- Then, set up Jessie test slaves
  - For now, slaves will continue to be separate VMs for each distro+arch+kernel combo
  - Still no provision for RIP build in new builder scripts
    - Current Buildbot builds separate RIP build for use in tests
    - Goal is package-only builds, but this would require packaging functional tests
    - TBD which can be set up most quickly

Dovetail Automata package repository:
- No significant differences planned other than adding Jessie and later, possibly Trusty and RPi

zultron, 2015-04-29:
On 04/29/2015 09:38 AM, Michael Haberler wrote:
>
> re CURVE: oops - this is a a really important todo item IMO, we have no
> authentication on the zeromq sockets right now (!): #204
> <https://github.com/machinekit/machinekit/issues/204> - we really should
> have the option in case we find somebody to work on it, that issue is
> not going away

If someone wants to work on it, I'll pick up the package again.  What 
I'd prefer is someone work with the Debian developers to rebuild the 
`zeromq3` and `pyzmq` packages with `libsodium`.  The pieces are all 
there, just needs to be enabled.

#204 neglects an important question:  what is the infrastructure for 
sharing the crypto data needed to establish trust?  It's been a while 
since I looked at CURVE, but talking in terms of other crypto systems, 
there's a certificate authority pki like ssl, pubkeys like ssh, and 
shared secrets like WPA personal, telnet, etc.  IIRC CURVE supports at 
least two of those.

IMO there's still a way to go before any kind of implementation can 
start, and my plate's already pretty full.

 > (I didnt know until now they are not included anymore?
 > maybe I missed something)

Upstream distro packages never had CURVE support.  The only change for 
Jessie is I'm dropping my Wheezy `zeromq3` etc. packages that did.

mhaberler, 2015-04-30:
I'll take your comment on #204 over there to keep this focused on the 'next iteration' issues. But I hope we have agreement that authentication is a must-have prerequisite, even if not implemented as of now. The situation now is worse than the LinuxCNC approach (which is: unencrypted TCP and a compiled-in default password which everybody can read in the repo).

re prerequisites packages which are not available upstream:

- I understand the desire to use stock packages only - building our own is a lot of work, so of course it would be best not to have any at all
- back when we decided to build a few of our own packages, the driving factor was: either the packages were not available at all, or the versions were so outdated that  we could not use them. So we decided to go for a particular released tree (eg zeromq 4.0.x) and use that API version in machinekit.
- the hope was: eventually the debian packages will catch up and we can stop producing our own; and for most of the packages this worked out. But this relies on an external factor we have no control over right now: some projects care less what goes into the debian stream. libwebsockets and zeromq are clearly in this 'dont care' category. So that has not happened, and it's questionable if we ever reach this state.

the question at hand now is - how do we deal with the situation? 
- we can stay on course with our own packages and hope for upstream.
- we can work with the people doing the debian packages in parallel to get those packages up to snuff.
- we can try to step back in terms of API's to whatever is provided upstream. 

In the case of zeromq and libwebsockets, I do not see a sensible way for the last option. And: if we decide to change course now, then we have been addressing the wrong problem in terms of packaging so far - we should have worked with the upstream maintainers right away.

I guess we need to make a decision how to proceed.

RunningLight, 2015-04-30:
This discussion illustrates our severe manpower limitation. Practically, to "work with the people doing the debian packages" means dividing John's time even more (although I suspect him of having at least one clone already, given how much he's doing!). Further, I suspect any forceful expression of need on our part would elicit the reasonable suggestion that we become co-maintainers of the debian packages in question. They have their manpower problems too. I can hope for the best, that nudging package maintainers will do the trick, but planning for the worst, I think we have to "stay on course." We will always face the problem that one package or another may diverge from our expectations.

As for authentication, I consider it irresponsible in this day and age not to build it into our architecture from the get-go even if what we are limited in what we can implement today. Just my 2-cents worth.

@zultron I'm unclear about parts of your Buildbot synopsis:

*  `Temporarily acquire additional ARM builder to develop Buildbot config (@cdsteinkuehler: nudge, wink ;)`  --- could you be more specific here? I first took "builder" to mean a human being, to wit, Charles, but it could also mean a builder in the Buildbot sense. What "Buildbot config" development is needed? Do you mean solely updating the ARM-arch Docker container to accommodate Jessie or are there other issues?

*  `For now, [Jessie] slaves will continue to be separate VMs for each distro+arch+kernel combo` --- I thought that's what all your slaves are now, except for BBB. If so, what would you change?

zultron, 2015-04-30:
I agree that we should've been working with Debian maintainers.  Also agree with these:
- we can stay on course with our own packages and hope for upstream.
- we can work with the people doing the debian packages in parallel to get those packages up to snuff.

So, I'll pick up the packages since we all feel it's important to have security, and on the assumption that we're going to start work on #204 soon, and the hope that we can start working with upstream packagers.

@RunningLight, by 'ARM builder' I mean the ARM board (Wandboard) that's doing the ARM builds.  It's been flakey, needs constant attention, and after spending several hours just now replacing the rootfs sdcard (mostly repairing packages after a hang in the update), I don't think it's going to be any better.  I'd like to remove it and do builds under qemu, but that may be too much work in the short-term (although I have resolved the Jessie-qemu problem).

As for *test* slaves, you're right, they're all VMs, except for BBB, and for now will stay that way.  I have ideas for a different scheme that I probably shouldn't have mentioned.

mhaberler, 2015-04-30:
@zultron - thanks, I really appreciate it!

mhaberler, 2015-05-01:
with a bit of luck we might have the libzmq/czmq package issue solved for us. [If I read this correctly, rsyslog uses zeromq plugins, and has a similar prerequisite issue like us](http://comments.gmane.org/gmane.network.zeromq.devel/26281) . I'll inquire with the author if this has any upside for us.

RunningLight, 2015-05-01:
@zultron, as for the ARM builder, I'm experimenting with my own Wandboard Quad, an Odroid-C1, and a CuBox-i4Pro. They all have Gigabit ethernet ports, so an nfs-mounted rootfs should have decent benefit, and the Wandboard and the CuBox-i include a SATA port. I'd like to finish getting some performance numbers, but any or all can be contributed to the Buildbot whether as remote of local slaves.

mhaberler, 2015-05-03:
zmq* and debian - studying who is doing what here:

libzmq:

http://zeromq.org/distro:debian says a Alessandro Ghedini maintains [debian libzmq3](http://anonscm.debian.org/cgit/collab-maint/zeromq3.git/)
however this repo has serious cobwebs, last change 2014-03-16 ie 14 months ago
Alessandro's debian [QA page](https://qa.debian.org/developer.php?login=ghedo%40debian.org) does not list any zmq-related packages any more: 

https://packages.debian.org/jessie/libzmq3-dev says the maintainer is [László Böszörményi](  https://plus.google.com/112158872274462443530/about), for stable this lists 4.0.5+dfsg-2 (John: I assume this is the one we could use modulo the missing curve support?)

this looks like a change in maintainers to me, I have no idea what the practices are but looks like László is the man to talk to - does it make sense approaching László and suggesting a jessie-backports or unstable libzmq+curve package (or maybe a different package - libzmq3-sodium; unsure)? 

czmq: 

http://zeromq.org/distro:debian  says Alessandro Ghedini and Gergely Nagy manage this
Gergely Nagy's [debian QA page](https://qa.debian.org/developer.php?login=algernon@madhouse-project.org) does not list czmq: 
Allesandro indicated [he lost interest](http://lists.zeromq.org/pipermail/zeromq-dev/2013-December/023852.html)
the last released commit was from a different person, [Arnaud Quette]( http://arnaud.quette.fr/post/2012/07/06/opening%3A-Hello-Planets), looks like an experienced maintainer

http://anonscm.debian.org/cgit/collab-maint/czmq.git/ has last change in Nov 2014, but API version 2.2

pyzmq:

current debian maintainer: Julian Taylor, jessie lists 14.4.0-1 
unsure if sodium-enabled
do we consider the alternative - installing via pypy?

mhaberler, 2015-05-03:
re libwebsockets & debian: I've filed an issue:   https://github.com/warmcat/libwebsockets/issues/285

cdsteinkuehler, 2015-05-03:
@zultron I've also got two quad-core ARM boards (CuBox and Wandboard) I'd be happy to setup as build slaves.  Throw the buildbot configs online somewhere and I'll see about setting one up to test.

zultron, 2015-05-06:
(The packaging plan above is now updated to reflect the fixed Debian Jessie `zeromq3` packages and fixed `qemu` in the build scripts.)

mhaberler, 2015-05-06:
wow, I'm really relieved that the zeromq situation is clearing up, modulo czmq. Thanks, John  - great move on the debian bug report!

mhaberler, 2015-05-13:
I'm confused wrt jessie install-from-source - is this now possible with the dovetail automata packages and the stock jessie package stream, or not yet?

I tried an x86 jessie install, mutated the website instructions /wheezy/jessie/ and got stuck here:

````
root@j1900:/home/mah# apt-get -f install
Reading package lists... Done
Building dependency tree
Reading state information... Done
Correcting dependencies... Done
The following packages were automatically installed and are no longer required:
  libgssglue1 libmpc2
Use 'apt-get autoremove' to remove them.
The following extra packages will be installed:
  libzmq3
The following NEW packages will be installed:
  libzmq3
0 upgraded, 1 newly installed, 0 to remove and 1249 not upgraded.
54 not fully installed or removed.
Need to get 0 B/442 kB of archives.
After this operation, 725 kB of additional disk space will be used.
Do you want to continue [Y/n]? y
(Reading database ... 139357 files and directories currently installed.)
Unpacking libzmq3:i386 (from .../libzmq3_4.0.5+dfsg-2+deb8u1_i386.deb) ...
dpkg: error processing /var/cache/apt/archives/libzmq3_4.0.5+dfsg-2+deb8u1_i386.deb (--unpack):
 trying to overwrite '/usr/lib/i386-linux-gnu/libzmq.so.3.1.0', which is also in package libzmq4:i386 4.0.4-2~1da~wheezy1
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/libzmq3_4.0.5+dfsg-2+deb8u1_i386.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
````

is the wheezy install as per website still what we recommend to do, or is it jessie and we have just not updated the website yet?

zultron, 2015-05-13:
The Dovetail Automata Jessie packages are super stale tests (and were never announced for either testing or GA).  Don't use them, and please remove `libzmq4` if it's installed.

I'll have time to push out a new repo for testing in a few days, and will call for testing then.  Thanks!

RunningLight, 2015-05-13:
@zultron Is your deb-testing repo still worth testing our revised procedure on?

RunningLight, 2015-05-14:
FYI, I just built MK in Jessie against your deb-testing repo ala the newly revised http://machinekit.io/docs/building-from-source page (changed as necessary for your repo and for jessie). I had no problems...until, that is, I tried to start the sim/axis-mm configuration from Axis and ran into tcl awfulness. 

In the process of trying to sort tcl/tk 8.5 vs tcl/tk 8.6 I realize there's a disagreement between the informative message from `src/configure` and the output from `src/configure --help` which I'm posting in a new issue.

Starting with a fresh VM, I got to:

```
kreed@jessie-amd64-xfce-virgin:~/Desktop/machinekit$ debian/configure -p -t8.5
debian/control:  copied base template
debian/rules:  copied base template
debian/control:  added POSIX threads package
debian/rules:  enabled posix threads
debian/control:  Set tcl/tk build deps to version 8.5
debian/control:  add final Build-Depends: list
```

and then, at the end of `src/configure`

```
...
checking version of python libraries... python2.7
checking match between tk and Tkinter versions... TCL mismatch: 8.5 vs 8.6
configure: error: Python requires use of Tcl 8.6 and Tk 8.6.
Install this version and specify --with-tclConfig and --with-tkConfig if necessary
```

At this point, versions tcl8.5, tcl8.5-dev, tcl8.6, tk8.5, tk8.5-dev, and tk8.6 are all installed, but I don't seem to be able to find the right incantation to satisfy the configure gods at this point. Forgive my poor attention span (Washington Capitals are playing Game 7 against the New York Rangers as I write this!). Perhaps you've resolved this in email exchanges in the past but I can't find it.

zultron, 2015-05-14:
@RunningLight, I don't have any Jessie repo online that I think might work.

Remove `tcl8.5` and `tk8.5`.  Apparently, `m4/tcl.sh` (a less-broken copy of the broken system-installed version from Jessie) searches for `/usr/lib/tkConfig.sh` before `/usr/lib/tk8.6/tkConfig.sh`, and the former may be a symlink to v. 8.5 if that version was installed after the `tcl8.6` package.

RunningLight, 2015-05-14:
@zultron Ok, I know you're working on the real deal so this is just for fun using deb-testing.

`tcl8.5` and `tk8.5` got in there because of my meddling. Sorry, I shouldn't have mentioned it. I had tried `debian/configure -t8.5` only after my first build succeeded but I couldn't run the sim/axis-mm configuration.

Back to a virgin build on a virgin jessie-amd64 VM: It builds fine (modulo the pub-key business, which I know you'll take care of when you publicly release a jessie repo). 

Invoking `. ./scripts/rip-environment`, then invoking `machinekit`, and selecting the sim/axis-mm  configuration, I get to the expected Axis window. In it, I can toggle Emergency Stop, but when I then toggle Machine Power, homing buttons don't light up and I can't home axes.

Meanwhile, I see the following reported on the command line from which I invoked `machinekit`

```
kreed@jessie-amd64-xfce-virgin:~/Desktop/machinekit$ machinekit
MACHINEKIT - 0.1
Machine configuration directory is '/home/kreed/Desktop/machinekit/configs/sim/axis'
Machine configuration file is 'axis_mm.ini'
Starting Machinekit...
io started
halcmd loadusr io started
task pid=12646
emcTaskInit: using builtin interpreter
//...at this point I toggle Machine Power...
TCL error in asynchronous code:
wrong # args: should be ".toolbar.program_step cmd ?arg ...?"
    while executing
".toolbar.program_step"
    invoked from within
"if {$::last_interp_state != $::INTERP_IDLE || $::last_task_state != $::task_state} {
                set_mode_from_tab
            }"
    (procedure "update_state" line 44)
    invoked from within
"update_state"
    ("after" script)
```

Granted, I did not consider that this may be an artifact of some recent change to the repo and not the result of building on jessie. I'll try the same on wheezy tomorrow.

Oh, yeah, the Washington Capitals were eliminated from the Stanley Cup Playoffs by the New York Rangers tonight in the first Overtime period of Game 7.

zultron, 2015-05-17:
(Updated issue:  added 3.18.12 + Xenomai-3 and RTAI kernel candidate)

zultron, 2015-05-17:
Ok, I've got a new `deb-testing` repo rsyncing right now for Jessie-x86.  Same deal, url is http://deb.dovetail-automata.com/deb-testing/, packages are signed with a random key.  100% untested, aside from verifying that MK packages can be built.  This is the package base I'm getting ready to support, so you're not wasting your time testing.

No `armhf` packages yet.  Anyone know the status of RCN's Jessie-BBB images?  Do they exist?  Do they have kernels?  Once I know what needs to be built, I'll do so.

cdsteinkuehler, 2015-05-17:
On 5/17/2015 3:41 PM, John Morris wrote:
> No `armhf` packages yet.  Anyone know the status of RCN's
> Jessie-BBB images?  Do they exist?  

Yes, for some time now.

> Do they have kernels?  

There's a 3.14 kernel with Xenomai patches (untested AFAIK), but it's
possible to continue running the known-good 3.8.13 as well.

-- 
Charles Steinkuehler
charles@steinkuehler.net

zultron, 2015-05-17:
On 05/17/2015 04:40 PM, cdsteinkuehler wrote:
> There's a 3.14 kernel with Xenomai patches (untested AFAIK), but it's
> possible to continue running the known-good 3.8.13 as well.

Ok, so RCN is handling armhf-architecture Xenomai and kernel packages. 
He picked up most other MK deps for Wheezy, too (ISTR he had trouble 
with Cython, but we drop that dep for Jessie), so I'll assume he'll 
continue carrying `czmq` and `libwebsockets`.

Wow, looks like I don't need to build any MK deps for armhf, just MK 
itself when the Jessie Buildbot comes online.

Hey, sorry for being out of the loop.

RobertCNelson, 2015-05-17:
@zultron yeap, just let me know which versions of czmq and libwebsockerts we need and i'll push them out to the armhf repo. ;)

RobertCNelson, 2015-05-17:
btw, on the qt5 side, i briefly played around with backporting 5.4.1 from stretch/sid, ran into some other fun problems. So we will stick with jessie's qt: 5.3.2 as default. I'm continuing to follow lxqt releases's and build them for armhf, so we have a full qt5 based desktop.
Regards,

zultron, 2015-05-18:
@RobertCNelson, let's defer to @mhaberler on the version questions:

`czmq` v. 3 is close to coming out; if we can't wait for that, you can probably just grab my packaging.

I forget how we picked the particular (seemingly random) git rev that the `libwebsockets` packaging is based on, and haven't found a tagged rev that MK builds correctly against.

mhaberler, 2015-05-18:
@zultron @RobertCNelson - to build right now we need czmq2; I'd have to go over to API to fix the breakage for czmq3 (mostly haltalk and webtalk). So ideally we had a bit of overlap with the current czmq2 package, and once API change is done I'll change the build deps.

I'll have a look at the libwebsockets history if anything is worth in the history suggests forwarding since we froze it; the czmq rev choice was based on a patch I added to libwebsockets to support interworking with the czmq event loop

mhaberler, 2015-05-18:
czmq3 : well it turns out there is no API breakage at all, they retained the legacy 2.2 API as well. Built czmq v3.0.0 and built machinekit against that, no issue.

This means we could in fact switch to czmq3 immediately (probably better to wait for Pieter tagging a release so we dont have to redo, but that is it).

The V3 API can come in piecemeal; it's really just stronger typing on sockets, better support library functions, and more efficient timer handling - nothing which has to be done right away

mhaberler, 2015-05-19:
well, the czmq V2->V3 transition wasnt that hard: https://github.com/mhaberler/machinekit/commits/czmq3

checking compatibility with libczmq2 now - for my development box I've switched to czmq3 already

zultron, 2015-05-19:
Here are [Linux 3.16.7][1] + [Xenomai-3.0-rc4][2] packages and [Machinekit building against them][3], completely untested as usual.  ;)  It looks like we retire `xenomai-kernel` threads with Xenomai-3.

[1]: https://github.com/zultron/linux-ipipe-deb/tree/3.16.7
[2]: https://github.com/zultron/xenomai-deb/tree/3.0-rc4
[3]: https://github.com/zultron/machinekit/tree/xenomai-3

ShabbyX has [merged `magma` into his RTAI repo][4], so that can be pulled into the kernel package now.

[4]: https://github.com/ShabbyX/RTAI/issues/23

I've gotta take a break for a day or two, but let's step back; it looks like many boxes are close to tickable as far as packages are concerned:

- CZMQ-3
- Xenomai-3
- Newer x86 Xenomai, RTAI, RT_PREEMPT (with @arceye's work) kernels
- TI kernels
- Machinekit updates for the above

A Jessie Buildbot is conspicuously absent.  Lots of small details to work out.  What's the big plan?

mhaberler, 2015-05-19:
My own priority would  be to get the packaging and build shaken out for jessie; if the packages per se are bleeding edge or not is a secondary question

I dont think there is any urgency in the czmq V2->V3 API transition - whenever Pieter does his thing. My experience is that one can perfectly go ahead with the czmq3 dev branch, a local V3 czmq build and work for instance on the authentication issue. It's fairly localized to a few daemons.

Xenomai-3+TI Kernel - as this is preparatory for the X15 platform, I think "in summer" is fine

RunningLight, 2015-05-20:
>  My own priority would be to get the packaging and build shaken out for jessie; if the packages per se are bleeding edge or not is a secondary question

I agree 100%. Jessie is now the Debian Stable release. To my way of thinking it is imperative for us to be building and packaging for it.

mhaberler, 2015-05-21:
further to the CZMQ V3 API issue - the https://github.com/mhaberler/machinekit/tree/czmq3 branch now works fine with czmq master, however I made two PR's against czmq master which should be picked onto the V3 release branch by Pieter before that release is usable for us:

e08fcec31a540501f80c86916bfa85ed10b726df and 82ef83b360aafcbefa8262992637f84e4568d415

mhaberler, 2015-05-23:
the two required commits are now mainlined in zeromq/czmq master, Pieter confirmed they will be part of the V3 release
the matching  https://github.com/mhaberler/machinekit/tree/czmq3-cleaned branch is ready to go as soon as we have V3 packages

zultron, 2015-05-28:
The Dovetail Automata Buildbot is now building/testing Jessie x86 arch.  Soon after the next PR merge we'll have MK packages.

These are still the old 3.8.13, tried-and-true kernel packages.  No newer kernels, no RT_PREEMPT, no ARM arch, no Xenomai 3.0, no RTAI 4.2/whatever.

zultron, 2015-05-31:
Experimental packages, 3.16.7 kernel with Xenomai-3 and updated ShabbyX/RTAI available in the new [jessie-bleeding][1] repo.  This should lower the bar for anyone interested in updating Machinekit to run against the updated software.

Note the separate issues:  #641 related to the Xenomai-3 port, and #653 related to where I left off on the RTAI updates.

[1]: http://www.dovetail-automata.com/articles/2015/05/30/jessie-bleeding-kernel-packages/

mhaberler, 2015-06-22:
debian/control.in still lists 

debian/control.in:    libzmq3-dev (>= 4.0.4) | libzmq4-dev (>= 4.0.4),

shouldnt the libzmq4-dev reference be removed?

zultron, 2015-06-22:
Yes.  The Dovetail Wheezy repo still has the libzmq4 package.  That can be removed in one of these cases:
- folks stop using the Dovetail Wheezy repo, or
- someone figures out the right incantation for the Wheezy libzmq3 `Breaks: libzmq4` etc. package relations and the Dovetail repo is updated with replacement libzmq3 packages.

I'm hoping we can just wait until Wheezy is retired.

zultron, 2015-06-22:
Looking over this issue once again (what a clutter), here's a summary of issues raised and status.

### Outstanding issues:
-  `libzmq3-dev | libzmq4-dev`:  file new issue?
- `czmq` v3 packages:  again I'd like to be lazy & wait for Wheezy's retirement party; file new issue?
- Jessie RT_PREEMPT kernel:  both @arceye and Ben Hutchings have kernels; investigate

### Resolved issues:
- CURVE:  package-related issues noted here resolved; work on #204 unblocked
- Updated Xenomai and RTAI kernels (3.16):  done; in Dovetail repo; see #641 and #653
- `libwebsockets`:  Nothing to do
- Buildbot now building and testing on Jessie (but RT_PREEMPT and BBB still missing)
- Most Jessie build issues resolved (Tcl/Tk 8.5 problem lingers)

### Unknown (to me) issues, mostly BBB:
- TI 3.14 Xenomai (and RT_PREEMPT?) kernel; @cdsteinkuehler (in https://github.com/machinekit/machinekit/issues/598#issuecomment-102861038) indicates this is resolved
- @cdsteinkuehler's `config-pin` utility
- BBB `czmq` and `libwebsockets` packages; not sure RCN ever got a solid answer on versions after https://github.com/machinekit/machinekit/issues/598#issuecomment-102869475

mhaberler, 2015-06-22:
czmq: Pieter quietly rolled a release: https://github.com/zeromq/czmq/commits/v3.0.2
trying linux-image-3.14.43-ti-xenomai-r67 right now
I didnt find the Hutchings kernels yet - where? better buy than make

zultron, 2015-06-22:
On 06/22/2015 12:19 PM, Michael Haberler wrote:
> I didnt find the Hutchings kernels yet - where? better buy than make

Mmm, I remember him writing that RT_PREEMPT was enabled in the 4.0.4 
kernels, and that he'd added the patches.  I don't see any kernels, but 
I assumed it's a matter of waiting.

	John

mhaberler, 2015-06-22:
re RT-PREEMPT/Am335x: Charles mentioned 

>Looks like OSADL has one in their QA farm:
https://www.osadl.org/Profile-of-system-in-rack-7-slot-8.qa-profile-r7s8.0.html
Kernel version 3.18.13-rt9, with some off-tree patches (provided).

RobertCNelson, 2015-06-23:
@zultron 

I grabbed czmq/libwebsockets from: http://deb.dovetail-automata.com/deb-testing/pool/main/
(not sure if that's the latest pkg version from you guys..)
```
czmq: 2.2.0-0.5%7e1jessie%7e1da
libwebsockets: 1.3-1%7egit95a8abb%7e1431844465git95a8abb%7e1jessie%7e1da
```

Regards,

zultron, 2015-06-23:
On 06/23/2015 09:08 AM, Robert Nelson wrote:
> I grabbed czmq/libwebsockets from:
> http://deb.dovetail-automata.com/deb-testing/pool/main/
> (not sure if that's the latest pkg version from you guys..)
>
> |czmq: 2.2.0-0.5%7e1jessie%7e1da
> libwebsockets: 1.3-1%7egit95a8abb%7e1431844465git95a8abb%7e1jessie%7e1da

Those should be ok, I think.  The stable packages are:

> http://deb.dovetail-automata.com/pool/main/libw/libwebsockets/libwebsockets_1.3-1%7egit95a8abb%7e1432455350git95a8abb%7e1jessie%7e1da.debian.tar.xz
> http://deb.dovetail-automata.com/pool/main/libw/libwebsockets/libwebsockets_1.3-1%7egit95a8abb%7e1432455350git95a8abb%7e1jessie%7e1da.dsc
> http://deb.dovetail-automata.com/pool/main/libw/libwebsockets/libwebsockets_1.3.orig.tar.xz
>
> http://deb.dovetail-automata.com/pool/main/c/czmq/czmq_2.2.0-0.5%7e1jessie%7e1da.debian.tar.xz
> http://deb.dovetail-automata.com/pool/main/c/czmq/czmq_2.2.0-0.5%7e1jessie%7e1da.dsc
> http://deb.dovetail-automata.com/pool/main/c/czmq/czmq_2.2.0.orig.tar.gz

Thanks!

zultron, 2015-06-23:
Oh, @RobertCNelson, the `~1da` should be removed, since that's a marker for the Dovetail Automata repo, and feel free to fix that nasty `libwebsockets` package version.

zultron, 2015-06-25:
Here's where my information about the Debian kernel came from.  Still nothing tangible, AFAICT.

http://thread.gmane.org/gmane.linux.rt.user/13655/focus=13673

luminize, 2015-06-25:
I thought I give it a shot, building on Jessie. This is (if memory serves me right) an upgrade from wheezy to jessie. So no fresh VM here.
I've added the deb-testing repo so my machinekit.list shows:
```
#deb http://deb.dovetail-automata.com wheezy main
deb http://deb.dovetail-automata.com/deb-testing jessie main
```

however I get the following message after `sudo apt-get update`, `sudo apt-get upgrade` and after getting a hint of the terminal of `sudo apt-get upgrade -f`

```
(Reading database ... 291571 files and directories currently installed.)
Preparing to unpack .../libzmq3_4.0.5+dfsg-3_amd64.deb ...
Unpacking libzmq3:amd64 (4.0.5+dfsg-3) ...
dpkg: error processing archive /var/cache/apt/archives/libzmq3_4.0.5+dfsg-3_amd64.deb (--unpack):
 trying to overwrite '/usr/lib/x86_64-linux-gnu/libzmq.so.3.1.0', which is also in package libzmq4:amd64 4.0.4-2~1da~wheezy1
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/libzmq3_4.0.5+dfsg-3_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
more output: http://pastebin.com/fYBb35XT

If I'm too soon with my trying to install then give me a nudge and tell me to wait :)

zultron, 2015-06-25:
On 06/25/2015 02:09 PM, Bas de Bruijn wrote:
> I thought I give it a shot, building on Jessie. This is (if memory
> serves me right) an upgrade from wheezy to jessie.

Yes, the upgrade contributes to the problem.

> Unpacking libzmq3:amd64 (4.0.5+dfsg-3) ...
> dpkg: error processing archive /var/cache/apt/archives/libzmq3_4.0.5+dfsg-3_amd64.deb (--unpack):
>   trying to overwrite '/usr/lib/x86_64-linux-gnu/libzmq.so.3.1.0', which is also in package libzmq4:amd64 4.0.4-2~1da~wheezy1

Manually remove all `zeromq4` packages and try again.

luminize, 2015-06-26:
I have removed zeromq4 with aptitude (apt-get got me into an endless circle with broken packages)

looks like I need some further package hand-holding

I run `sudo scripts/apt-installbuilddeps`
and after a looooot of lines get the message:
```
[......]
Note, selecting 'python-rosunit' instead of 'python-rosunit:any'
Note, selecting 'python-roswtf' instead of 'python-roswtf:any'
Note, selecting 'python-topic-tools' instead of 'python-topic-tools:any'
Note, selecting 'linux-headers-3.8-1-rtai.x86-amd64' instead of 'linux-headers-rtai.x86'
Note, selecting 'linux-headers-3.8-1-xenomai.x86-amd64' instead of 'linux-headers-xenomai.x86'
Note, selecting 'linux-image-3.8-1-rtai.x86-amd64' instead of 'linux-image-rtai'
Note, selecting 'linux-image-3.8-1-rtai.x86-amd64' instead of 'linux-modules-3.8-1-rtai.x86-amd64'
Note, selecting 'linux-image-3.8-1-rtai.x86-amd64' instead of 'rtai-modules-3.8-1-rtai.x86-amd64'
Note, selecting 'linux-image-3.8-1-xenomai.x86-amd64' instead of 'linux-image-xenomai'
Note, selecting 'linux-image-3.8-1-xenomai.x86-amd64' instead of 'linux-modules-3.8-1-xenomai.x86-amd64'
Note, selecting 'linux-image-rtai.x86-amd64' instead of 'linux-latest-modules-3.8-1-rtai.x86-amd64'
Note, selecting 'linux-image-xenomai.x86-amd64' instead of 'linux-latest-modules-3.8-1-xenomai.x86-amd64'
Note, selecting 'python-rtai' instead of 'python2.7-rtai'
Note, selecting 'linux-image-3.16.0-0.bpo.4-amd64' instead of 'linux-modules-3.16.0-0.bpo.4-amd64'
E: Unable to locate package libzmq4-dev
dpkg-checkbuilddeps: Unmet build dependencies: autoconf (>= 2.63) automake libmodbus-dev (>= 3.0) libncurses-dev libreadline-dev cython (>= 0.19) libxaw7-dev libzmq3-dev (>= 4.0.4) | libzmq4-dev (>= 4.0.4) libczmq-dev (>= 2.2.0) python-zmq (>= 14.0.1) liburiparser-dev uuid-runtime libavahi-client-dev protobuf-compiler (>= 2.4.1) python-protobuf (>= 2.4.1) libtk-img python-pyftpdlib tcl8.6-dev tk8.6-dev
```

ArcEye, 2015-06-26:
There are a shed load of dependencies for building generally on Jessie

I never use the scripts/apt-installbuilddeps

This is the script I use (there may be a couple of double entries)
```
#!/bin/bash

apt-get install -y  build-essential debhelper kernel-package libpth-dev libgtk2.0-dev tcl-dev tk-dev bwidget python-old-doctools python-tk python-dev libglu1-mesa-dev libgnomeprintui2.2-dev \
		    libncurses5-dev libxaw7-dev gettext libreadline-gplv2-dev lyx texlive-extra-utils imagemagick texinfo groff libmodbus-dev;

apt-get install -y  libudev-dev libmodbus-dev libboost-python-dev libboost-serialization-dev libboost-thread-dev libtk-img automake autoconf libtool libusb-dev;

apt-get install -y  automake1.11 libtool liburiparser-dev cmake libssl-dev  openssl python-setuptools  libusb-1.0-0-dev libudev-dev  uuid-dev libavahi-client-dev libavahi-compat-libdnssd-dev \
		     avahi-daemon libprotobuf-dev protobuf-compiler python-protobuf libprotoc-dev uuid-runtime python-avahi python-netifaces avahi-discover;

apt-get install -y  libssl-dev uuid-dev libavahi-client-dev libavahi-common-dev libdbus-1-dev python-pyftpdlib libmodbus-dev libudev-dev libusb1.0-0-dev libglib2.0-dev \
		    libgl1-mesa-dev libxmu-dev python-netifaces liburiparser python-protobuf ;
```

Then manually purge of all *zmq* packages, update apt etc

apt-get install python-zmq libczmq-dev
then
apt-get install libzmq3-dev 
should do it

I have only built on rt and rtai so don't know about xenomai quirks

luminize, 2015-06-26:
I'm building in a VM, only posix threads
after package wrestling I start with compiling and I get as far as this:
```
Linking webtalk
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/4.9/../../../../lib/libwebsockets.a(client.c.o): undefined reference to symbol 'BIO_ctrl@@OPENSSL_1.0.0'
//usr/lib/x86_64-linux-gnu/libcrypto.so.1.0.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
machinetalk/webtalk/Submakefile:53: recipe for target '../bin/webtalk' failed
make: *** [../bin/webtalk] Error 1
make: Leaving directory '/home/bas/machinekit/src'
bas@mbp-jessie:~/machinekit/src$ dpkg -l | grep libssl
ii  libssl-dev:amd64                       1.0.2c-1                                          amd64        Secure Sockets Layer toolkit - development files
ii  libssl-doc                             1.0.2c-1                                          all          Secure Sockets Layer toolkit - development documentation
ii  libssl1.0.0:amd64                      1.0.2c-1                                          amd64        Secure Sockets Layer toolkit - shared libraries

```

ArcEye, 2015-06-26:
The versions are rather strange, I have just done an update and dist-upgrade and you still have newer versions than me
```
ii  libssl-dev:amd64                      1.0.1k-3+deb8u1                         amd64        Secure Sockets Layer toolkit - development files
ii  libssl-doc                            1.0.1k-3+deb8u1                         all          Secure Sockets Layer toolkit - development documentation
ii  libssl1.0.0:amd64                     1.0.1k-3+deb8u1                         amd64        Secure Sockets Layer toolkit - shared libraries
```
Do you have sid enabled?

ArcEye, 2015-06-26:
As I suspected, for some reason you are pulling in sid packages
https://packages.debian.org/sid/libssl-dev

We can look forward to the next jessie ssl package versions stopping working with MK by the sound of it

luminize, 2015-06-26:
I've been botching things. I've been working on VM which I set up for CAN, pythonOCC, ROS etcetera. I'll first do a fresh install of a Jessie VM. Apologies for the inconvenience. If i run into trouble I'll report back.

mhaberler, 2015-06-27:
Robert Nelson kindly posted a Xenomai 3 kernel for the BB (and presumably X15):

````
root@epig:/home/mah# dmesg |grep -i xenomai
[    0.000000] Linux version 3.14.45-ti-xenomai-r69 (root@a6-imx6q-wandboard-2gb) (gcc version 4.6.3 (Debian 4.6.3-14) ) #1 SMP PREEMPT Fri Jun 26 01:04:36 UTC 2015
[    2.724259] [Xenomai] scheduling class idle registered.
[    2.724280] [Xenomai] scheduling class rt registered.
[    2.724482] I-pipe: head domain Xenomai registered.
[    2.731982] [Xenomai] Cobalt v3.0-rc4 (Exact Zero) [DEBUG]
[    3.218876] usb usb1: Manufacturer: Linux 3.14.45-ti-xenomai-r69 musb-hcd
[ 1823.793598] CPU: 0 PID: 4557 Comm: sshd Not tainted 3.14.45-ti-xenomai-r69 #1
````

I guess this is our first Xenomai3 BB kernel - thanks, Robert!

So far we used Xenomai 2.6.x patches - this gives us an excellent opportunity to fix up loose ends for Xenomai 3 - first notes:

- we will have to patch up kernel autodetection a bit, right now it is autodetected as flavor 'posix':
- we need to check if we need a new userland libxenomai library as well
- /dev/rtheap is history in xenomai 3: https://xenomai.org/pipermail/xenomai-git/2014-September/003456.html so the code breaks here:
https://github.com/machinekit/machinekit/blob/5a1a3241aa643053bb2275a3d97581fd38eb9979/src/rtapi/rtapi_compat.c#L69
- xenomai_gid needs love: https://github.com/machinekit/machinekit/blob/5a1a3241aa643053bb2275a3d97581fd38eb9979/src/rtapi/rtapi_kdetect.h#L80

I'll check what needs to be done, I guess a stronger xenomai2/xenomai3 distinction in the autodetect code might help downstream

also for consideration - we might introduce a xenomai3 flavor (not sure yet about this)?

[114 more replies]

closed 2018-08-01 by ArcEye