machinekit/machinekit #601: delta calibration problems

gorf25, 2015-04-30:
Bas De Bruijn asked me to open a new issue and to refer to this 
google groups posting

lineardetla calibration 
i have not been able to calibrate the printer

the link to group

https://groups.google.com/forum/#!topic/machinekit/MRTmRa3saX0

thanks gary

luminize, 2015-05-02:
I experienced the following.
While the SEARCH_HOME_VEL was positive, while homing the direction was negative.
* meaning negative in motion (down)
* as well as negative in commands (the positions in the screen increased negatively, which match with going down)

Negating the HOME_SEARCH_VEL and HOME_LATCH_VEL to a `-` and a `+` which should be incorrect, but results in the right homing direction (up)

Then the next day, when I started up the config, the homing still went down. But now that the HOME_SEARCH_VEL in the ini file is negative, this is to be expected. So setting these to the correct values again `+` search_vel and `-` latch_vel gives the correct behaviour.

My system consists of a BBB rev C, BeBoPr-Bridge, recent build with commit 6ca42cfeb3de5c770041eb0c53a0fc931d7a3313

luminize, 2015-05-02:
pressed wrong button

strahlex, 2015-05-02:
Does the build-plate move or the printing head? Sometimes the direction just looks wrong.

gorf25, 2015-05-02:
the print head its a kossel delta printer

luminize, 2015-05-03:
@gorf25 can you verify (and only this, don't look at scale and direction) that:
* HOME_SEARCH_VEL in your ini file is positive, like +10
* HOME_LATCH_VEL in your ini file is negative, like -1
* *while* homing the joints shows the positions of the joints being positive and increasing. like screenshow below, yellow box.

[image]

gorf25, 2015-05-03:
HOME_SEARCH_VEL in your ini file is positive, like +10
HOME_LATCH_VEL in your ini file is negative, like -1
while homing the joints shows the negative pos and moves towards the bed 
numbers decreasing,

HOME_SEARCH_VEL in your ini file is positive, like -10
HOME_LATCH_VEL in your ini file is negative, like +1
while homing the joints shows the positive pos and moves towards home joints being positive and
increasing like your picture

luminize, 2015-05-03:
well, that's obviously wrong behaviour IMO

luminize, 2015-06-07:
I captured a situation where the search velocity is plain wrong. This time in 2 of the 3 joints. All should go up as per the INI file.

[image]

- I did a compile last week, from 0febbc7cf0934c287a7842bb57f19afdaecea92c
- printed w/o problems
- have had a long power down

for all joints I have these ini values.
```ini
HOME_SEARCH_VEL = 30
HOOME_LATCH_VEL = -1
```

luminize, 2015-06-07:
next step I did:
changed the for the joints 0 and 2 the following:
```ini
HOME_SEARCH_VEL = -30
HOME_LATCH_VEL = 1
```
- pressing the home button moves all the joints up, with positive numbers in the preview window
- then homing didn't stop at the endswitch
- power down
- e-stop
- e-stop clear
- power up
and then the joints 0 and 2 start moving down when I re-home.

- exit Machinekit
- change the ini values again to the correct ones
```ini
HOME_SEARCH_VEL = 30
HOME_LATCH_VEL = -1
```
- start Machinekit
- homing works as it should

I'm flabbergasted.

Will report back if I see something which triggers this.

darenschwenke, 2015-06-08:
verified your end stop signals are good before you started homing?
Noticing you have ~30:1 difference there.

luminize, 2015-06-08:
good observation! If that were the problem it would be an embarrassing one :) I moved my board, and come to think of it, these exact joints were disconnected. I'll have to do some more rewiring so I'll be extra careful to check the endstops.

gorf25, 2015-06-10:
I tried to duplicate what you have done to fix but haven't been successful still have to have set HOME_SEARCH_VEL = -30

closed 2018-08-04 by ArcEye