machinekit/machinekit #607: hal_bb_gpio missing from BBB Xenomai packages

strahlex, 2015-05-02:
the `hal_bb_gpio` HAL component shared object seems to be missing from the current version of the Debian packages.

Looking at the buildbot output I found this error message, not sure it is related:
```
../scripts/check-build-vs-configure-sha: line 11: not: command not found
```

I also noticed that the `configure` step is run without the `--with-platform-beaglebone` argument. However, I tried to build machinekit from source without this option and the missing component was still created.

strahlex, 2015-05-02:
Looking at the buildbot output I cannot find that the component is linked so this seems to be a problem with build not the packaging itself. Has anything with the build infrastructure changed lately?

http://buildbot.dovetail-automata.com/builders/d7-bb-pkg/builds/750/steps/chroot-build-binary-package/logs/stdio

strahlex, 2015-05-02:
I think I found the problem with the `--build=arm-linux-gnueabihf` argument configure seems to disable the beaglebone libraries: `checking platform-beaglebone... disabled for this architecture`

strahlex, 2015-05-02:
The solution to fix this problem is explicitely adding `--with-platform-beaglebone` to the ./configure command. I guess the `--build` and `--host` arguments where added recently to the switch to the docker infrastructure.

zultron, 2015-05-02:
The Jessie changes include multi-arch changes that revealed weakness in #257.  #608 isn't a very good fix; it really should run those checks in one place and set a variable.  I figure we're going to have to address this more fully later, though, so good enough for now....

commit 09f7e2b6 (2015-05-02): configure.ac:  account for Debian multi-arch

When running `./configure` without `--host` and `--build`, the
`AC_CANONICAL_*` macros run `config.guess` to determine host and build
architectures.

Commit ccf3bc98b from #594 sets `--host` and `--build` to Debian
multi-arch names from variables defined in
`/usr/share/dpkg/architecture.mk` during package builds.

This exemplifies the difference:

    # ./config.guess
    armv7l-unknown-linux-gnueabihf
    # dpkg-architecture -qDEB_BUILD_MULTIARCH
    arm-linux-gnueabihf

This patch is a quick fix for platform detection errors in package
builds, an [... cut]
closed 2015-05-02 by zultron