machinekit/machinekit #610: HAL C API: deprecate less-than-useful params, aliases

mhaberler, 2015-05-03:
The HAL C API provides two features with dubious usefulness:

- HAL params: very similar to pins, except they cannot be linked to.
- HAL name aliases: an obscure and unused feature to provide an alias name for a pin or a param. The only usage is in a demo HALTcl script: sim/axis/simtcl/twopass_demo.tcl

Params are supported by the legacy hal module, but aliases cannot be created by it. Due do how halpr_find_pin_by_name() works, they can be looked up. One of the reasons for params - they can have non-default initial values - has gone away since pins can have that too. Many comps using params have been rewritten piecemeal to used pins instead (eg pid - the P/I/D gains) to no ill effect.

Both params and aliases are not supported by the Cython bindings and haltalk for that matter, and so far nobody complained.

I propose to deprecate params as well as aliases, and to go about it as follows:
- the code for aliases is safe to remove. It needs to be removed from the docs as well.
- params: we leave in the supporting code in, but add a deprecation warning at the C/C++/comp/instcomp tools.
- when a component using params is in need for interacting with the Cython bindings or haltalk, it should be rewritten to use a pin instead.
- eventually the code for params can be disabled.

I'm willing to take this on and mutate params into pins as far as they are exported by the HAL C library, but would like to see consensus beforehand

strahlex, 2015-05-03:
So I am the first one to complain about that the Cython bindings cannot write params.

At the moment I am working on implementing the Halanduino project only using the Cython API. So far I found these two problems:
- `hal_pru_generic` and `hal_bb_gpio` use params so I can't set them easily from Python
- `pid` is not yet instantiable 

So I am all in favor of deprecating the HAL params. I see no use either compared to HAL pins. However, I would suggest replacing them with I/O pins so default values will not be overwritten when bound to a signal.

RunningLight, 2015-05-03:
`alias` always struck me as a solution in search of a problem, easier to implement than to justify. The only time I used it, other than to test that it worked at some point in the life of halcmd, was to interactively overlay a hal file with a running configuration which named pins differently and I was too lazy to edit the file first. An `alias` of an `alias` (of an `alias` ...) is a nightmarish concept even if it is explicitly mentioned in the docs. I don't believe this feature will be missed by users.

I see no reason not to deprecate param for the reasons you have stated. Changing the existing mindset is a bigger challenge than changing the existing code.

mhaberler, 2015-05-05:
ok, i will #ifdef out the aliases code out for now such it can be turned on just in case

for params we might add a warning to comp/instcomp suggesting the pin alternative

in both cases I will add a conditional __attribute__(deprecated) to the underlying data structures so offending code can be easily identified except if configured with --disable-deprecation-warnings

ArcEye, 2015-06-22:
Have a look at
https://github.com/ArcEye/machinekit/commit/1f605ba4beb60a0ad6f6d6f038eeebebda3c4532

params: I have changed 
`param rw` to `pin io`
`param r` to `pin out` (since it is presumably just there for debugging purposes)

variables:
**float** or double become `hal_float_t`
**int** defaults to hal_s32_t, unless using it to hold large number (eg round over a float) or where cannot hold a neg value, then hal_u32_t
**char** occured in 3 places and replaced with hal_bit_t, as that was what it was holding or being compared to

instcomp still uses params but warns where they occur that they should be deprecated

Not done a PR yet until another pair or two of eyes make sure there are no gotchas

Only downside is my editor has changed tabs to spaces, resulting in many more changes than intended.;)

ArcEye, 2015-06-22:
Found a few more and trying to fix the white space issues, commit reset for a bit

ArcEye, 2015-06-22:
Now at
https://github.com/ArcEye/machinekit/commit/bc5b64d58d715f3da7b9248c2687dc61e51b3884
with whitespace changes squashed

commit d31205c3 (2015-06-26): Merge pull request #694 from ArcEye/master

Fix unsafe type usages - close #663 & use of params - close #610
closed 2015-06-26 by strahlex