machinekit/machinekit #611: IPV6 support missing in haltalk, msgd, webtalk

mhaberler, 2015-05-03:
currently those daemons bind to IPv4 addresses only if
this should be fixed so by default binding would happen ob IPv4 and IPv6

strahlex, 2015-05-03:
This is problem because in mixed IPv4/IPv6 networks service announcement defaults to IPv6 (seems to be an Bonjour/MDNS thing). Therefore, hostname resolution may also return IPv6 addresses and client connection may fail.

mhaberler, 2015-05-04:
here is a first cut, please test if this works for you:

https://github.com/mhaberler/machinekit/tree/ipv6

koppi, 2015-05-04:
Tested, it works for me.

mhaberler, 2015-05-04:
tested - how? using ipv4, or v6?

koppi, 2015-05-04:
Tested the new default binding for IPv4 and IPv6.

```ini
# IPv6 support - default on
# disable by uncommenting next line
IPV6=0
```

Both connecting via IPv4 and IPv6 works for me.

strahlex, 2015-05-04:
I suggest an additinal parameter to enable/disable IPv4

Also bind should use any interface if `REMOTE` is set 1 (bind with * ) so it works for network interfaces started afterwards too. I am not sure if the `INTERFACES` ini parameter is required anymore. 

Avahi has following behaviour:
- `PROTO_UNSPEC` creates IPv4 and IPv6 announcements
- `PROTO_INET``creates IPv4 announcements
- `PROTO_INET6` creates IPv6 announcements
default should be announcements on both IPv4 and IPv6 interfaces

Furthermore, an additional txtRecord or something similar is required to indicated wether IPv6 is supported or not at the client. This is necessary since the client has the opportunity to resolve the `.local` hostname to either an IPv4 (type A DNS request) or an IPv6 (type AAAA DNS request) address. In case IPv6 is supported IPv6 is will be preferred at the client side.

mhaberler, 2015-05-05:
@koppi: actually if you have the line 'IPV6=0' _uncommented_ as you show it above the behavior should be exactly as before

if it is _commented out_ announcements and binding of sockets should entail ipv6 as well. The sockets will listen on mapped v6 addresses, which is the default behavior of zeromq on linux if the ZMQ_IPV6 option is set on a socket.

mhaberler, 2015-05-05:
@strahlex  - could well be, I think the INTERFACES= idea was not that great afterall and I did not understand the binding behavior that well back then

we can add a txt record; if we do it we might just as well go all the way and cover all options for conveying a preference:

- IPV4 only
- IPV6 only
- both, preference for IPV4 
- both, preference for IPV6
so that could be something like 'TXT=<key> <stringvalue>

I'll think a bit how this API might be generalized a bit to add arbitrary TXT values, maybe on a per-service level - you never know

since avahi is a bit tricky to work with, probably the announcement directive (PROTO_UNSPEC/PROTO_INET/PROTO_INET6) should be a separate knob in the ini file at least for now, until we know better and maybe can simplify things again

strahlex, 2015-05-05:
Or we use the type of service announcement (the host address of the service announcer tells the client wether it is IPv4 or IPv6) to select wether to use IPv4 or IPv6. However, this requires that the announcements are done correctly (IPv4 for IPv4 ZMQ sockets, ...).

So in case the client operates IPv4 only it will not receive service announcements that it can't connect to.

mhaberler, 2015-05-10:
just a note on OSX Yosemite & IPv6:

resolving .local names (eg slogin user@host.local) is disabled by default

this helps:

 sudo discoveryutil mdnsactivedirectory yes

found at: https://discussions.apple.com/message/26927843#26927843

mhaberler, 2015-05-10:
I'm about to rewrite how zeromq socket binding and mDNS announcements are configured. 

the idea is to completely separate port selection, interface/address selection, and mDNS announcements.

This is how it will look like:

````
# enable remote service access - defaults to local; set to 1 for enabling remote operation
#
# REMOTE=1 means: zeroMQ sockets will use TCP as specified below for address and port
# REMOTE=0 means: zeroMQ will use IPC sockets in RUNDIR/<rtapi_instance>.<service>.<uuid>
#
REMOTE=1

# -------------- network setup -----------------
# only relevant when using TCP, not needed in local case (REMOTE=0)

# If REMOTE=0, services are bound to Unix IPC sockets, meaning the service cannot
# be reached from outside the current host.

# -------------- TCP Port selection -----------------
#
# this is the default for group status (STP) - ephemeral port (next free port number)
#GROUP_STATUS_PORT=*
# alternatively, bind to a specific port number like so:
#GROUP_STATUS_PORT=6200

# default for remote component status protocol - ephemeral port
#RCOMP_STATUS_PORT=*
#RCOMP_STATUS_PORT=6201  # specific port number

# default for remote command interaction - ephemeral port
#COMMAND_PORT=*
#COMMAND_PORT=6202

# default for rtapi msgd log service - ephemeral port:
#LOG_PORT=*
#LOG_PORT=6203  # specific port number

# -------------- interface and protocol selection -----------------
#
# binding zeroMQ sockets and mDNS announcements are conceptually separate,
# but of course better add up.
# the default is to bind on in6addr_any (ipv4 and mapped ipv6 on Linux).
#
# The binding algorithm is as follows:
#
# Bind any addresses or interfaces in the BIND_IPV4 list (none if empty).
# if a port number was explicitly specified above, use that port
# else bind as ephemeral port, retrieve that assigned port number, and use
# that in subsequent bind operations.
#
# if the BIND_IPV6 list is not empty, set the zeromq ZMQ_IPV6 socket option,
# then, for each interface/address in BIND_IPV6:
#   if a port number was already assigned (eg by explicit port number setting or
#   through an ephemeral binding of an IPv4 interface/address, use that.
#   bind the current interface/address
#
# default binding spec is equivalent to:
#BIND_IPV4=
#BIND_IPV6=*

# mDNS announcement - protocol selection:
#ANNOUNCE_IPV4=0 and ANNOUNCE_IPV6=0: suppress any Avahi announcements.
#ANNOUNCE_IPV4=1 and ANNOUNCE_IPV6=0: use AVAHI_PROTO_INET
#ANNOUNCE_IPV4=1 and ANNOUNCE_IPV6=1: use AVAHI_PROTO_UNSPEC
#ANNOUNCE_IPV4=0 and ANNOUNCE_IPV6=1: use AVAHI_PROTO_INET6

#defaults are:
#ANNOUNCE_IPV4=1
#ANNOUNCE_IPV6=1
````

strahlex, 2015-05-10:
Is it possible to bind a ZMQ socket to more than one interface?

mhaberler, 2015-05-10:
it is possible to execute several bind operations on a socket, and mix wildcard and specific binding like so:
````
socket.bind("tcp://*:5500")      
socket.setsockopt(zmq.IPV6, 1)
socket.bind("tcp://[fd61:b29a:8a73:99cd:a00:27ff:fef0:6219]:5500")
````
this yields 
````
mah@nwheezy:~$ ss -an|grep 5500
LISTEN     0      100      fd61:b29a:8a73:99cd:a00:27ff:fef0:6219:5500                    :::*     
LISTEN     0      100                       *:5500                     *:*     
````

also, the point where the ZMQ_IPV6 option is set is significant, if you do this:
````
socket.setsockopt(zmq.IPV6, 1)
socket.bind("tcp://*:5500")      
socket.bind("tcp://[fd61:b29a:8a73:99cd:a00:27ff:fef0:6219]:5500")
````
this yields 
````
zmq.error.ZMQError: Address already in use
````

However,
````
socket.setsockopt(zmq.IPV6, 1)
socket.bind("tcp://127.0.0.1:5500")
socket.bind("tcp://193.228.47.221:5500")
socket.bind("tcp://[fd61:b29a:8a73:99cd:a00:27ff:fef0:6219]:5500")
````
yields 
````
mah@nwheezy:~$ ss -an|grep 5500
LISTEN     0      100      fd61:b29a:8a73:99cd:a00:27ff:fef0:6219:5500                    :::*     
LISTEN     0      100       ::ffff:193.228.47.221:5500                    :::*     
LISTEN     0      100        ::ffff:127.0.0.1:5500                    :::*     
````

(note mapped v6 addresses)

that's why I split into BIND_IPV4 and BIND_IPV6 address groups with an implicit ZMQ_IPV6 option between them - this is the most general solution I could find

mhaberler, 2015-05-14:
I have a working branch which implements the above scheme for haltalk and rtapi_msgd, webtalk still needs a bit of work but that can wait

https://github.com/mhaberler/machinekit/commits/new-binding

This works for me at least in the gladevcp case 

HOWEVER I observed one situation where avahi had renamed the host (ps -ax showed avahi announcing nwheezy-14.local), the library code would announce URI's using nwheezy.local and client resolution for this name failed

now the mk_service.cc library code uses gethostname() and this could be wrong, but I have not found out yet how to retrieve a renamed hostname as set  by avahi, which probably is the better solution

I could not reproduce the host renaming effect after restarting avahi-daemon - any hints welcome as to what is going on

that said, the situation is as before since we're announcing "<gethostbyname()>.local." already

I would appreciate some testing and a second opinion before going for cleanup/a PR

mhaberler, 2015-05-14:
update: seems the proper way is to use avahi_server_get_host_name_fqdn() to retrieve the hostname, instead of constructing our own by using gethostname() and appending ".local."

avahi_server_get_host_name_fqdn() docs says:
"Return the currently chosen host name as a FQDN ("fully qualified domain name", i.e. the concatenation of the host and domain name)."

http://avahi.sourcearchive.com/documentation/0.6.25-1ubuntu3/core_8h_073972d64569e2e26a45f1a08fecf748.html

mhaberler, 2015-05-15:
ok, the preview branch is here, it implements the new behavior as outlined above and should also properly deal with avahi hostname renaming: https://github.com/mhaberler/machinekit/tree/new-binding

it's good for testing; I'll prepare a PR shortly

mhaberler, 2015-05-22:
fixed in #637

closed 2015-05-22 by mhaberler