machinekit/machinekit #613: error - cp: cannot copy cyclic symbolic link - during "make"

koppi, 2015-05-05:
I'm getting errors during "make":

```
koppi@x220:~/Desktop/machinekit/src$ ./configure 
checking for cython... /usr/bin/cython
checking cython version... 0.21.1
checking build toplevel... /home/koppi/Desktop/machinekit
checking installation prefix... run in place
checking for grep... /bin/grep
checking for egrep... /bin/egrep
checking for arch... /usr/bin/arch
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking how to run the C preprocessor... gcc -E
checking for objcopy... objcopy
checking for ld... ld
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking for pkg-config... pkg-config
checking platform-pc... default enabled for this architecture
checking platform-beaglebone... disabled for this architecture
checking platform-raspberry... disabled for this architecture
checking platform-zedboard... disabled for this architecture
checking for uuidgen... /usr/bin/uuidgen
setting unique Machinekit UUID to e9a48114-9593-4f2d-af16-6a44d85f7bcf
checking whether to enable remote operation... local operation on IPC sockets only, zeroconf not used
checking which directory to use for IPC sockets and other transient files... using /tmp for IPC sockets and other transient files
checking whether to build programming examples... not building example programs
checking for proto2js... none
checking whether to build protobuf Javascript bindings... not protobuf Javascript bindings
checking whether to use the common shared memory driver... default is no
checking whether to use inb/outb or ppdev ioctl on the x86 for parport I/O... using defaults: no
checking whether to build the emcweb interface... not building the emcweb interface
checking whether to build unstable development code... not building  development code
checking pkg-config is at least version 0.9.0... yes
checking for CZMQ... yes
checking for PROTOBUF... yes
checking for protoc... /usr/bin/protoc
checking python module: google.protobuf.descriptor... yes
checking python module: google.protobuf.descriptor_pb2... yes
checking python module: google.protobuf.compiler.plugin_pb2... yes
checking for JANSSON... yes
checking for URIPARSER... yes
checking for LWS... yes
checking for SSL... yes
checking for UUID... yes
checking for AVAHI... yes
checking whether to enable NML-using parts... not building non-essential NML-using parts
checking python module: pyftpdlib.authorizers... yes
checking python module: pyftpdlib.handlers... mayes
checking python module: pyftpdlib.servers... yes
checking for configured kernel source in /usr/src/linux-headers-3.18.12-xenomai-2.6.4... found
checking if /usr/src/linux-headers-3.18.12-xenomai-2.6.4 is a xenomai kernel... yes
checking if /usr/src/linux-headers-3.18.12-xenomai-2.6.4 is an RTAI kernel... no
checking for configured kernel source in /usr/src/linux-headers-3.19.0-14... not found
checking for configured kernel source in /usr/src/linux-headers-3.19.0-16... not found
checking for configured kernel source in /usr/src/linux-headers-3.19.0-16-generic... found
checking if /usr/src/linux-headers-3.19.0-16-generic is a xenomai kernel... no
checking if /usr/src/linux-headers-3.19.0-16-generic is an RTAI kernel... no
checking for configured kernel source in /home/koppi/linux-3.18.12... found
checking if /home/koppi/linux-3.18.12 is a xenomai kernel... yes
checking if /home/koppi/linux-3.18.12 is an RTAI kernel... no
checking -lrt for RT_PREEMPT and POSIX... kok
[... 213 lines cut ...]
Done reading dependencies

mkdir -p objects/xenomai-kernel/3.18.12-xenomai-2.6.4
cp -rlf emc hal libnml rtapi workaround machinetalk objects/xenomai-kernel/3.18.12-xenomai-2.6.4
cp: cannot copy cyclic symbolic link ‘rtapi/posix’
cp: cannot copy cyclic symbolic link ‘rtapi/rt-preempt’
cp: cannot copy cyclic symbolic link ‘rtapi/xenomai’
Makefile:675: recipe for target 'objects/xenomai-kernel/3.18.12-xenomai-2.6.4/.separate-kbuild-artifacts-hack.stamp' failed
make[1]: *** [objects/xenomai-kernel/3.18.12-xenomai-2.6.4/.separate-kbuild-artifacts-hack.stamp] Error 1
make[1]: Leaving directory '/home/koppi/Desktop/machinekit/src'
Makefile:121: recipe for target 'modules' failed
make: *** [modules] Error 2
make: Leaving directory '/home/koppi/Desktop/machinekit/src'
koppi@x220:~/Desktop/machinekit/src$ cp -rlf emc hal libnml rtapi workaround machinetalk objects/xenomai-kernel/3.18.12-xenomai-2.6.4
cp: cannot copy cyclic symbolic link ‘rtapi/posix’
cp: cannot copy cyclic symbolic link ‘rtapi/rt-preempt’
cp: cannot copy cyclic symbolic link ‘rtapi/xenomai’
koppi@x220:~/Desktop/machinekit/src$ ls -dl rtapi/posix
lrwxrwxrwx 1 koppi koppi 1 Mai  5 22:26 rtapi/posix -> .
koppi@x220:~/Desktop/machinekit/src$ ls -dl rtapi/rt-preempt
lrwxrwxrwx 1 koppi koppi 1 Mai  5 22:27 rtapi/rt-preempt -> .
koppi@x220:~/Desktop/machinekit/src$ ls -dl rtapi/xenomai
lrwxrwxrwx 1 koppi koppi 1 Mai  5 22:27 rtapi/xenomai -> .
koppi@x220:~/Desktop/machinekit/src$ 
```

mhaberler, 2015-05-05:
could you please report what kind of linux platform this is? Guessing will not help

koppi, 2015-05-06:
Just toying around on Ubuntu 15.04. Will post an update tomorrow about the findings of my journey.

zultron, 2015-05-06:
On 05/05/2015 03:39 PM, koppi wrote:
> koppi@x220:~/Desktop/machinekit/src$ cp -rlf emc hal libnml rtapi workaround machinetalk objects/xenomai-kernel/3.18.12-xenomai-2.6.4
> cp: cannot copy cyclic symbolic link ‘rtapi/posix’
> cp: cannot copy cyclic symbolic link ‘rtapi/rt-preempt’
> cp: cannot copy cyclic symbolic link ‘rtapi/xenomai’

Can you try to find options to `cp` that might work?  For example,

     cp -rlfd [...]

Curious error; this works fine in Wheezy, Jessie and Trusty.

	John

mhaberler, 2015-05-06:
you will get quickest results by using Debian stable - you are pretty much on your own with other distros.

koppi, 2015-05-09:
@zultron  cp -rlfd worked for me.

koppi, 2015-05-10:
merged.

closed 2015-05-10 by koppi

zultron, 2015-05-11:
@koppi, thanks for PR #617 (adding reference for documentation purposes).