machinekit/machinekit #619: machinekit-dev package missing ccache dependy

strahlex, 2015-05-07:
ccache support was recently added to speed up compiling on the BBB. Now I get the following error on fresh install with machinekit-dev:
´´´
make: ccache: Command not found
´´´
a simple ´sudo apt-get install ccache´ fixes the problem. Adding ccache to the Debian package dependencies might enough.

zultron, 2015-05-08:
Ok, but `ccache` should not be required to build.

Can you include any logs that might show what is requiring `ccache`?

zultron, 2015-05-08:
Are you building packages?  I only see [one place][1] where `ccache` is explicitly mentioned in PR #609, in the `debian/rules` `build` target.

[1]: https://github.com/machinekit/machinekit/pull/609/files#diff-15054b3a95e7be339432e2184473e43aR82

strahlex, 2015-05-08:
No, I just used `comp --install <componen>`  to build and install a component.

zultron, 2015-05-08:
Ah, maybe I know what's going on.  `env CC="ccache gcc" ./configure` stores `CC` in the generated `Makefile.modinc`.  I'll have to think about how to work around that.

cdsteinkuehler, 2015-05-08:
Perhaps use a bash alias to enable ccache instead of defining CC ?

-- 
Charles Steinkuehler
charles@steinkuehler.net

zultron, 2015-05-08:
@cdsteinkuehler, yeah, that's about the right idea.  If I'd bothered to read the `ccache(1)` man page, I'd have seen this:

    [It's] easiest to just prepend /usr/lib/ccache to your PATH. /usr/lib/ccache contains
    symlinks for all compilers currently installed as Debian packages.

This is a Buildbot-only fix, though the related #623 improves `CC`/`CXX` handling better in packages, allowing e.g. `AC_PROG_CC` to select a multi-arch compiler like `x86_64-linux-gnu-gcc` rather than the `make` default `cc`.  :P

zultron, 2015-05-08:
@strahlex, `comp` will invoke a different compiler command from the past after #623 is merged (`arm-linux-gnueabihf-gcc` rather than `gcc`).  Can you test it out after the merge and report results (close this issue if applicable)?

zultron, 2015-06-01:
Assuming this is fixed.

closed 2015-06-01 by zultron