machinekit/machinekit #627: Forward declaration of #typedef , breaks gcc 4.4 ability to build Machinekit

ArcEye, 2015-05-10:
The C11 support in gcc-4.4 is incomplete and starting to cause problems with newer code.
https://gcc.gnu.org/gcc-4.4/cxx0x_status.html

Specifically forward declarations of #typedef tp_shared_t *tp_shared_t 
when building machinetalk produce this error
```
Linking liblinuxcnc-pb2++.so.0
ln -sf liblinuxcnc-pb2++.so.0 ../lib/liblinuxcnc-pb2++.so
Compiling machinetalk/support/sizes.c
In file included from ./emc/motion/motion.h:88,
                 from machinetalk/support/sizes.c:2:
emc/tp/tp.h:22: error: redefinition of typedef ‘tp_shared_t’
emc/tp/tp_types.h:88: note: previous declaration of ‘tp_shared_t’ was here
In file included from ./emc/motion/motion_debug.h:19,
                 from machinetalk/support/sizes.c:3:
emc/tp/tp_shared.h:66: error: redefinition of typedef ‘tp_shared_t’
emc/tp/tp.h:22: note: previous declaration of ‘tp_shared_t’ was here
make: *** [objects/machinetalk/support/sizes.o] Error 1
make: Leaving directory `/usr/src/machinekit-arceye/src'
```

This can be got around by commenting out the declarations in tp.h and tp_types.h and inserting
#include tp_types.h prior to tp.h in motion.h, so that the full type is declared before usage.

However the next error is the same sort, just a different typedef

```
In file included from hal/lib/hal_lib.c:63:
hal/lib/hal_priv.h:437: error: redefinition of typedef ‘hal_funct_t’
hal/lib/hal_priv.h:376: note: previous declaration of ‘hal_funct_t’ was here
hal/lib/hal_priv.h:459: error: redefinition of typedef ‘hal_thread_t’
hal/lib/hal_priv.h:375: note: previous declaration of ‘hal_thread_t’ was here
make: *** [objects/hal/lib/hal_lib.o] Error 1
make: Leaving directory `/usr/src/machinekit-arceye/src
```

This was not so easy to sidestep and at that point I realised what was likely to be going on.

Installed gcc/g++/cpp 4.6 and the same code built without issues

As will be seen, 4.6 has better implementation of C11
https://gcc.gnu.org/gcc-4.6/cxx0x_status.html

but only 4.8 has almost complete implementation
https://gcc.gnu.org/gcc-4.8/cxx0x_status.html

Suggest we don't 'dumb down' the code, just change configure to require >= gcc-4.6
which should be available for all supported Wheezy and Ubuntu variants

strahlex, 2015-05-10:
This raises the question: Do we need to support GCC 4.4? Which distribution ships with an compiler that old?

Adding GCC >= 4.6 to the dependency sounds legit to me.

ArcEye, 2015-05-10:
It obviously came with Wheezy 7 beta on the i386 distro, I would not have installed it specially.

I agree, I don't think we need to support it, just ensure it is not used.

mhaberler, 2015-05-10:
agreed - not needed, let's require gcc 4.6 minimum

fixing this is not obvious because these forward typedefs are required if one does mutually dependent datastructure definitions

RunningLight, 2015-05-10:
Seems more than safe enough

A quick perusal of my hosts and the web-based repo search engines shows

Linux Repo | gcc 4.4 | gcc 4.6 | gcc 4.7 | gcc 4.8 | gcc 4.9 | gcc 5.0 
--------------- | :---: | :---:   | :---:  | :---:  | :---:  | :---:
Debian 7 / Wheezy| 4.4.7 | 4.6.3 | 4.7.2 
Debian 8 / Jessie / Stable| | | | 4.8.4 | 4.9.2 
Ubuntu 14.04.2 LTS / Trusty | | | | 4.8.1 
Ubuntu 14.10 / Utopic | | | | | 4.9.1
Ubuntu 15.04 / Vivid | | | | | 4.9.2
Linux Mint 16/Petra [1] | | | | 4.8.1 
Linux Mint 17.1 Rebecca [2] | | | | 4.8.2 | 4.9.1

[1] based on Ubuntu 13.1
[2] based on Ubuntu 14.04

Don't know when Wheezy stopped installing 4.4, but on a freshly built Wheezy VM configured ala Building From Source,

```
$ gcc --version
gcc (Debian 4.7.2-5) 4.7.2
```

ArcEye, 2015-05-14:
Issue resolved as part of a larger series of commits to a development branch of @mhaberler 

Will be merged in due course when branch finalised

closed 2015-05-14 by ArcEye