machinekit/machinekit #631: Ethercat Integration with MK missing

jpmoniz, 2015-05-12:
Opening issue to capture work focused towards solving this.

koppi, 2015-05-13:
subscribed!

mhaberler, 2015-05-18:
I got delivery of a (nanotec N5 EtherCAT drive)[http://de.nanotec.com/produkte/1597-n5-motorsteuerung-fuer-ethercat-oder-canopen/] for testing.

amitgoradia, 2015-05-28:
I got delivery of a Panasonic A5B ethercat servo drive (http://industrial.panasonic.com/ww/products/motors-compressors/fa-motors/ac-servo-motors/minas-a5b) for testing. Have to start working on it.

jpmoniz, 2015-05-28:
Hopefully I will be back at it, in a week or two.  In the middle of a house move right now.  I'll be working with a control techniques M700 drive.  Looks like we have a good mix of different devices.

claudiolorini, 2015-05-28:
...ok, i've got a Beckhof AX5106 on my desk, some digital IO modules and a Ethercat/CANOpen
gateway, if there is something to test, i'm in!
(not sure if the Zedboard Eth phisical is compatible with IgH Etherlab but i think i can buy a BBB)

claudiolorini, 2015-08-07:
...some small steps ahead:
I've managed following @koppi 's guide to activate fast digital outputs (EL2202) from Machinekit, adding the few lines of code needed for support and changing the configuration xml.... but:
Trying to do the same with the fast digital inputs module (EL1252) i obtain the following error
```
sudo tail -f /var/log/kern.log | grep EtherCAT
...
Aug  7 13:10:48 PC104 kernel: [14528.822109] EtherCAT ERROR 0 0:1: PDO entry 0x6010:01 is not mapped.
Aug  7 13:10:48 PC104 kernel: [14528.822477] EtherCAT 0: Releasing master...
...
```

the PDOs that the devices is showing are the following:

```
ethercat pdos -p1 

SM0: PhysAddr 0x1000, DefaultSize    1, ControlRegister 0x22, Enable 1
  TxPDO 0x1a00 "Channel 1"
    PDO entry 0x6000:01,  1 bit, "Input"
  TxPDO 0x1a01 "Channel 2"
    PDO entry 0x6000:02,  1 bit, "Input"
    PDO entry 0x0000:00,  6 bit, ""
  TxPDO 0x1a02 "Reserved"
SM1: PhysAddr 0x09ae, DefaultSize    0, ControlRegister 0x00, Enable 4
  TxPDO 0x1a13 "Latch"
    PDO entry 0x1d09:ae,  8 bit, "Status1"
    PDO entry 0x1d09:af,  8 bit, "Status2"
    PDO entry 0x1d09:b0, 64 bit, "LatchPos1"
    PDO entry 0x1d09:b8, 64 bit, "LatchNeg1"
    PDO entry 0x1d09:c0, 64 bit, "LatchPos2"
    PDO entry 0x1d09:c8, 64 bit, "LatchNeg2"
SM2: PhysAddr 0x0910, DefaultSize    0, ControlRegister 0x00, Enable 4
```
afaict the master is requesting the PDO 0x6010 that the device does not provide.... is it correct?

I've added the following to the @sittner source:

lcec_el1xxx.h:
```
...
#define LCEC_EL1252_PID 0x04E43052
...
#define LCEC_EL1252_PDOS 2
...
```
lcec_main.c:
```
...
  { lcecSlaveTypeEL1252, LCEC_EL1xxx_VID, LCEC_EL1252_PID, LCEC_EL1252_PDOS, lcec_el1xxx_init},
...
```
lcec_conf.h:
```
...
  lcecSlaveTypeEL1252,
...
```
lcec_conf.c:
```
...
  { "EL1252", lcecSlaveTypeEL1252 },
...
```

...any suggestion?

koppi, 2015-08-07:
@claudiolorini 3 months ago I had a similar issue with the PDO mapping for the EL7041-1000 device driver, see http://etherlab-users.etherlab.narkive.com/eqQWt2Pw/el7041-1000-al-status-message-0x001e-invalid-input-configuration

Run ``` ethercat -p1 cstruct``` and compare it's output to the PDO entries defined in ```LCEC_EL1252_PDOS```. The message in ```/var/log/kern.log``` suggests that the PDO structure defined in ```LCEC_EL1252_PDOS``` is different to the PDO structure reported by the device.

koppi, 2015-08-07:
Sorry, my above comment was not quite correct: you defined ```LCEC_EL1252_PDOS``` to be ```2```.

This is the line, where the PDO mapping happens: https://github.com/sittner/linuxcnc-ethercat/blob/master/src/lcec_el1xxx.c#L52 in the driver from what you report. In your case the addresses are ```0x6000:01``` and ```0x6000:02```.

claudiolorini, 2015-08-11:
...I've seen the code and i understand better the things now, i think i have to write two specific 'drivers' for EL2202 (has a slightly different behaviour from the generic EL2xxx) and EL1252 (that has a completly different set of PDOs from EL2xxx)! 
thanks, C.

claudiolorini, 2015-08-24:
...ok, i managed to put to work the IO terminals (EL2202 and EL1252), if someone is interested i've made a PR to @sittner, otherwise the code can be accessed here: 
https://github.com/claudiolorini/linuxcnc-ethercat

Next step is to operate the EL6751 CANopen Master with the Elmo Harmonica CANOpen drives.

dbskccc, 2015-11-30:
hello, is somobody try this on beaglebone black,  i am just lost on it.

AceXIE, 2016-04-18:
Hi, everyone.
Are you successed? Now, I am try this on my BBB. I use the .img, so there is no problem running machinekit. However, when I try etherlab master on BBB, this is some errors. Do you have some idea on how to make.
Thanks.

sirop, 2016-04-18:
@AceXIE 
Could you please share your error logs?

My suspicion about ethercat on BBB is: linuxcnc-ethercat depends on https://sourceforge.net/p/etherlabmaster/ which in its turn depends on adopted network drivers
https://sourceforge.net/p/etherlabmaster/code/ci/default/tree/devices/ .

You may still use the generic network driver from /default/tree/devices/ but it is not the fastest.
Thus - even if you configured everything properly - it is not improbable that your master will complain about skipped datagrams...

AceXIE, 2016-04-18:
Dear  @sirop.
Thanks a lot.
I follow the tutorial [tutorial](https://docs.google.com/document/d/1GiB065ZIAaoMHPtVfTg9JV1Kn-19xGQl2X9DM9-THNM/edit#)

```
1st Download ethercat master from IgH
2nd Untar the file: tar xvfj ethercat-1.5.2.tar.bz2
3rd download debian folder from sittner
4th copy debian folder (3rd step) to /ethercat-1.5.2/ (2nd step)
5th sittner instructions:
$cd ethercat-1.5.2
$ cd debian
$ ./configure -a _//This step, I use ./configure -r, because my kernel version is 3.8.13._
$ cd ..
$ dpkg-checkbuilddeps (and install missing deps)
$ dpkg-buildpackage
```
There is no error log before this.
The error logs is as follows.

```
/usr/bin/ld: error: ethercat uses VFP register arguments, ../master/soe_errors.o does not
/usr/bin/ld: failed to merge target specific data of file ../master/soe_errors.o
collect2: ld returned 1 exit status
make[3]: *** [ethercat] Error 1
make[3]: Leaving directory `/home/machinekit/ethercat-1.5.2/tool'
make[2]: *** [install-recursive] Error 1
make[2]: Leaving directory `/home/machinekit/ethercat-1.5.2'
make[1]: *** [install] Error 2
make[1]: Leaving directory `/home/machinekit/ethercat-1.5.2'
make: *** [install] Error 2
dpkg-buildpackage: error: debian/rules binary gave error exit status 2

```
It seems that BBB is hard float, I guess.
My kernel version is
```
cat /proc/version 
Linux version 3.8.13-xenomai-r78 (root@a6-imx6q-wandboard-2gb) (gcc version 4.6.3 (Debian 4.6.3-14) ) #1 Sat Sep 26 16:54:43 UTC 2015
```
My gcc version is
```
machinekit@beaglebone:~/ethercat-1.5.2$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/4.6/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Debian 4.6.3-14' --with-bugurl=file:///usr/share/doc/gcc-4.6/README.Bugs --enable-languages=c,c++,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.6 --enable-shared --enable-linker-build-id --with-system-zlib --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.6 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --enable-plugin --enable-objc-gc --disable-sjlj-exceptions --with-arch=armv7-a --with-fpu=vfpv3-d16 --with-float=hard --with-mode=thumb --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 4.6.3 (Debian 4.6.3-14) 
```
Do you have some suggestions? 
When I just install the etherlab, there is no error. And, I can start the EtherCAT Master, and ``` ethercat slaves ``` can also receive my slaves. I don't know how to connect machinekit to Etherlab Master. I want to  control my drivers via EtherCAT. I am complete beginner to ARM, Could you help me.
Thanks.

sirop, 2016-04-18:
@AceXIE 
I do not know if the usual stackoverflow search would help about
`/usr/bin/ld: error: ethercat uses VFP register arguments, ../master/soe_errors.o does not`
http://stackoverflow.com/q/9753749/4599792 ?

However  you say:
>When I just install the etherlab, there is no error. And, I can start the EtherCAT Master, and ethercat >slaves can also receive my slaves.  

It means you can run Etherlab on the BBB but you did not install it like in https://github.com/sittner/ec-debianize/ ?

If you follow this newer tutorial https://github.com/koppi/mk/blob/daa3fd64a0b2b61933cdda100c8679dcee8232cb/Machinekit-Xenomai-Thinkpad-X200.md#installation-und-konfiguration-ethercat-master , then
examine and adopt  https://github.com/sittner/ec-debianize/blob/master/debian/extras/usr/sbin/update-ethercat-config  .

If you find out that Distributed Clock handling is not fine grained enough within https://github.com/sittner/linuxcnc-ethercat , then you can try to code your own MK component
for the very slaves you use like:
https://github.com/sirop/EtherCAT_HAL/blob/master/pdortapi2dc.comp

AceXIE, 2016-04-18:
@sirop 
Thank you very much.

> When I just install the etherlab, there is no error. And, I can start the EtherCAT Master, and ethercat >slaves can also receive my slaves.

I mean the EtherCAT master can work well when I install following the instruction of [Etherlab.](http://www.etherlab.org/download/ethercat/ethercat-1.5.2.pdf).
I will have a try, as [this](http://stackoverflow.com/questions/9753749/arm-compilation-error-vpf-registered-used-by-executable-not-object-file).
By the way, how to use .comp file. I can't find the ```comp``` command. Which package I forget to install on debian?

mhaberler, 2016-04-18:
@AceXIE compile everyhing non-machinekit for armhf. Mixing softftp and hardfp doesnt work well.

sirop, 2016-04-18:
>I mean the EtherCAT master can work well when I install following the instruction of Etherlab.   

If you still want to install it this way https://github.com/sittner/ec-debianize ,
try to change the configure options of this debianized packet https://github.com/sittner/ec-debianize/blob/master/debian/configure#L79-L95 . As you see `3.8.13-xenomai-r78` is not among the choices, but only `3.8-1-xenomai.x86-amd64`. You might also need to edit/adopt  https://github.com/sittner/ec-debianize/blob/master/debian/patches/40_xenomai_config.dpatch .

>By the way, how to use .comp file. I can't find the comp command. Which package I forget to install on debian?  

Try to follow this:
https://groups.google.com/d/msg/machinekit/r8oRB0AnJt8/0944IqkDCwAJ

AceXIE, 2016-04-19:
Thank you.
@mhaberler My gcc use hardfp as default.
```
machinekit@beaglebone:~/ethercat-1.5.2$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/4.6/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Debian 4.6.3-14' --with-bugurl=file:///usr/share/doc/gcc-4.6/README.Bugs --enable-languages=c,c++,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.6 --enable-shared --enable-linker-build-id --with-system-zlib --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.6 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --enable-plugin --enable-objc-gc --disable-sjlj-exceptions --with-arch=armv7-a --with-fpu=vfpv3-d16 --with-float=hard --with-mode=thumb --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 4.6.3 (Debian 4.6.3-14) 
```
When I use https://github.com/sittner/ec-debianize, I change the configure options just as @sirop said. Unfortunately， there are some errors. I found someone had the same question. http://comments.gmane.org/gmane.network.etherlab.user/2555.
> As a workaround, I discovered that if I "touch master/soe_errors.c"
> when this happens
> and "make install" again, then it works. Have not had time to go back to it.

I use ``dpkg-buildpackage`` to package, because I want install etherlab*-dev.deb which is needed by linuxcnc-ethercat when ```dpkg-checkbuilddeps``` https://github.com/koppi/mk/blob/daa3fd64a0b2b61933cdda100c8679dcee8232cb/Machinekit-Xenomai-Thinkpad-X200.md#linuxcnc--ethercat-hal-module

koppi, 2016-04-30:
I received a small package from Beckhoff Automation GmbH & Co. KG via UPS two days ago:

[image].
The [CX9020](http://www.beckhoff.de/english.asp?embedded_pc/cx9020.htm) is a compact, DIN rail-mountable Ethernet control system with 1 GHz ARM Cortex™-A8 CPU with an FPGA that talks to the EtherCAT fieldbus.

I will document the setup with Debian/Jessie + Machinekit [here](https://github.com/koppi/mk/blob/master/Machinekit-Beckhoff-CX9020.md) (– work in progress as of 2016-04-30).

luminize, 2016-04-30:
> On 30 Apr 2016, at 15:55, Jakob Flierl <notifications@github.com> wrote:
> The CX9020 is a compact, DIN rail-mountable Ethernet control system with 1 GHz ARM Cortex™-A8 CPU with an FPGA that talks to the EtherCAT fieldbus
> 
My knowledge is not really up to par, so forgive my ignorance. Do you mean that you would  use the Ethercat FPGA fieldbus like you would instantiate components same as you would a stepgen/encoder in a Mesa FPGA card?

koppi, 2016-04-30:
@luminize For now I will use the LinuxCNC [EtherCAT realtime HAL driver](http://wiki.linuxcnc.org/cgi-bin/wiki.pl?EtherCatDriver) to glue the cards on the EtherCAT fieldbus into the HAL layer. In my current Thinkpad X200 based [CNC mill setup](https://github.com/koppi/mk/blob/master/Machinekit-Xenomai-Thinkpad-X200.md) the component instantiation for the IO- and stepper-driver cards is configured with this [xml file](https://github.com/koppi/mk/blob/master/linuxcnc/configs/koppi-cnc/koppi-cnc.xml). – So the only real news from my previous post is, that I will give a try at replacing the Thinkpad X200 with a Beckhoff CX9020.

koppi, 2016-05-27:
I just published a new [Video](https://www.youtube.com/watch?v=Xs5yllN3u6Q) with showing a basic [siggen HAL component test](https://github.com/koppi/mk/tree/master/linuxcnc/configs/cx9020-siggen) on the CX9020.

sirop, 2016-05-27:
@koppi 
That's music. :)

mhaberler, 2016-05-27:
@koppi first, this is a superb demo video!

(curious - how do you do the terminal-window-and-machine-in-background video?)

mhaberler, 2016-05-27:
btw if packaging that effort reaches stability, I am happy to offer resources and handholding on jenkins.machinekit.io for package building and apt-repo serving

koppi, 2016-05-27:
Thank's @sirop and @mhaberler !

The CNC mill in the background is streamed to the X11 desktop with cvlc from my Logitech Inc. Pro 9000 webcam, something like:
```bash
cvlc v4l2:///dev/video1:chroma=MJPG:width=1280:height=720:fps=30
```
The X11 desktop + PulseAudio is then recorded with my X11 desktop [capture.sh](https://gist.github.com/koppi/1d7da7471a80bee339dde251d6b34a2d) Bash script.

In case I only want to record the webcam part, I also have a [capture-webcam.sh](https://gist.github.com/koppi/83d19855140575ef26bee1a50915347b) Bash script.

The YouTube video posted here consists of three sub-videos, which I post-processed with [Kdenlive](https://kdenlive.org/) – I still need to learn more about video post production..

@mhaberler As for the packaging efforts, thank's for your generous offer. – This still requires a bit of more research and time from my side, I will contact you, when I'm ready to give the packaging efforts it a try.

koppi, 2016-06-02:
Handwheel [ProNumeric OP 50M by Schleicher Electronic](https://www.bibus.ch/fileadmin/editors/countries/bagch/4_Mechatronik/schleicher/Downloads/OP_50_M_05b.pdf) supports EtherCAT. [Video](https://www.youtube.com/watch?v=kJarKvvv194) by SKCNC.

closed 2018-08-04 by ArcEye