machinekit/machinekit #632: LTTNG integration missing

mhaberler, 2015-05-13:
[LTTNG](http://lttng.org) provides a great tracing toolkit which can be used from userland as well as kernel code. It is very lightweight on the code being traced, and can be used from code running in RT threads.

Having the option to trace code with lttng would ease nailing problems like #412 and problems beyond.

Lttng is has its home at the Polytechnique Montreal, under direction of Prof. Michel Dagenais. Michel's student, Rafaël Blanchet-Lecomte, has indicated he'd be willing to help in integration and nailing #412 . 

So this issue is intended as dual-use: discuss topics related to lttng integration and use, and also to get Rafaël through the learning curve quickly. Let's help him to get to the core of the problem fast!

Rafael-BL, 2015-05-14:
Hi everyone,

My name is Rafael and I am the student that will be looking at integrating LTTng to the BBB as well as using it with machinekit in order to try and find the source of various problems (notably issue 412).

Since I have no experience neither with BBB or machinekit, I started by flashing an image to the BBB's eMMC. I used the newest realease for machinekit from elinux.org which is flasher-debian-8.0 (Jessie). the command uname -a outputs the following:

Linux beaglebone 3.14.40-ti-r62 #1 SMP PREEMPT Thu Apr 30 18:28:06 UTC 2015 armv7l GNU/Linux

Following Alex's tutorial (https://github.com/strahlex/asciidoc-sandbox/wiki/Creating-a-Machinekit-Debian-Image), I am now to the step of installing machinekit and I must admit I am a little bit confused. On the Debian packages documentation page (http://www.machinekit.io/docs/packages-debian/) I can either install what seems like a new xenomai kernel (sudo apt-get install linux-image-xenomai.beaglebone-omap) or simply the machinekit binaries. In order to get machinekit working as soon as posssible, which route should I take.

I would also like to familiarize myself with machinekit. Do the folks here have any suggestions as to where to start. Perhaps reading the LinuxCNC official documentation?

Finally, once machinekit is up and running and once I have a basic understanding as to how it works, I will need to reproduce issue 412 in order to try and identify the cause of the problem (using LTTng). Would someone be kind enough to lend me a test case that triggers the problem in order for me to reproduce it?

Thank you for your time.

Rafael

RunningLight, 2015-05-14:
On 05/14/2015 01:06 PM, Rafael-BL wrote:
> Hi everyone,
>
> My name is Rafael and I am the student that will be looking at
> integrating LTTng to the BBB as well as using it with machinekit in
> order to try and find the source of various [...]

Hi, Rafael, and welcome aboard!

I haven't set up a fresh BBB is quite a while so I'll let others comment 
on your installation question. I am concerned to see the string "debian 
8.0 (Jessie)". We don't yet support Jessie officially. Were I doing 
this, I'd drop back to Wheezy.

As for learning Machinekit, you can get a lot done faster if you install 
it to a Virtual Machine running Debian Wheezy on a x86 host. Lots more 
room, lots more horsepower, and nothing lost if you break the VM :)

Good luck and good hunting!

Regards,
Kent

cdsteinkuehler, 2015-05-14:
On 5/14/2015 12:06 PM, Rafael-BL wrote:
> Since I have no experience neither with BBB or machinekit, I started
> by flashing an image to the BBB's eMMC. I used the newest realease
> for machinekit from elinux.org which is flasher-debian-8.0 [...]

Please use the Machinekit image, also from elinux.org:

http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#BBW.2FBBB_.28All_Revs.29_Machinekit

This should be able to run Machinekit without any other changes (no need
to install anything).

Once you verify Machinekit runs, you'll probably want to build from
source in preparation for working with lttng:

http://www.machinekit.io/docs/building-from-source/

Once you can build and run from source, you can begin making changes.

-- 
Charles Steinkuehler
charles@steinkuehler.net

Rafael-BL, 2015-05-15:
First of all, I wanted to thank everyone for the warm welcome.

> I haven't set up a fresh BBB is quite a while so I'll let others comment on your installation question. I am concerned to see the string "debian 8.0 (Jessie)". We don't yet support Jessie officially. Were I doing this, I'd drop back to Wheezy.

Noted. I switched to Debian 7.8 (Wheezy).

> As for learning Machinekit, you can get a lot done faster if you install it to a Virtual Machine running Debian Wheezy on a x86 host. Lots more room, lots more horsepower, and nothing lost if you break the VM :)

Thank you for the advice. I will get a VM up and running with machinekit during the weekend.

> Once you verify Machinekit runs, you'll probably want to build from source in preparation for working with lttng:

Indeed. I just wanted to get a first experience with the BBB and machinekit before actually building from source. Since I was advised to play with machinekit in a VM, I decided to build it from source right away.

In short, I have machinekit working and built from source on the BBB. I also have LTTng working on the BBB (traced both the kernel and some userspace application). I will try to reproduce issue 412 in the upcoming days and see what LTTng has to say about it. Ill keep you posted!

Rafael

mhaberler, 2015-05-15:
@Rafael-BL - wow, watch your speed ;) that was very quick. Dont hesitate to ask in case the speed drops or the road gets bumpy!

Rafael-BL, 2015-05-19:
Hi,

I am actually having trouble reproducing issue 412. Since I had a few issues running machinekit with CRAMPS, I will go over those problems and the "fixes" I found since they perhaps introduced the errors I am having when trying to run the gcode.

First of all, when trying to launch machinekit with CRAMPS, I ran into the following error:

    Loading cape-universal overlay
    bash: line 0: echo: write error: File exists
    Error loading device tree overlay file: cape-universal
    CRAMPS.hal:10: program './setup.sh' failed, returned 1

Disabling only the audio cape (cape_disable=capemgr.disable_partno=BB-BONELT-HDMI) did not fix the error and so I ended up uncommenting the following line in /boot/uEnv.txt: 

    cape_disable=capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN

That fixed the issue.

After that, I had a "X11 connection rejected because of wrong authentication" issue. Again, I found a simple fix which consisted of :

    export XAUTHORITY=~/.Xauthority

After resolving those issues, I am able to open machinekit (axis GUI) with CRAMPS configuration. However, when running the gcode that produced issue 412, I encounter other errors. I started by attempting to run the following code:

    G21        ;metric values
    G90        ;absolute positioning
    G64 P0.1    ; path mode
    
    G92  A881.308240
    ;TYPE:FILL
    G1 X-44.982000 Y16.959000 Z6.5 A881.308240 F4800
    G1 X-45.503000 Y16.439000 A881.318030 
    G0 X-45.012 Y16.364
    G1 X-44.416000 Y16.959000 A881.329230 
    G1 X-45.012000 Y16.364000 A881.340440 
    G0 X-45.011 Y15.799
    G1 X-43.851000 Y16.959000 A881.362260
    G1 X-45.011000 Y15.799000 A881.384090 
    G0 X-44.806 Y15.438
    G1 X-43.699000 Y16.545000 A881.404920 
    G1 X-44.806000 Y15.438000 A881.425740 
    M2

It resulted in this error:

    hpg: stepgen.03.maxvel is too big for current step timings & position-scale, clipping to max possible
    
    USRMOT: ERROR: invalid command
    Linear move on line 7 would exceed joint 0's negative limit

I also tried running the following code (which is also supposed to produced the unexpected realtime delay on RT thread 1 error):

    G21        ;metric values
    G90        ;absolute positioning
    G64 P0.25   ; path mode
    G92  A881.340440
    
    F4800
    G1 X44.988 Y16.364000 A881.340440    
    G0 X44.989 Y15.799
    G1 X46.149 Y16.959000 A881.362260    
    G1 X44.989 Y15.799000 A881.384090    
    M2

The error output was the following:

    joint 1 following error
    emc/task/taskintf.cc 617: Error on axis 1, command number 213

Since my experience with linuxcnc / machinekit is non-existent, I figured I would read the linuxcnc user manual in hope for answers. However, halfway through it, I am still completely clueless as to how to fix those errors and so I decided to ask for some help here.

Thank you,
Rafael

cdsteinkuehler, 2015-05-19:
The "hpg: stepgen.03.maxvel is too big ..." error is expected when
running that configuration.  That's my fault for not tweaking the
maximum velocities when I switched from 16x to 32x micro-stepping drivers.

The "joint 1 following error" is I believe at least one of the actual
errors folks were complaining about, and the theory is the ARM side
has gotten 'stuck' somewhere for long enough the lower-level step/dir
generation has continued past where the ARM code thinks it _should_ be.

-- 
Charles Steinkuehler
charles@steinkuehler.net

Rafael-BL, 2015-05-29:
I had to recompile machinekit adding an extra CFLAG (-finstrument-functions). I modified the makefile to do so : 

    DEBUG := -g -Wall -funwind-tables -finstrument-functions (on line 357 of the makefile)

I did a few tests and it seems like the compilation with -finstrument-functions worked ok.
However, afterwards, when trying to launch machinekit with CRAMPS configuration I get the following error:

    CRAMPS.hal:29: 
    Invalid argument 'num_chan=2' to instantiated component
    NB. Use of personality or cfg is deprecated
    Singleton components cannot have multiple instances

Any idea why?

Thank you,
Rafael

mhaberler, 2015-05-29:
see current mailing list topics

it wasnt you, it was us ;) fix in the works, or see as per list

Rafael-BL, 2015-05-30:
Ah thank you very much. I should have checked the mailing list before asking but I was certain I was the one who had introduced the problem since I recompiled everything after a few modifications.

Rafael-BL, 2015-06-04:
I know issue 412 has been fixed by changing the mathematical library used in machinekit, thus removing the functions (cosinus most likely) that took too long to execute. However, from my point of view, it is still interesting to try and find exactly where the problem was. I also think that making LTTng work with machinekit might be of some use in the future, in the eventuality that a similar problem happens again.

That being said, I have a couple of questions that could help me trace machinekit.

I know the problem was coming from the trajectory planner and so I need, in order to add some tracepoints and various instrumentation functions to the file, to recompile it. Previously the command:
    make -n | grep tp.c
would output the gcc call to compile this c file. However, since I rebuilt machinekit with a more recent version, that command doesn't output anything anymore. My knowledge of autotools and makefiles in general being limited, I am having a hard time only recompiling that file. Does anyone know what command could be used to do so?

Also, I am having a hard time finding where and when the trajectory planner gets called. Is there a main that does so? Figured it might be the motion planner or the rtapi but I cant seem to find exactly where.

Thank you,
Rafael

RunningLight, 2015-06-04:
> However, from my point of view, it is still interesting to try and find exactly where the problem was. I also think that making LTTng work with machinekit might be of some use in the future, in the eventuality that a similar problem happens again.

+1

mhaberler, 2015-06-04:
hi @Rafael-BL - skype me and I'll walk you through it

Rafael-BL, 2015-06-04:
Thank you! I am ready when you are.

bitbangr, 2015-09-30:
I'd love to hear the results of the rest of the above testing.

I've just finished my CRAMPs board assembly and now in the process of testing.  I found this thread when googling for the "hpg: stepgen.03.maxvel is too big ..." error which I have also encountered.  

Was there a fix or a setting that can be made?

closed 2018-08-04 by ArcEye