machinekit/machinekit #634: src/configure messages conflict

RunningLight, 2015-05-14:
In building on Jessie just now, I noted that `src/configure` helpfully emits the message

```
...
checking version of python libraries... python2.7
checking match between tk and Tkinter versions... TCL mismatch: 8.5 vs 8.6
configure: error: Python requires use of Tcl 8.6 and Tk 8.6.
Install this version and specify --with-tclConfig and --with-tkConfig if necessary
```

However, in the middle of the output from `src/configure --help`

```
...

  --with-tcl              directory containing tcl configuration
                          (tclConfig.sh)
  --with-tk               directory containing tk configuration (tkConfig.sh)
```

So which is it? `---with-tcl` and `---with-tk` or `---with-tclConfig` and `---with-tkConfig`? And what is the correct response syntactically?

zultron, 2015-05-14:
`configure` found `tcl` v. 8.5, but `python-tk` is linked against v. 8.6 libs.  Make sure the `tcl8.6` and `tk8.6` packages are installed.  If problems continue, report back here, and then make sure the `tcl8.5` and `tk8.5` packages are uninstalled.

If `configure --help` doesn't show a `--with-tclConfig` option, then it doesn't really exist.

zultron, 2015-05-14:
OMG, we get TclOO with v. 8.6, and here I was thinking we should tie cement blocks to its feet and toss it into the Gulf.

RunningLight, 2015-05-14:
Thought I'd tried all possible combinations. I'll try again after the hockey game is over (tied 1-1 after regulation time).

TclOO? Even if we found a guru what are the odds of keeping one around?

commit f1d951fb (2015-05-14): Merge pull request #636 from RunningLight/config_patch

fix #634
closed 2015-05-14 by zultron