machinekit/machinekit #635: bool undefined when using comp in RIP installs

strahlex, 2015-05-14:
For some reason bool seems to be undefined (#include <stdbool.h> missing?) when running comp on a RIP install:
`error: unknown type name ‘bool’`

mhaberler, 2015-05-14:
bool is a C++ type. Comps and the comp tool use C.

pls show code

strahlex, 2015-05-14:
`stdbool.h` provides bool support for C. See https://github.com/thecooltool/Uni-print-3D/blob/ve4/reset.comp
Btw. why don't we use the HAL data types in the components?

mhaberler, 2015-05-14:
excellent point. that's what they are here for. I dont know why people cook up their own types.

strahlex, 2015-05-15:
Using `bit`, `float`, `u32` and `s32` would be simple. Comp just needs to add a simple typedef to the head of the file. Alternatively `hal_bit` and so on could be used.

strahlex, 2015-05-15:
hmm, short data types would colide with the data types from `rtapi_int` so I suggest using the HAL data types `hal_bit_t`, `hal_float_t`, `hal_s32_t` and `hal_u32_t`

mhaberler, 2015-05-15:
while we are at it - have a look at rtapi_int.h at git.linuxcnc.org - it had an overhaul a while ago by Jeff, might be worth adopting

ArcEye, 2015-06-27:
Following on from resolution of Issues #663 and #610 

All variables that were essentially bools (only ever held 0 or 1) were changed from int or char to
hal_bit_t

Both comp and instcomp will #define true/TRUE false/FALSE by default
(unless `option no_convenience_defines` is set)

@strahlex and I concluded that it would be preferable to have such values assigned using true or false, as it would make the type and usage of the variable much clearer within the code.

I assume from the preceding that there is no thought of defining a hal_bool_t type or using bool itself.

If that is so, I can change the affected icomps accordingly, so that they are all completely 'politically correct' type wise, as shining examples for future components ;)

strahlex, 2015-06-27:
@ArcEye I think it is a wise idea to only use only lower or uppercase bool types. I would prefer lowercase `true` and `false` since they are usually highlighted by syntax highlighters. `#include <stdbool.h>` should be fine in this case.

I agree that using the bool types should make understanding of the components easier.

ArcEye, 2015-06-27:
I agree with lower case, that is the normal C / C++ syntax

We don't even need to `#include stdbool.h`, if we continue to use hal_bit_t as the data type.
Just define true as 1 and false as 0, which is already done in the preprocessing in instcomp and comp.

strahlex, 2015-06-27:
According to http://pubs.opengroup.org/onlinepubs/9699919799/basedefs/stdbool.h.html stdbool.h does nothing except defining `true` as 1 and `false` as 0. If we use our own `#define` we might get compiler warnings if someone includes `stdbool.h` somewhere in the other included headers.

ArcEye, 2015-06-28:
http://pubs.opengroup.org/onlinepubs/9699919799/basedefs/stdbool.h.html
""An application may undefine and then possibly redefine the macros bool, true, and false.""

instcomp and comp  #ifdef - #undef - #define , so does not really matter, it will just be defined again to the same, not defined twice in a way that will get warnings.

I will look at it next week, probably take out the capitalised defines, so that if lowercase not used will get an error, should enforce lowercase usage.

ArcEye, 2018-08-01:
The redefined bool defs have been in use for years w/o problems AFAIK

closed 2018-08-01 by ArcEye