machinekit/machinekit #640: Broken xhc-hb04.tcl. XHC-HB04 MPG pendant module not working anymore.

OttoMeister, 2015-05-17:
I use a ***XHC-HB04 MPG pendant module*** on BeagleBoneBlack Revision C element14 on BeagleBoneBlack Revision C element14. I install *bone-debian-7.8-machinekit-armhf-2015-04-26-4gb.img.xz* and made apt-get update and upgrade and reboot. So i end up with kernel **3.8.13-xenomai-r72** (uname -r)

Start linuxcnc and select sample configuration:
sim -> axis-> xhc-hb04 -> xhc-hb04-layout2_mm. I get a lot off errors. Some thing with the **xhc-hb04.tcl** is not working. Did not had the problem with my old installation (*bone-debian-7.8-machinekit-armhf-2015-03-15-4gb.img.xz*). There it worked after I added udev rule and a few type cast (unsigned char) in the xhc-hb04.cc file.

You do not need a XHC-HB04 to test it. In the file */home/machinekit/linuxcnc/configs/sim.axis.xhc-hb04/xhc-hb04-layout2_mm.ini* on row 22 you can change "require_pendant = yes" to no. Now you see the same error messages:

    xhc-hb04.tcl: !!! <xhc-hb04.button-goto-zero> pin does not exist, continuing
    xhc-hb04.tcl: !!! <xhc-hb04.button-half> pin does not exist, continuing
    xhc-hb04.tcl: !!! <xhc-hb04.button-home> pin does not exist, continuing
    xhc-hb04.tcl: !!! <xhc-hb04.button-macro-1> pin does not exist, continuing
    xhc-hb04.tcl: !!! <xhc-hb04.button-macro-2> pin does not exist, continuing
    xhc-hb04.tcl: !!! <xhc-hb04.button-macro-3> pin does not exist, continuing
    xhc-hb04.tcl: !!! <xhc-hb04.button-macro-6> pin does not exist, continuing
    xhc-hb04.tcl: !!! <xhc-hb04.button-macro-7> pin does not exist, continuing
    xhc-hb04.tcl: !!! <xhc-hb04.button-mode> pin does not exist, continuing
    xhc-hb04.tcl: !!! <xhc-hb04.button-probe-z> pin does not exist, continuing
    xhc-hb04.tcl: !!! <xhc-hb04.button-reset> pin does not exist, continuing
    xhc-hb04.tcl: !!! <xhc-hb04.button-rewind> pin does not exist, continuing
    xhc-hb04.tcl: !!! <xhc-hb04.button-safe-z> pin does not exist, continuing
    xhc-hb04.tcl: !!! <xhc-hb04.button-spindle> pin does not exist, continuing
    xhc-hb04.tcl: !!! <xhc-hb04.button-step> pin does not exist, continuing
    xhc-hb04.tcl: !!! <xhc-hb04.button-stop> pin does not exist, continuing
    xhc-hb04.tcl: !!! <xhc-hb04.button-zero> pin does not exist, continuing

OttoMeister, 2015-05-17:
Cannot put labels on. Why?

marksue, 2015-06-21:
Because I want to use the MPG too, I analysed the problem more in detail. Here what I find out so far:
The device is initialized by the HALFILE script xhc-hb04.tcl. The script tries to connect some pins, but before, it checks whether the pin exists. To do this, the hal command “show pin 'pinname'” is called, and the response will be analysed, but the response is not as expected.
When executing the command “show pin 'pinname'”, the function print_pin_info(…) in the file halcmd_commands.c is called. The output is defined as:

halcmd_output("Component Pins:\n");
halcmd_output("  Comp   Inst Type  Dir         Value  Name                             Epsilon         Flags\n");

but the script only receive

"Component Pins:\n"
"  Comp   Inst Type  Dir         Value  Name                             Epsilon"

The second line was truncated after 80 characters and therefore the HALFILE script xhc-hb04.tcl fails with the reported error messages.
The truncation can also be observed in the “HAL Configuration” dialog when executing “show pin” in the “Test HAL command:” box.

Up to now I don’t know why the line will be truncated.

Other issues with xhc-hb04: 

1) The driver for the xhc-hb04 is not platform independent, because cast expressions are used with the assumption the data type ‘char’ is signed. But on the BBB ‘char’ seems to be unsigned.

The line  

   *(xhc.hal->jog_counts) += ((char)in_buf[4]);

in the file xhc-hb04.cc must therefore be changed to 

   *(xhc.hal->jog_counts) += ((signed char)in_buf[4]);
  
to get a correct left\rigth jog behaviour. Maybe instead of ‘char’, ‘int’, ‘short’ etc. <stdint.h> types (int8_t, …) should be used.

2) Due to the last commit in xhc-hb04.cc to rename all math functions, the display of the MPG does not show factional digits anymore. The reasons are additional cast expressions to (int) that trancate the value to be displayed.
Example: 

Lines with

xhc_encode_float((int)rtapi_rint(...)

have to be changed to

 xhc_encode_float(rtapi_rint(...)

Before commit:
        if (xhc->axis == axis_a) p += xhc_encode_float(round(1000 * *(xhc->hal->a_wc)) / 1000, p);
        else p += xhc_encode_float(round(1000 * *(xhc->hal->x_wc)) / 1000, p);
	p += xhc_encode_float(round(1000 * *(xhc->hal->y_wc)) / 1000, p);
	p += xhc_encode_float(round(1000 * *(xhc->hal->z_wc)) / 1000, p);
	if (xhc->axis == axis_a) p += xhc_encode_float(round(1000 * *(xhc->hal->a_mc)) / 1000, p);
	else p += xhc_encode_float(round(1000 * *(xhc->hal->x_mc)) / 1000, p);
	p += xhc_encode_float(round(1000 * *(xhc->hal->y_mc)) / 1000, p);
	p += xhc_encode_float(round(1000 * *(xhc->hal->z_mc)) / 1000, p);
	p += xhc_encode_s16(round(100.0 * *(xhc->hal->feedrate_override)), p);
	p += xhc_encode_s16(round(100.0 * *(xhc->hal->spindle_override)), p);
	p += xhc_encode_s16(round(60.0 * *(xhc->hal->feedrate)), p);
	p += xhc_encode_s16(round(60.0 * *(xhc->hal->spindle_rps)), p);

After commit:
        if (xhc->axis == axis_a) p += xhc_encode_float((int)rtapi_rint(1000 * *(xhc->hal->a_wc)) / 1000, p);
	else p += xhc_encode_float((int)rtapi_rint(1000 * *(xhc->hal->x_wc)) / 1000, p);
	p += xhc_encode_float((int)rtapi_rint(1000 * *(xhc->hal->y_wc)) / 1000, p);
	p += xhc_encode_float((int)rtapi_rint(1000 * *(xhc->hal->z_wc)) / 1000, p);
	if (xhc->axis == axis_a) p += xhc_encode_float((int)rtapi_rint(1000 * *(xhc->hal->a_mc)) / 1000, p);
	else p += xhc_encode_float((int)rtapi_rint(1000 * *(xhc->hal->x_mc)) / 1000, p);
	p += xhc_encode_float((int)rtapi_rint(1000 * *(xhc->hal->y_mc)) / 1000, p);
	p += xhc_encode_float((int)rtapi_rint(1000 * *(xhc->hal->z_mc)) / 1000, p);
	p += xhc_encode_s16((int)rtapi_rint(100.0 * *(xhc->hal->feedrate_override)), p);
	p += xhc_encode_s16((int)rtapi_rint(100.0 * *(xhc->hal->spindle_override)), p);
	p += xhc_encode_s16((int)rtapi_rint(60.0 * *(xhc->hal->feedrate)), p);
	p += xhc_encode_s16((int)rtapi_rint(60.0 * *(xhc->hal->spindle_rps)), p);

mhaberler, 2015-06-22:
@marksue - great find! can you make this into a PR?

marksue, 2015-06-23:
@mhaberler: Yes, I will write PRs for the issues (next week when I am back from a short mountain bike trip in austria :-))

mhaberler, 2015-06-23:
@marksue - great! where are you biking? I'm near Graz https://goo.gl/maps/o39y0

marksue, 2015-07-03:
I created sperate PRs for issue 1 (#699) and 2 (#700). For this PRs I could submit a pull request. 

Although the true reason for the "pin does not exist" error message is the transcation of the received output message (see descrition above) the problem could also be fixed by merging
http://sourceforge.net/p/emc/mailman/message/33333328/ into machinekit. 

@mhaberler : I started in Germany (Teggernsee) crossed over the Geißeljoch and Pfitscherjoch and finished in Italy (Sterzing). This time Graz was not on my route ;-)

marksue, 2015-07-05:
Further investigations of truncation:

With introduction of multiple instantiation, hal messages were changed. Now there are messages
which exceed 80 characters. For example halcmd_output(" Comp Inst Type Dir Value Name Epsilon Flags\n");

The output buffer length used in function halcmd_output() (halsh.c) is define as 
"#define BUFFERLEN 80" in the file configure.ac.

Implementation of halcmd_output: 

#ifndef BUFFERLEN
#define BUFFERLEN 1024
#endif

void halcmd_output(const char *format, ...) {
    char buf[BUFFERLEN + 1];
    va_list ap;
    int len;
    va_start(ap, format);
    vsnprintf(buf, BUFFERLEN, format, ap);
    va_end(ap);

=> BUFFERLEN in configure.ac needs to be increased.

mhaberler, 2015-07-06:
@marksue  - instead of patching up kludgy haltcl files which parse halcmd output to somehow figure what is going on, I'd recommend to drop this script altogether and replace it by a Python script which uses the cython bindings. See examples in nosetests/*.py.

Unfortunately this habit of parsing halcmd output permeates through all of tests/* and other locations, and is a surefire way to blow up if something in HAL changes. And a _log_ has changed in HAL.

hboterman, 2015-09-13:
@mhaberler - Is there any chance to have marksue's PR (left/right and pins problems solutions) merged soon ? I have an HXC-HB04 that doesn't work anymore (lost keys functionnality after last update and the jog only moves any axis upward).

mhaberler, 2015-09-13:
sorry - it was unclear this PR was still current, I'll have a look

hboterman, 2015-09-13:
Ok, thanks !

OttoMeister, 2015-09-14:
Maybe we could also somehow install by default the needed udev rule. The 
xhc-hb04 executable needs permission for reading the pendant’s USB 
device. I put this text line in /etc/udev/rules.d/90-xhc.rules:

ATTR{idProduct}=="eb70", ATTR{idVendor}=="10ce", MODE="0666", 
OWNER="root", GROUP="plugdev"

mhaberler, 2015-09-14:
please see how `shmdrv.rules` is handled, do likewise and post  a PR

OttoMeister, 2015-09-15:
machinekit/src/rtapi/shmdrv/shmdrv.rules
# make the shared memory allocator publically accessible
KERNEL=="shmdrv", MODE="0666"

Sorry, I don't understand it. Could somebody else please help me with 
this issue?

mhaberler, 2015-09-15:
sorry for being unclear - I meant - follow how  shmdrv.rules is installed and do the same for your .rules file

hboterman, 2015-09-17:
Hi,
I have upgraded Machinekit to the last version available: 0.1.1469-1da.bbot.pr756.gitbca988b~1wheezy
Jog works now on both direction on every axis, and decimal digits are displayed.
Sadly, there is still no button working. Is BUFFERLEN been increased ?

hboterman, 2015-09-17:
Here is messages output:
...
xhc-hb04: waiting for XHC-HB04 device
xhc-hb04: found XHC-HB04 device
xhc-hb04.tcl: !!! <xhc-hb04.button-goto-zero> pin does not exist, continuing
xhc-hb04.tcl: !!! <xhc-hb04.button-half> pin does not exist, continuing
xhc-hb04.tcl: !!! <xhc-hb04.button-home> pin does not exist, continuing
xhc-hb04.tcl: !!! <xhc-hb04.button-macro-1> pin does not exist, continuing
xhc-hb04.tcl: !!! <xhc-hb04.button-macro-2> pin does not exist, continuing
xhc-hb04.tcl: !!! <xhc-hb04.button-macro-3> pin does not exist, continuing
xhc-hb04.tcl: !!! <xhc-hb04.button-macro-6> pin does not exist, continuing
xhc-hb04.tcl: !!! <xhc-hb04.button-macro-7> pin does not exist, continuing
xhc-hb04.tcl: !!! <xhc-hb04.button-mode> pin does not exist, continuing
xhc-hb04.tcl: !!! <xhc-hb04.button-probe-z> pin does not exist, continuing
xhc-hb04.tcl: !!! <xhc-hb04.button-reset> pin does not exist, continuing
xhc-hb04.tcl: !!! <xhc-hb04.button-rewind> pin does not exist, continuing
xhc-hb04.tcl: !!! <xhc-hb04.button-safe-z> pin does not exist, continuing
xhc-hb04.tcl: !!! <xhc-hb04.button-spindle> pin does not exist, continuing
xhc-hb04.tcl: using std_start_pause_button
xhc-hb04.tcl: !!! <xhc-hb04.button-step> pin does not exist, continuing
xhc-hb04.tcl: !!! <xhc-hb04.button-stop> pin does not exist, continuing
xhc-hb04.tcl: !!! <xhc-hb04.button-zero> pin does not exist, continuing
task pid=7258
...

mhaberler, 2015-09-18:
this piece of code, in particular the TCL scripts, are some of the worst legacy which we carried over from linuxcnc. The  idea of the TCL scripts - use  halcmd output, parse that and use it as an API - is just silly, and bound to break time and again. Nevermind there is barely anybody around who understands TCL, which has fallen into disuse and we really want to get rid of it.

My recommendation is:
- take the TCL code, try to understand it, then throw it away. All of it. No exceptions.
- work with the author - you find Frederic on the list.
- rewrite the TCL scripts from scratch to use a proper API which is not bound to break everytime some line in a utility like halcmd - which is designed for interactive use, not as an API replacement - us changed.

Sorry to have no better news. But at some point one needs to make a cut and start over. This is one. I fear you are out on your own on this one.

ArcEye, 2016-08-19:
This is an interim fix for the `pin does not exist` problem
https://github.com/ArcEye/machinekit/tree/halcmd-print-pexists

It needs testing by running the sim with a pendant plugged in

The problem is caused as @mhaberler stated, by tcl use of the output from a command line utility ie. halcmd and strictly checking the output against what was output at a fixed point in time, making no allowances for future changes.

I have changed halcmd by adding a function that outputs a simple string confirmation of existence, which tcl can use, which will keep it going for now.

The answer undoubtedly is to replace tcl, but I don't have one of these pendants to even begin that..

If someone wants to lend or give me one to test with ..........................

ArcEye, 2016-12-28:
I received one a few weeks later and confirm that this fixed the particular issue.

I completely rewrote it to use Qt5 in my own repo, but this is not ready for release until Qt5 becomes the default version (stretch?).
Then it may be viable to completely remove tcl.

closed 2016-12-28 by ArcEye