machinekit/machinekit #644: MULTIARCH problem configuring tcl in Jessie i386

RunningLight, 2015-05-19:
I should have paid closer attention to @mhaberler remarks about passing tcl/tk options into configure https://groups.google.com/d/msg/machinekit/HVa1MtrWlBc/gg--sCgcqosJ

When I try building Machinekit from sources on a Jessie i386 VM I encounter a problem which wasn't present in my build on a Jessie amd64 VM.

yada yada yada
$ ./autogen.sh
$ ./configure
...
checking for Tcl configuration... found /usr/lib/tcl8.6/tclConfig.sh
checking for existence of /usr/lib/tcl8.6/tclConfig.sh... loading
/usr/lib/tcl8.6/tclConfig.sh: line 2: /usr/lib/i686-linux-gnu/tcl8.6/tclConfig.sh: No such file or directory

I note that tcl, tcl8.6, tcl8.6-dev:i386, tk, tk8.6, and tk8.6-dev:i386 packages have all been installed by this point so that's not the problem.

Looking at the specific script which is failing

$ more /usr/lib/tcl8.6/tclConfig.sh
#! /bin/sh
. /usr/lib/`dpkg-architecture -qDEB_HOST_MULTIARCH`/tcl8.6/tclConfig.sh

Trying the meat of that command at the command line

$ dpkg-architecture -qDEB_HOST_MULTIARCH
i386-linux-gnu

and, sure enough, the sought-after file is located at /usr/lib/i386-tcl8.6-linux-gnu/tcl8.6/tclConfig.sh

For reference,

$ uname -a
Linux debian80-i386-lxde-test 3.16.0-4-586 #1 Debian 3.16.7-ckt9-3-deb8u1 (2015-04-24) i686 GNU/Linux

@zultron it looks like this conflict between i386 and i686 arises in configure.ac where it constructs its own DEB_HOST_MULTIARCH=${host_cpu}-${host_os}. What a PITA!

Obviously, this problem can be sidestepped by passing explicit options as @mhaberler did but it would be nice to nip it in the bud.

[as an aside, I tried to submit this issue several times with no response from github.com. Now it's later and also I'm using a different browser. My apologies if it suddenly shows up more than once.]

zultron, 2015-05-28:
@RunningLight, d'oh, I'm embarrassed.

closed 2015-05-28 by mhaberler

RunningLight, 2015-05-28:
All's well that ends well!

PR #658 (2015-05-28): Fixup e94b7ad1:  Unbreak tclConfig.sh
Setting `DEB_HOST_MULTIARCH` to the value detected by `config.guess`
will be wrong in many cases, and indeed breaks the multiarch
`tclConfig.sh` script on i?86:

    $ dpkg-architecture -qDEB_HOST_MULTIARCH
    i386-linux-gnu
    $ ./config.guess
    i686-pc-linux-gnu
    $ dpkg-query -L tcl8.6-dev | grep tclConfig.sh
    /usr/lib/tcl8.6/tclConfig.sh
    /usr/lib/i386-linux-gnu/tcl8.6/tclConfig.sh
    $ cat /usr/lib/tcl8.6/tclConfig.sh
    #! /bin/sh
    . /usr/lib/`dpkg-architecture -qDEB_HOST_MULTIARCH`/tcl8.6/tclConfig.sh

Fixes #644
commit 0ab6d2b9 (2015-05-28): Fixup e94b7ad1:  Unbreak tclConfig.sh

Setting `DEB_HOST_MULTIARCH` to the value detected by `config.guess`
will be wrong in many cases, and indeed breaks the multiarch
`tclConfig.sh` script on i?86:

    $ dpkg-architecture -qDEB_HOST_MULTIARCH
    i386-linux-gnu
    $ ./config.guess
    i686-pc-linux-gnu
    $ dpkg-query -L tcl8.6-dev | grep tclConfig.sh
    /usr/lib/tcl8.6/tclConfig.sh
    /usr/lib/i386-linux-gnu/tcl8.6/tclConfig.sh
    $ cat /usr/lib/tcl8.6/tclConfig.sh
    #! /bin/sh
    . /usr/lib/`dpkg-architecture -qDEB_HOST_MULTIARCH`/tcl8.6/tclConfig.sh

Fixes #644