machinekit/machinekit #651: src/configure options

RunningLight, 2015-05-22:
So far, all I've proposed in #650 is superficial rebranding of `src/configure.ac` and hence `src/configure`. However, the options offered by `configure` seem to be suffering bit-rot.

1) I propose to modify the wording in and below the `--prefix` and `--exec-prefix` options to clarify what is now meant by "run-in-place". 

2) concerning alternative platforms, there is ambiguity in the twice-repeated "default for ARM arch"---Beaglebone and Raspberry can't both be the default, and why are they differentiated from other ARM-based boards such as the zedboard we already acknowledge---as well as ambiguity in "Raspberry" now that there exists both the original RPi (armv6) and the new RPi2 (armv7). I'm still making sure I understand the magic occurring in Section 2.3 before proposing any changes.

3) options related to building documentation: haven't these all been "rendered inoperative" as the Nixon White House was fond of saying? Isn't it time to remove them?

zultron, 2015-05-24:
Re. 1), ugh, can't wait to refer to 'run-in-place' by its acronym's more general meaning.

Re. 2), it means 'default *enabled* for ARM arch'.  Those two are differentiated because they have special, separate pieces of the build that other ARM boards don't have.  It's also that way because originally, one would select the one or the other, until we decided it's better to build for both by default (I don't remember why that's better).  I actually don't like those wordy status messages, and would prefer just 'yes' or 'no', but of course that's in disagreement with the original authors.

Re. 3), I think so.  @luminize would know for sure.

luminize, 2015-05-24:
regarding 3) you could remove the building of docs since building docs can't be done unless you get the machinekit-docs repo back into machinekit and I don't see a reason to do that.

It would be better _if_ needed that building is done from the machinekit-docs repo. But why use PDF/HTML if there are the viewable asciidocs?

So my conclusion: would be best to clean up by removing.

ArcEye, 2018-08-01:
Overtaken by events

closed 2018-08-01 by ArcEye