machinekit/machinekit #653: RTAI /procfs support broken in 3.10+ kernels

zultron, 2015-05-24:
Two commits in #647, b30eab3f and 29f46a07, attempt to fix `procfs` support after an API change in 3.10.  In fact, they don't work:
```
May 24 01:57:48 d8-64-rtai msgd:0: hal_lib:0:kernel hal_xinit:138 HAL error: could not init /proc files
May 24 01:57:49 d8-64-rtai kernel: [ 6433.831744] WARNING: CPU: 0 PID: 12979 at /build/linux-MYNz_0/linux-3.16.7/debian/build/source_rtai.x86/fs/proc/generic.c:434 proc_create_data+0x87/0xb0()
```

mhaberler, 2015-05-30:
I think this is a great opportunity to put most of this to rest

the only feature needed in the kernel proper is the rtapi command interface and that better become a sysfs entry. The rest is just poking through the RTAPI data segment and printing stuff, that can be very well done in halcmd or Python if so desired.

zultron, 2015-05-30:
Reading through the RTAPI data seg. would be great, sounds like it would be general for all flavors.  RTAI is destined to become our only kthreads flavor, and one of the least used, and I hate putting so much work into it.

zultron, 2015-07-26:
Closing (wontfix):  Issue should be fixed using @mhaberler's more general method.

closed 2015-07-26 by zultron