machinekit/machinekit #660: State engine component

luminize, 2015-05-29:
While working towards #589 and knowing of #631I have the conjecture I'm going to need a state engine component. Thinking a little but further, a state engine component can not only find its use in "embedding in other components" like when communicating with busses (startup of nodes behaviour) but could also make Machinekit better suited for industrial control uses.

I can imagine sending a message (doesn't matter over which wires) to the HAL which acts as a transition between states, but I can also imagine a message being sent (action) when a state becomes active.

thoughts?

mhaberler, 2015-05-29:
I see where you're coming from, but I dont think making a state machine (FMS) a generalized component is going to help much

however, it is quite common for components to integrate a FSM; in the case of the CANopen component I would see one FSM per perpheral device, for instance for NMT status

here's an example component which uses a FSM internally: http://git.mah.priv.at/gitweb?p=emc2-dev.git;a=blob;f=src/hal/components/mvjun.comp;h=51847726efcad93823c3dec11961e694bd389c2a;hb=fadc472db6b0feb94c96cbb8f8f809ff8b49d5bc

it is a tool changer comp with several internal states, including time dependent ones (eg signal timeout on the Z pin in the MS_INDEXING state). I would think the state would become part of the instance data.

pJanez, 2015-05-29:
I was thinking about state machine too. CANopen has NMT(network management). NMT master somehow starts all nodes on startup and network becomes operational. This should not be complicated.

It may be good idea to look into DS402, state machine for drives(FSA - finite state automation). Probably all CANopen based drives use it, including Ethecat. Machinekit could have it's own state machine, which could be the "sum" of states of the remote drives (it is necessary to switch-on and enable-operation of the drives somehow). Machinekits state machine could be somehow controlled by switch, which powers-up the drives and actuators in CNC machine.
I don't know, how this is solved now in Machinekit, but I think this part should be standardized. 

controlWord and statusWord are used to control the states. Here is picture from DS402:
[image]

I think, Machinekit should have implemented own Object dictionary, which could be similar in some points with Object dictionary in DS402 drives.

mhaberler, 2015-05-29:
Janez - have a look at the code link I posted above, this is _exactly_ how a RT comp state machine should be done

to understand the code, just assume FUNCTION is called by a timer (1mS)

luminize, 2015-05-29:
maybe @jpmoniz could chime in.
I've been thinking about a state engine for a long time. The background of the question might make the water a bit murky. What I'm getting at is getting a way of the behaviour of a machine formalised. 

For example: 
- Action "output A high" is only active in state 4 OR state 7
- Action "send message foobar" when state 3 becomes active.
- State 4 becomes active if state 3 is active, AND sensor 1 rises
- State 7 becomes active if state 6 is active AND (sensor 1 OR sensor 2 rises)

When you're an integrator, it should be fairly easy to change the behaviour. For the historical Desktop CNC this is not an issue, since you don't change your machine daily, and the machine is fairly simple. But when you're doing something industrial automation-ish you will adapt the machine while you're commissioning the machine and working towards the Site Acceptance Test when the customer's Project Manager is breathing down your neck. Just change the logic of transitions, and then add another state in the OR of an "Action".

The reason is that I was thought so when I programmed PLC's. It's easy to formalise machine behaviour (and for that matter process behaviour).
if you were to wire up everything in HAL with signals and pins. This will make for spaghetti logic (which of course we don't want).
* you can quickly lose the "when does Output A need to become active" logic
* it makes looking for errors (not in the code, but in the logic) very hard.
* it makes changing the behaviour hard.

It's more of a rung higher on the ladder, maybe the location of this proposed component regarding the "H" of Hardware in "HAL" is maybe making this a murky place to do machine/process logic. Maybe we need a "PAL" (Process Abstraction Layer) :)

Anyway, a state engine could be used from inside components (instantiating a state-engine like the marvelous ring-buffer is done) as well as formalising machine/process states.

In a way this is more about making machine behaviour easier to understand.

mhaberler, 2015-05-29:
this is all neat in theory, and there is also a loooong history of FSM-based design methods and software.

There is one key problem with FSM-based design methods and it is called state explosion.

IOW: if you do anything non-trivial with FSMs the state space becomes unwieldy very quickly (I happened to do a PhD thesis on protocol analysis and FSM's communicating by messages was one of the "hoped for" tools to nail the analysis problem. It was an utter failure and barely anybody goes down that route nowadays).

So I would be extremely cautious by using FSM's for manual coding with anything more than say 10 states. There are FSM-based tools which go beyond that like flex (generally - regex matchers) but I hope nobody ever needs to nail a bug in such a tool. Give it a try: take any non-trivial regex syntax (a recognizer for a regex can be built as a FSM so its a isomorphic problem) like for instance C identifiers. Run it through flex and have it generate the FSM description. My prediction: the output will not comprehensible.

that said, the diagram Janez posted a FSM is the right tool in terms of problem size. Manual coding ala mvjun.comp entirely possible and comprehensible by mortals.

pJanez, 2015-05-30:
Michael,

I looked into your FMS example. It's wery clear. switch(state){case: ...} inside 1ms timer thread. I use the same principle in much places in my programs.

For now I can't experiment or work with machinekit directly, but it is my strong point of interest in future. I'm now more focused on "low level" CANopen, I can't do both. (That doesn't mean I don't have many ideas :) ) I agree with Bas, it will have to be possible to configure Machinekit when it is part of something industrial automation-ish.

It may be pain for existing developers of Machinekit or LinuxCNC, but I think **Object Dictionary**, as specified by CANopen standard, should be integrated into the central part of the Machinekit. Even if some existing parts of Machinekit would be replaced by it. Machinekit should be configurable and monitorable via this standardized interface. Master functionality may also be handled with Machinekit, but it can be separate part.

---

Let me explain the basic principle of **Object Dictionary**, which is part of CANopen specification and is also common to Ethercat and similar protocols. **Object Dictionary** is a collection of variables inside one device. Each variable is accessible via (CAN) network by knowing node-id of device, index-subindex pair of the variable and a description of a variable. Description of a variable is part of device documentation and is also part of eds file(electronic data sheet), supplied with each device.

There are usually many OD variables in each device. Variables at indexes 0x1000...0x1FFF are used for communication protocol. At 0x2000...0x5999 are manufacturer specific variables, not standardized. At 0x6000+ are device profile specific variables. All variables from DS401, DS402, etc. are in this section.

SDO client(master) can access to any of those variables. This is so called "Service Data Object" access to variables. This is basically a pair (or multiple pairs) of CAN messages exchanged between a client and a server. SDO communication is basicaly used only for the configuration of the devices.
For process communication (fast, real-time) are used PDO messages. These are basically CAN messages (configurable via SDO) into which are mapped necessary variables from the **Object Dictionary**.

What are **OD variables**: these are ordinary C variables or arrays or structs of any type (from char to long long int, float, double, string, etc.). Variable can be read-only or read-write from network side. If variable is up to 4 bytes long, it is transfered in one pair of SDOs, othervise it is split into segments(handled by protocol). In CANopenNode variables can also be functions, where client read or write request trigers function in server(slave). Function eather accepts argument and do some processing or do some processing and returns the result.

---

How DS402 state machine (it's quite sophisticated) uses Object Dictionary: by four variables only: control_word, status_word, mode_of_opertion and mode_of_operation display.

What about state machine from Michaels example: put some control variables and "pstate" into Object Dictionay. It is then easy also for debug (via Wireshark for example).

There can be many state machines in one device, of course.

---

If there will be interest for integrating state machine from DS402 into Machinekit, I can contribute the code. FSA from DS402 may seems complicated from the beginning, but it is quite useful acually. My code is full implementation of FSA, as shown on picture above. Different modes of operation are in separate files.

luminize, 2015-06-25:
Let me close this issue since my initioal thought was reusability betwen various busses which seem to use some form of state engine. For now I'll put specific a state engine in code in the component it belongs. We can always yank it out later if we need to reuse the code.

closed 2015-06-25 by luminize

luminize, 2015-07-05:
@pJanez could you show me where your code regarding the DS402 state machine can be found? I'm having a hard time finding out the command structure for getting states changed for DS402 devices. By hand first before I do any coding. I seem to miss a small bit of knowledge how to interpret the SDO messages and their replies w.r.t changing the states.

pJanez, 2015-07-06:
I didn't publish it yet. It wouldn't help much for machinekit directly.
Best source for DS402 state machine is the standard itself or some operating manuals of the drives. Here is excerpt from my code, which may be interesting:

    /**
     * Constants: Statusword
     *
     * Statusword bits and internal states according to CiA 402.
     */
        #define DRV_SW_readyToSwitchOn            0x0001
        #define DRV_SW_switchedOn                 0x0002
        #define DRV_SW_operationEnabled           0x0004
        #define DRV_SW_fault                      0x0008
        #define DRV_SW_voltageEnabled             0x0010
        #define DRV_SW_quickStopNot               0x0020
        #define DRV_SW_switchOnDisabled           0x0040
        #define DRV_SW_warning                    0x0080
        #define DRV_SW_08_ms                      0x0100
        #define DRV_SW_remote                     0x0200
        #define DRV_SW_targetReached              0x0400
        #define DRV_SW_internalLimitActive        0x0800
        #define DRV_SW_12_oms                     0x1000
        #define DRV_SW_13_oms                     0x2000
        #define DRV_SW_14_ms                      0x4000
        #define DRV_SW_15_ms                      0x8000

        #define DRV_STATE_NOT_READY_TO_SWITCH_ON  0x0000     /* use MASK_NO_QS */
        #define DRV_STATE_SWITCH_ON_DISABLED      0x0040     /* use MASK_NO_QS */
        #define DRV_STATE_READY_TO_SWITCH_ON      0x0021     /* use MASK */
        #define DRV_STATE_SWITCHED_ON             0x0023     /* use MASK */
        #define DRV_STATE_OPERATION_ENABLED       0x0027     /* use MASK */
        #define DRV_STATE_QUICK_STOP_ACTIVE       0x0007     /* use MASK */
        #define DRV_STATE_FAULT_REACTION_ACTIVE   0x000F     /* use MASK_NO_QS */
        #define DRV_STATE_FAULT                   0x0008     /* use MASK_NO_QS */
        #define DRV_STATE_MASK                    0x006F
        #define DRV_STATE_MASK_NO_QS              0x004F

        #define DRV_STATE_CHANGE(SW, NEW_STATE)  {SW &= 0xFF90; SW |= NEW_STATE;}

    /**
     * Constants: Controlword
     *
     * Controlword bits and commands according to CiA 402.
     */
        #define DRV_CW_04_oms                     0x0010
        #define DRV_CW_05_oms                     0x0020
        #define DRV_CW_06_oms                     0x0040
        #define DRV_CW_faultReset                 0x0080
        #define DRV_CW_halt                       0x0100
        #define DRV_CW_09_oms                     0x0200
        #define DRV_CW_10_reserved                0x0400
        #define DRV_CW_11_ms                      0x0800
        #define DRV_CW_12_ms                      0x1000
        #define DRV_CW_13_ms                      0x2000
        #define DRV_CW_14_ms                      0x4000
        #define DRV_CW_15_ms                      0x8000

        #define DRV_COMM_SHUTDOWN                 0x008F, 0x0006
        #define DRV_COMM_SWITCH_ON                0x008F, 0x0007
        #define DRV_COMM_DIS_VOLTAGE              0x008F, 0x0000
        #define DRV_COMM_QUICK_STOP               0x0086, 0x0002
        #define DRV_COMM_DISABLE_OPER             0x008F, 0x0007
        #define DRV_COMM_ENABLE_OPER              0x008F, 0x000F

        #define DRV_IS_COMM_SHUTDOWN(cw)          ((cw&0x0087) == 0x0006)
        #define DRV_IS_COMM_SWITCH_ON(cw)         ((cw&0x008F) == 0x0007)
        #define DRV_IS_COMM_DIS_VOLTAGE(cw)       ((cw&0x0082) == 0x0000)
        #define DRV_IS_COMM_QUICK_STOP(cw)        ((cw&0x0086) == 0x0002)
        #define DRV_IS_COMM_DISABLE_OPER(cw)      ((cw&0x008F) == 0x0007)
        #define DRV_IS_COMM_ENABLE_OPER(cw)       ((cw&0x008F) == 0x000F)