machinekit/machinekit #661: jessie BB install fails in configure with out of memory - checking for boost::python shared library...

mhaberler, 2015-05-29:
upgraded a BB to jessie

during configure cc1plus runs out of memory (no swap space configured)
mah@epig:~/machinekit/src$ g++ --version
g++ (Debian 4.9.2-10) 4.9.2

looks like gcc template processing sets new records for memory usage with gcc-4.9 :-/

possible solutions:
- add swap space 
- pass -O0 to g++ at least during this configure test

we'll see what happens to the boost-using source files once we get past configure

mhaberler, 2015-05-29:
seems to be related: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=61183

RunningLight, 2015-05-29:
I think we're doomed without physical swap space...

Very unscientifically, I did a quick test in an Jessie x86 VM set to 508MB base memory to mimic the BBB. With several trials of `src/configure`, each time starting with a fresh boot and `debian/configure`, etc to clear out the cobwebs, swap usage peaked anywhere from 100MB to 200MB (determined by eyeballing `top` in a separate terminal window; I'm sure I could get different results with better probes but I don't see the point.). Ouch! 

I then naively tried setting environment variables to pass -O0 to the compiler/precompiler...didn't seem to help much (which is what the bugzilla report you pointed out suggested to me would be the outcome, but maybe I just did the test wrong).

Having successfully configured with swap on, I turned swap off and tried to make machinekit. It failed while compiling emctask.cc: "virtual memory exhausted: Cannot allocate memory." Trying again from scratch, this time keeping swap on, I saw a peak swap usage of 430MB (didn't repeat this time).

mhaberler, 2015-05-29:
doom, doom.. we dont do doom around here ;)

another option would be clang
maybe swapping to a file for the build is easier than redoing partitions

RunningLight, 2015-05-29:
My bad. I forgot about that option.

mhaberler, 2015-05-29:
trying this now: https://wiki.archlinux.org/index.php/Swap#Swap_file_creation

````
 fallocate -l 2048M /swapfile
 chmod 600 /swapfile
 mkswap /swapfile
````

RunningLight, 2015-05-29:
Did almost the identical thing In a x86 VM (just 500MB) and it worked fine.

RunningLight, 2015-05-29:
I'll be curious to hear how much time swapping adds to the build process on a real BBB.

PS - my Marvel-comics loving grandson has been known to call me "Dr. Doom".

mhaberler, 2015-05-30:
works for me with the swapfile.

cdsteinkuehler, 2015-05-30:
On 5/29/2015 10:23 PM, Michael Haberler wrote:
> works for me with the swapfile.

It's possible to create a swapfile on the default BBB images, but it
means less free space on the uSD card.  What size swap file are you
using?  Is 512MB big enough, or should it be larger?

NOTE: When I was using the scripts to write uSD images instead of the
current pre-made image (ie: a 'dd' clone of the block device), I
always created a 512 MB swap file, but I didn't push that into the
pre-built images (which I've regretted on more than one occasion, even
though I don't do builds on the BBB).

-- 
Charles Steinkuehler
charles@steinkuehler.net

mhaberler, 2015-05-30:
I am pretty sure 512MB does it, I will test with smaller swap files and report

build time with swapfile:
real	89m56.414s
user	74m28.631s
sys	5m33.234s