machinekit/machinekit #663: Unsafe type usage in comps/icomps

strahlex, 2015-05-29:
In most comps/icomps float, int and other standard data types are whereas the HAL data types should. Unwanted type conversions in C can lead almost unfindable heisenbugs. Therefore I propose a cleanup session changing all data types to the appropritae HAL data types. The most dangerous Problem I see so far is that float is used in many places but the HAL float type is a double.

mhaberler, 2015-05-29:
seconded. Loose typing is asking for trouble, and there are lots of smelly dead bodies in that cellar,
happy to review and changes

mhaberler, 2015-06-01:
I am about to extend the HAL definition of types, primarily with an eye towards standardizing the use of basic types and type tags (HAL_FLOAT etc) within and above HAL:

````
/** HAL "data union" structure
 ** This structure may hold any type of hal data
*/
typedef union {
    hal_bit_t b;
    hal_s32_t s;
    hal_s32_t u;
    hal_float_t f;
    hal_s64_t s64;
    hal_u64_t u64;
} hal_data_u;

typedef enum {
    HAL_TYPE_UNSPECIFIED = -1,
    HAL_BIT = 1,
    HAL_FLOAT = 2,
    HAL_S32 = 3,
    HAL_U32 = 4,

    // the following type tags are for uniformity and run-time typing
    // there are currently no HAL values of this type (!)
    HAL_S64 = 5,
    HAL_U64 = 6,
    HAL_STRING = 7,
    HAL_BYTES = 8

} hal_type_t;
````

the protobuf Value tag will be a superset thereof:
````
enum ValueType {
    //  the following tags correspond to hal.h: hal_type_t;
    HAL_TYPE_UNSPECIFIED = -1;
    HAL_BIT   = 1;
    HAL_FLOAT = 2;
    HAL_S32   = 3;
    HAL_U32   = 4;
    HAL_S64 = 5;
    HAL_U64 = 6;
    HAL_STRING = 7;
    HAL_BYTES = 8;

    // any tags for non-RT HAL types:
    CARTESIAN = 10;
    LEGACY_CARTESIAN = 11;
    POSE      = 12;
    LEGACY_POSE    = 13;
}
````
as will be the basic Value message:
````
message Value {
    option (nanopb_msgopt).msgid = 1500; // see README.msgid

    required ValueType   type = 15;
    // actual values

    // One of the following must be filled in,
    // depending on type.

    // scalars
    optional bool      halbit   = 1;
    optional double    halfloat = 2;
    optional sfixed32  hals32   = 3;
    optional fixed32   halu32   = 4;
    optional sfixed64  hals64   = 5;
    optional fixed64   halu64   = 6;
    optional string    halstring = 7    [(nanopb).max_size = 41];
    optional bytes     halbytes  = 8    [(nanopb).max_size = 41];

    // compound types
    optional PmCartesian       carte   = 10;
    optional EmcPose           pose    = 12;
}
`````
the 64bit types could actually quite easily be supported in HAL; strings and blobs cannot be made  pins/signals/params due to atomicity requirements, but it does make sense to have the tags and matching defintions in protobuf

zultron, 2015-06-01:
Looks great.

How about naming the `hal_data_u` members consistently with `hal_type_t`, `ValueType` and `Value`, even if slightly more verbose?  `bit`, `float`, `s32` etc.

How does `(nanopb).max_size = 41` get enforced in non-nanopb code?  That hard-coded value that could become inconsistent worries me.

mhaberler, 2015-06-01:
another issue is the indiscriminate use of 'int' throughout HAL - even in places where _size_t_ would be the obvious choice; same for 'unsigned'

mhaberler, 2015-06-01:
(nanopb).max_size = 41 means: if this protobuf message is used in RT through nanopb, the maximum string size is 41 - all this means a protobuf can be decoded with a fixed-size struct in nanopb, i.e. avoiding malloc/free

that sie limit is a consequence of the (currently example only) code in rtprintf.c which was done to get around the incomprehensible dbuf.c/h code eventually. It would be kthreads only and not a priority for me.

but you're right, a specialized message for rtprintf/dbug would get around that case. Nb Value is currently not used.

mhaberler, 2015-06-01:
naming: fixed in value.proto

strahlex, 2015-06-01:
One thing to mention is here is that when initializing C variables one should explicitly specify the data type. e.g.:
```
doubles: 0.0
unsigned: 0u
unsigned long: 0ul
```
and so on. We already had a problem that was a result of shifting a value more than 31 places to the left that was not typed as long value. Also to mention is that certain values can not be represented as floating point numbers. Therefore, floating point point number should never be compared using the equals operator. See also https://github.com/machinekit/machinekit/issues/391

mhaberler, 2015-06-01:
maybe a clang build catches this

RunningLight, 2015-06-01:
@mhaberler I've notice clang comes up a lot in your remarks. Do we have much experience with it? The llvm site says

> The support for X86 (both 32-bit and 64-bit) is considered stable on Darwin (Mac OS X), Linux, FreeBSD, and Dragonfly BSD: it has been tested to correctly compile many large C, C++, Objective-C, and Objective-C++ codebases.

while 

> The support for ARM (specifically ARMv6 and ARMv7) is considered stable on Darwin (iOS): it has been tested to correctly compile many large C, C++, Objective-C, and Objective-C++ codebases. Clang only supports a limited number of ARM architectures. It does not yet fully support ARMv5, for example.

Is the fact that Linux isn't mentioned in the ARM statement significant?

RunningLight, 2015-06-01:
Oops, so you already did some work with clang (https://github.com/mhaberler/asciidoc-sandbox/wiki/Building-machinekit-with-clang-(aka-llvm). By the time I remembered this, I had already started down the path on a jessie-amd64 box, explicitly setting CC and CXX in `./configure`, but didn't get far. The `make` process quickly compiled 82 files, then failed on `src/rtapi/rtapi_compat.c`, reporting

```
clang: error: argument unused during compilation: '-fno-strict-overflow'
```

Strangely, the same `argument unused...` message is reported as only a warning on the previous 82 compiles. I'm not a compiler-options guru. I'm going to back away slowly :blush:

ArcEye, 2015-06-02:
I will go through the icomps, when I can bear to look at them again:D

float is being done away with in favour of double in a lot of code elsewhere too

strahlex, 2015-06-06:
@ArcEye Ideally the HAL data types should be used for all occasions where data is exposed to HAL.

An example comp can be found here: https://github.com/thecooltool/Uni-print-3D/blob/ve4/thermistor_check.comp#L56
I tried to use HAL data types where suitable.

ArcEye, 2015-06-07:
I can see that you have used some hal types
```
variable hal_float_t last_temp = 0.0;
variable hal_bit_t last_enable = 0;
variable hal_float_t oldwait = 0.0;
variable int t_secs = 0;
variable int t_nsecs = 0;
variable int c_secs = 0;
variable int c_nsecs = 0;
```
However int is probably the most contentious variable of the lot.

We probably need some sort of common types usage convention as per the previous posts.

I think the reason so many people simply use int, is that the compiler will tend to use a type sensible for the architecture, so it just works.

strahlex, 2015-06-08:
@ArcEye You are right this was a bad example. The data type should be long as the period variable passed to the thread function is also long. Furthermore, there where some implicit type conversions between hal_float_t and int ongoing.

ArcEye, 2015-06-08:
If you want a really fun example, look at ilowpass
(description and groff formatting removed for clarity)
```
component ilowpass "Low-pass filter with integer inputs and outputs";

pin in s32 in;
pin out s32 out ;
param rw float scale = 1024 ;
param rw float gain = .5 ;
variable double value;

function _ "Update the output based on the input and parameters";

license "GPL";
author "Jeff Epler <jepler@unpythonic.net>";
;;

value += (in - value) * gain;
out = (int)(__s64)(value * scale);

return 0;
```
(hal_float_t * double), cast to __s64, cast to int and assigned to hal_s32_t

Think I am going to leave that one alone:)

strahlex, 2015-06-08:
Interesting method.

Now that I see it here: We should get rid of these params too: https://github.com/machinekit/machinekit/issues/610 Could be done in the same cleanup/review process.

ArcEye, 2015-06-22:
Have a look at
https://github.com/ArcEye/machinekit/commit/1f605ba4beb60a0ad6f6d6f038eeebebda3c4532

params: I have changed 
`param rw` to `pin io`
`param r` to `pin out` (since it is presumably just there for debugging purposes)

variables:
**float** or double become `hal_float_t`
**int** defaults to hal_s32_t, unless using it to hold large number (eg round over a float) or where cannot hold a neg value, then hal_u32_t
**char** occured in 3 places and replaced with hal_bit_t, as that was what it was holding or being compared to

instcomp still uses params but warns where they occur that they should be deprecated

Not done a PR yet until another pair or two of eyes make sure there are no gotchas

Only downside is my editor has changed tabs to spaces, resulting in many more changes than intended.;)

ArcEye, 2015-06-22:
Found a few more and trying to fix the white space issues, commit reset for a bit

ArcEye, 2015-06-22:
Now at
https://github.com/ArcEye/machinekit/commit/bc5b64d58d715f3da7b9248c2687dc61e51b3884
with whitespace changes squashed

strahlex, 2015-06-23:
@ArcEye Can you please create a PR so I can add some comments to the changes.

ArcEye, 2015-06-23:
On 23/06/15 15:45, Alexander Rössler
      wrote:
    
    
      @ArcEye Can you please create a PR so
        I can add some comments to the changes.
      —
          it on GitHub.
      
        
          
          
        
        
      
    
    Done, found one error, variable got clobbered and was missing, but
    seems to build OK

closed 2015-06-26 by strahlex

marksue, 2015-07-05:
There was the wish for request #701 to take this discussion into account, but the question
is, which types should be used? 

In the affected user component (xhc-hb04.cc) the standard data types int, short, char etc. are used for 
local variables, for cast expressions and calculations. As discussed here, the usage of this 
types should be avoided, because their implementation depends on the current compiler and\or 
platform. On the BBB for example 'char' seems to be 'unsigned', whereas on a X86 it is 'signed' 
per default. Worse, whether char is 'unsigned' or 'signed' can be set by a compiler option. 

But replacing the native C types through hal_ types at all isn't (imo) a good idea because:

1) hal_ types (hal_u32_t, hal_s32_t, ... ) are defined as 'volatile' in hal.h     

	typedef volatile hal_bool hal_bit_t;
	typedef volatile __u32 hal_u32_t;
	typedef volatile __s32 hal_s32_t;

2) hal.h does not define 8 and 16 Bit types.  

I wonder which types should be used instead (where hal_ types are not suitable)? stdint.h?

mhaberler, 2015-07-05:
this is a corner of HAL which needs cleaning up badly 
- the use of volatile in the HAL typedefs is most likely wrong and counterproductive to start with (see https://www.kernel.org/doc/Documentation/volatile-considered-harmful.txt)
- the types can and should be extended to cover 1/2/4/8 byte signed and unsigned without relying on a particular userland or kernel include file
- the hal_object_t enum should be extended to cover the new types
- the enum ObjectType in  protobuf/proto/types.proto must match these types
- NB the enum ObjectType is critical with respect to backwards compatibility. Only add new values, do NOT change existing ones.

commit d31205c3 (2015-06-26): Merge pull request #694 from ArcEye/master

Fix unsafe type usages - close #663 & use of params - close #610