machinekit/machinekit #666: Issue in debian package: couldn't load file "/usr/lib/tcltk/linuxcnc/bin/../linuxcnc.so": librtapi_math.so

darenschwenke, 2015-05-30:
I experienced this also.  Builds fine from source.

Copied from https://groups.google.com/forum/?utm_source=digest&utm_medium=email#!topic/machinekit/W1WJVYDQPyM

brad@gypsy:~$ machinekit
MACHINEKIT - 0.1
Error in startup script: couldn't load file "/usr/lib/tcltk/linuxcnc/bin/../linuxcnc.so": librtapi_math.so: cannot open shared object file: No such file or directory
    while executing
"load [file join [file dirname [info script]] linuxcnc[info sharedlibextension]]"
    (file "/usr/lib/tcltk/linuxcnc/bin/../linuxcnc.tcl" line 60)
    invoked from within
"source [file join [file dirname [info script]] .. linuxcnc.tcl]"
    (file "/usr/lib/tcltk/linuxcnc/bin/pickconfig.tcl" line 29)

launching with '-l' for the last config gets a little farther but fails with the same error. At least it writes to the linuxcnc log file which is pretty empty:

May 30 11:32:44 gypsy msgd:0: startup pid=7536 flavor=xenomai rtlevel=1 usrlevel=1 halsize=524288 shm=Posix gcc=4.7.2 version=v~~
May 30 11:32:44 gypsy msgd:0: ØMQ=4.0.4 czmq=2.2.0 protobuf=2.4.1
May 30 11:32:44 gypsy msgd:0: configured: sha=not in a git repo
May 30 11:32:44 gypsy msgd:0: built:      May 30 2015 14:06:46 sha=not in a git repo
May 30 11:32:44 gypsy msgd:0: register_stuff: actual hostname as announced by avahi='gypsy.local'
May 30 11:32:44 gypsy msgd:0: zeroconf: registering: 'Log service on gypsy.local pid 7536'
May 30 11:32:45 gypsy msgd:0: zeroconf: registered 'Log service on gypsy.local pid 7536' _machinekit._tcp 49152 TXT "uuid=a42c8c6b-4025-4f83-ba28-dad21114744a" "instance=7fc0fba6-06e9-11e5-8953-e091f53f2c58" "service=log" "dsn=tcp://gypsy.local:49152"
May 30 11:32:49 gypsy msgd:0: msgd:0: Terminated - shutting down
May 30 11:32:49 gypsy msgd:0: zeroconf: unregistering 'Log service on gypsy.local pid 7536'
May 30 11:32:49 gypsy msgd:0: log buffer hwm: 0% (0 msgs, 0 bytes out of 524288)
May 30 11:32:49 gypsy msgd:0: normal shutdown - global segment detached

zultron, 2015-05-30:
Can you see what version machinekit package you upgraded from?  This seems to work for me:

```
{ cat /var/log/dpkg.log.*[0-9]; gzip -cd /var/log/dpkg.log.*.gz; } | grep 'status installed machinekit:' | sort
```

darenschwenke, 2015-05-30:
Didn't have a /var/log/dpkg.log, but I did have a /var/log/apt/history.log:
Start-Date: 2015-05-28  22:48:13
Commandline: apt-get upgrade
Upgrade: machinekit-xenomai:armhf (0.1.1257-1da.bbot.pr639.gitb63c0b0~1wheezy, 0.1.1303-1da.bbot.pr658.gita802cff~1wheezy), xenomai-runtime:armhf (2.6.4-git20150129-1~bpo70+20150211+1, 2.6.4-git20150311-1~bpo70+20150518+1), nodejs-dev:armhf (0.10.29~dfsg-1~bpo70+20141125+1, 0.10.38~dfsg-1~bpo70+20150526+1), libv8-3.14-dev:armhf (3.14.5.8-8~bpo70+20141125+1, 3.14.5.8-8.1~bpo70+20150526+1), libfuse2:armhf (2.9.0-2+deb7u1, 2.9.0-2+deb7u2), nodejs:armhf (0.10.29~dfsg-1~bpo70+20141125+1, 0.10.38~dfsg-1~bpo70+20150526+1), libv8-3.14.5:armhf (3.14.5.8-8~bpo70+20141125+1, 3.14.5.8-8.1~bpo70+20150526+1), fuse:armhf (2.9.0-2+deb7u1, 2.9.0-2+deb7u2), machinekit:armhf (0.1.1257-1da.bbot.pr639.gitb63c0b0~1wheezy, 0.1.1303-1da.bbot.pr658.gita802cff~1wheezy), nodejs-legacy:armhf (0.10.29~dfsg-1~bpo70+20141125+1, 0.10.38~dfsg-1~bpo70+20150526+1), libxenomai1:armhf (2.6.4-git20150129-1~bpo70+20150211+1, 2.6.4-git20150311-1~bpo70+20150518+1), ntfs-3g:armhf (2012.1.15AR.5-2.1, 2012.1.15AR.5-2.1+deb7u2)
End-Date: 2015-05-28  22:50:09

darenschwenke, 2015-05-30:
That was to.  Here is from:
Start-Date: 2015-05-18  00:26:11
Commandline: apt-get -y --force-yes install machinekit-xenomai dovetail-automata-keyring
Install: libjansson4:armhf (2.7-1~bpo70+20141214+1, automatic), python-pyftpdlib:armhf (1.2.0-3~1da~wheezy1, automatic), python-opengl:armhf (3.0.1-1, automatic), python-gst0.10:armhf (0.10.22-3, automatic), libgnomevfs2-0:armhf (2.24.4-2, automatic), liburiparser1:armhf (0.7.5-1, automatic), libboost-python1.49.0:armhf (1.49.0-3.2, automatic), machinekit-xenomai:armhf (0.1.1257-1da.bbot.pr639.gitb63c0b0~1wheezy), python-gnome2:armhf (2.28.1+dfsg-1, automatic), liblcms1:armhf (1.19.dfsg-1.2, automatic), libgnome2-0:armhf (2.32.1-3, automatic), python-protobuf:armhf (2.4.1-3, automatic), libgnomeui-common:armhf (2.24.5-2, automatic), libart-2.0-2:armhf (2.3.21-2, automatic), xenomai-runtime:armhf (2.6.4-git20150129-1~bpo70+20150211+1, automatic), python-libxml2:armhf (2.8.0+dfsg1-7+wheezy4, automatic), python-gtksourceview2:armhf (2.10.1-2, automatic), protobuf-compiler:armhf (2.4.1-3, automatic), python-gtkglext1:armhf (1.1.0-9.1, automatic), libbonoboui2-common:armhf (2.24.3-1, automatic), bc:armhf (1.06.95-2, automatic), libbonoboui2-0:armhf (2.24.3-1, automatic), python-imaging-tk:armhf (1.1.7-4+deb7u1, automatic), python-pyorbit:armhf (2.24.0-6, automatic), python-gconf:armhf (2.28.1+dfsg-1, automatic), python-simplejson:armhf (2.5.2-1, automatic), dovetail-automata-keyring:armhf (0.1-1~wheezy1), libfam0:armhf (2.7.0-17), libboost-thread1.49.0:armhf (1.49.0-3.2, automatic), libboost-serialization1.49.0:armhf (1.49.0-3.2, automatic), blt:armhf (2.4z-4.2, automatic), libgnome2-common:armhf (2.32.1-3, automatic), python-avahi:armhf (0.6.31-2, automatic), python-tk:armhf (2.7.3-1, automatic), libgnomeprint2.2-data:armhf (2.18.8-3), libgtksourceview2.0-common:armhf (2.10.4-1, automatic), python-imaging:armhf (1.1.7-4+deb7u1, automatic), libgtksourceview2.0-0:armhf (2.10.4-1, automatic), libprotobuf7:armhf (2.4.1-3, automatic), python-netifaces:armhf (0.8-1, automatic), libgnomecanvas2-0:armhf (2.30.3-1.2, automatic), python-vte:armhf (0.28.2-5, automatic), libgnomeui-0:armhf (2.24.5-2, automatic), python-zmq:armhf (14.3.0-2~1mk~wheezy1, automatic), libidl0:armhf (0.8.14-0.2, automatic), libbonobo2-0:armhf (2.24.3-1, automatic), libgnomecanvas2-common:armhf (2.30.3-1.2, automatic), libgnomevfs2-extra:armhf (2.24.4-2), tcl8.5:armhf (8.5.11-2, automatic), tcl-tclreadline:armhf (2.1.0-12, automatic), libzmq4:armhf (4.0.4-2~1da~wheezy1, automatic), python-gdbm:armhf (2.7.3-1, automatic), gnome-mime-data:armhf (2.18.0-1), libczmq2:armhf (2.2.0-0.2~1da~wheezy1, automatic), liborbit2:armhf (2.14.19-0.1, automatic), tk8.5:armhf (8.5.11-2, automatic), bwidget:armhf (1.9.5-1, automatic), libbonobo2-common:armhf (2.24.3-1, automatic), libmodbus5:armhf (3.0.3-1, automatic), machinekit:armhf (0.1.1257-1da.bbot.pr639.gitb63c0b0~1wheezy, automatic), python-xlib:armhf (0.14+20091101-1, automatic), libgnomeprint2.2-0:armhf (2.18.8-3, automatic), libgtkglext1:armhf (1.2.0-2, automatic), libprotoc7:armhf (2.4.1-3, automatic), libxenomai1:armhf (2.6.4-git20150129-1~bpo70+20150211+1, automatic), freeglut3:armhf (2.6.0-4, automatic), libgnomeprintui2.2-0:armhf (2.18.6-3, automatic), libgnomeprintui2.2-common:armhf (2.18.6-3), libgnomecups1.0-1:armhf (0.2.3-5), libtk-img:armhf (1.3-release-12, automatic), libgnomevfs2-common:armhf (2.24.4-2), libsodium:armhf (0.5.0-0.1mk~wheezy1, automatic)
End-Date: 2015-05-18  00:27:57

zultron, 2015-05-31:
@strahlex, can you check this out?  I don't see `librtapi_math.so` being installed in the `machinekit` package.

zultron, 2015-05-31:
Thanks for answering my irrelevant questions.

Hopefully PR #667 will take care of this.

closed 2015-05-31 by mhaberler

strahlex, 2015-05-31:
@zultron The packages worked last week so librtapi_math.so was definitely installed.

PR #667 (2015-05-31): `rtapi_math` cleanups
Clean ups from #652
- Fix creation of `rtapi_math.so.0` DSO and `....so` symlink
  - Fixes #666
- Remove dupe `$(INCLUDES)` entry
- Remove dupe header install rules
- Remove apparently unused file `rtai_math.h`
commit 136de2e1 (2015-05-31): `rtapi_math` cleanups

Clean ups from #652
- Fix creation of `rtapi_math.so.0` DSO and `....so` symlink
  - Fixes #666
- Remove dupe `$(INCLUDES)` entry
- Remove dupe header install rules
- Remove apparently unused file `rtai_math.h`