machinekit/machinekit #670: Shared library linking needs attention

zultron, 2015-06-01:
Issues #666, #667 and #669 point out the inconsistent treatment of linking against shared libs.

@mhaberler pointed to Wheeler's [Program Library HOWTO][1], from which I'll use the terminology:
- 'soname':  e.g. `liblinuxcncini.so.0`; symlink to 'real name'
  - created by `ldconfig`, according to Wheeler; apparently included in run-time packages
- 'real name': e.g. `liblinuxcncini.so.0.0`; the actual dynamic shared object file name
  - needed at run-time; in `machinekit` package
- 'linker name': e.g. `liblinuxcncini.so`; symlink to either 'soname' or 'real name'
  - needed for build; in `machinekit-dev` package

[1]: http://tldp.org/HOWTO/Program-Library-HOWTO/shared-libraries.html

### Library file names, paths, packages

A number of problems are evident from file names in packages:
```
$ { dpkg-query -L machinekit | sed 's/^/run /'; \
    dpkg-query -L machinekit-dev | sed 's/^/dev /'; } | \
    sort -k2 | column -t | grep '/usr/lib/[^/]*\.so\(\|\.0\)'
```

**Cython extensions:**
```
dev  /usr/lib/compat.so
dev  /usr/lib/hal.so
dev  /usr/lib/rtapi.so
dev  /usr/lib/shmcommon.so
```
- These binaries installed in `-dev` package should be removed
- Already installed in correct place in base package
  - e.g. `/usr/lib/python2.7/dist-packages/machinekit/hal.x86_64-linux-gnu.so`

**libpyplugin.so.0:**
```
run  /usr/lib/libpyplugin.so.0
```
- `/usr/lib/libpyplugin.so.0` needs a symlink with the linker name for the `-dev` package

**Other libraries:**
```
dev  /usr/lib/libcanterp.so  
run  /usr/lib/libcanterp.so.0
dev  /usr/lib/liblinuxcnc-npb.so
run  /usr/lib/liblinuxcnc-npb.so.0
dev  /usr/lib/liblinuxcnc-pb2++.so  
run  /usr/lib/liblinuxcnc-pb2++.so.0
dev  /usr/lib/liblinuxcnchal.so  
run  /usr/lib/liblinuxcnchal.so.0
dev  /usr/lib/liblinuxcncini.so  
run  /usr/lib/liblinuxcncini.so.0
dev  /usr/lib/liblinuxcncshm.so  
run  /usr/lib/liblinuxcncshm.so.0
dev  /usr/lib/liblinuxcnculapi.so
run  /usr/lib/liblinuxcnculapi.so.0
dev  /usr/lib/libmtalk.so
run  /usr/lib/libmtalk.so.0
dev  /usr/lib/libnml.so  
run  /usr/lib/libnml.so.0   
dev  /usr/lib/libposemath.so
run  /usr/lib/libposemath.so.0
dev  /usr/lib/librs274.so
run  /usr/lib/librs274.so.0
dev  /usr/lib/librtapi_math.so
run  /usr/lib/librtapi_math.so.0
run  /usr/lib/libpyplugin.so.0
```
- According to Wheeler, the 'real name' file names should have additional `.0` appended.  The soname symlink must be included in packages; I'm unclear how this happens.
- These libraries should be installed into the multiarch lib directory, e.g. `/usr/lib/x86_64-linux-gnu/`.

## Bizarre case, `/usr/lib/liblinuxcnc.a`:
- Built in `src/emc/nml_intf/Submakefile`
- Statically-linked to:
  - `../bin/*` executables built in `src/emc/` directory and `mb2hal`
   - `../lib/python/linuxcnc.so` (`src/emc/usr_intf/axis/Submakefile`)
  - `../tcl/linuxcnc.so` (`src/emc/usr_intf/Submakefile`)
- Can this be turned into a DSO?

## Shared lib linker options
Wheeler says a shared lib with soname `liblinuxcncshm.so.0` should be linked to using `-llinuxcncshm`.  The MK build system instead uses `../lib/liblinuxcncshm.so.0`; e.g. [2].  Does Wheeler's recommendation apply here, since `liblinuxcncshm.so.0` is not yet a system library?

[2]: https://github.com/machinekit/machinekit/blob/master/src/rtapi/Submakefile#L164-L175

Assuming the current method is correct:

**Linking against `[...].so` vs. `[...].so.0`:**
@strahlex points out in #669 that binaries should be linked against the soname file path (`[...].so.0`), not the linker file path (`[...].so`), and fixes this for `rtapi_math.so.0`.  However, other libraries are not treated consistently.

For example, [`genserkins`][3]:  Links against mixture of `[...].so` and `[...].so.0`.  The linker file won't be present when the `machinekit-dev` package is not installed.  Similar examples abound.

[3]: https://github.com/machinekit/machinekit/blob/6c09f5252de39e335feff36862ed7493357e606e/src/emc/kinematics/Submakefile#L17-L22

### `automake`
This mess is a result of this silly build system.  If it were swapped out for e.g. an `automake`-based system, all of this would be templated for us and would be much harder to get wrong.

mhaberler, 2015-06-11:
this is a good reference on shared libraries: http://www.akkadia.org/drepper/dsohowto.pdf

zultron, 2015-07-26:
Closing (wontfix):  no problem reports, and lack of interest

closed 2015-07-26 by zultron