machinekit/machinekit #671: Are identical 6 instances ReadTemp.py + the old version of hal_temp_bbb still needed ?

the-snowwhite, 2015-06-01:
I'n my first dive into Machinekit with my CRAMPS board there were some issues getting the temp readings to work correctly.
Finding the need for a separate ReadTemp.py confusing
I found out there was a user component:
machinekit/src/hal/user_comps/hal_temp_bbb.py
which provided the necessary functionality, however only for the BeBoPr board at that time, so
A PR. updated it to include support for the CRAMPS board also with a -b CRAMPS option.
So it now should support Both cramps and bebopr(++) based capes + war easily updatable to support other thermistor input electronics.
______________
This file (should) provide(s) all the functionality in the manually included ReadTemp files
So I see no reason not to use this component instead universally
_________________________________
I'm wondering why all other BBB board config examples are still running with a custom readtemp
including the replicape which is using what looks looks the unupdated old version of hal_temp_bbb 
???
______
looking forward towards modulised configs:
Having all BBB cape boards using 1 temperature read component (providing the correct options) instead of the 6 instances of ReadTemp.py + (the old haltemp file) that include thermistor tables already provided in the:
machinekit/share/fdm/thermistor_tables 
folder
would seem much easier to deal with in future.
________
However I have no BeBoPr(++) or replicape to verify that the hal_temp_bbb.py user component works correctly on these boards to make a specific commit/PR ....

strahlex, 2015-06-01:
You are right. The hal_bbb_temp component should replace the `ReadTemp.py`. Since I have created the component I can verify that it works with the BeBoPr(++).

darenschwenke, 2015-06-01:
Here is the working CRAMPS line I use.  hal_temp_bbb does use the share thermisistor tables now:
loadusr -Wn Therm hal_temp_bbb -n Therm -c 04:3950-100k,05:3950-100k -b CRAMPS

Also, you may find the following useful:
https://github.com/strahlex/Machinekit-FDM-Standard

strahlex, 2015-06-01:
I should note that I am working on modular Python based HAL configs for the 3D printing world right now. This means that we might be able to use a common config base for all 3D printers.

the-snowwhite, 2015-06-01:
I thought I saw something like that looking for clues to get my config  based upon you uni-print-3d setup to work again.... 
Will the final selection and assembly of the modules be based upon commandline arguments / textfile editing or would it be beneficial to try building  a qt-creator json based config wizard that can select different config files and maybe add the necessary glue code ?

the-snowwhite, 2015-06-02:
Well the target configs are:
machinekit/configs/ARM/BeagleBone/BeBoPr

    36: loadusr -Wn Therm ./ReadTemp.py -n Therm --num_chan 2 -t 1 1 -a 1 3

machinekit/configs/ARM/BeagleBone/BeBoPr++

    ???

machinekit/configs/ARM/BeagleBone/BeBoPr++/Pepper

    36: loadusr -Wn Therm ./ReadTemp.py -n Therm --num_chan 2 -t 1 1 -a 4 5

machinekit/configs/ARM/BeagleBone/BeBoPr++/Pololu

    36: loadusr -Wn Therm ./ReadTemp.py -n Therm --num_chan 2 -t 1 1 -a 4 5

machinekit/configs/ARM/BeagleBone/BeBoPr-Bridge

    36: loadusr -Wn Therm ./ReadTemp.py -n Therm --num_chan 2 -t 1 1 -a 4 5

machinekit/configs/ARM/BeagleBone/PocketNC

    46: loadusr -Wn Therm /home/machinekit/machinekit-dev/configs/ARM/BeagleBone/BeBoPr-Bridge/ReadTemp.py -n Therm --num_chan 2 -t 1 1 -a 4 5

machinekit/configs/ARM/BeagleBone/Replicape
loadusr -Wn Therm ./hal_temp_bbb -n Therm -c 04:epcos_B57560G1104,06:epcos_B57560G1104 -b Replicape

    35: loadusr -Wn Therm ./hal_temp_bbb -n Therm -c 04:epcos_B57560G1104,06:epcos_B57560G1104 -b Replicape

!!! The hal_temp_bbb file seems actually to be an update to hal_temp_bbb.ly that includes an addition of the  Replicape to the -b (board) option:
So this update should also be PR'ed into: .......
However there are also some other changes/updates to the file compared to:
machinekit/src/hal/user_comps/hal_temp_bbb.py
The file is:

So it would be nice with some feedback as to if these changes now safely can go into hal_temp_bbb.py

the loadline I find most usefull is:

    loadusr -Wn Therm hal_temp_bbb -n Therm -c 04:[HBP]THERMISTOR,05:[EXTRUDER_0]THERMISTOR -b [CAPE]BOARD

This way the thermistors and the Cape can be specified externally in the ini file instead of being tied to the hal file itself, making it more portable.

Last there is to figure out the thermistor input pin values:
On the CRAMPS board the input pins are 1:04 , 2:05, 3:06, 4:03
How does the  -a 4 5 option in ReadTemp.py compare to the input pin options in hap_temp_bbb.py
same ?

cdsteinkuehler, 2015-06-02:
On 6/2/2015 9:25 AM, Michael Brown wrote:
> 
> the loadline I find most usefull is:
> 
>     loadusr -Wn Therm hal_temp_bbb -n Therm -c 04:[HBP]THERMISTOR,05:[EXTRUDER_0]THERMISTOR -b [CAPE]BOARD
> 
> This way the thermistors and the Cape can be specified externally in the ini file instead of being tied to the hal file itself, making it more portable.

I like this as well, but the channel values (04: and 05:, above) might
change if the board changes, and a *LOT* of pin definitions in the HAL
file will also need to change, so I don't see much use in having the
board in the INI file.

Proper support for migrating configurations between boards will likely
have to wait for a python based HAL file for the BBB, since it's pretty
hard to do relational database style lookups with the INI and shell
variable expansions currently supported by halcmd.  So you can't
(easily) do something like:

setp x.step.pin = [CAPE]BOARD:(X_step_pin)

...but that would be easy-peasy in Python by importing a Board file,
assuming we use consistent names for the channels across the various capes.

> Last there is to figure out the thermistor input pin values:
> On the CRAMPS board the input pins are 1:04 , 2:05, 3:06, 4:03
> How does the  -a 4 5 option in ReadTemp.py compare to the input pin options in hap_temp_bbb.py
> same ?

./ReadTemp.py -n Therm --num_chan 2 -t 1 1 -a 4 5

is the same as

./hal_temp_bbb -n Therm -c 04:epcos_B57560G1104,05:epcos_B57560G1104

-- 
Charles Steinkuehler
charles@steinkuehler.net

the-snowwhite, 2016-04-30:
https://groups.google.com/forum/#!topic/machinekit/6_eZbtENExw

I followed instructions from this guide to install machinekit on the CRAMPS board.

 https://github.com/the-snowwhite/CRAMPS2/wiki/BBB-CRAMPS-Machineface-3D-Printer-Out-Of-The-Box-Experience

Looks like another instance of this problem
https://groups.google.com/forum/#!topic/machinekit/F8YK6EjLxRI

The type should be hal_bool, or bool needs to be defined seperately by #include stdbool.h

I don't think thermistor_check.comp is in the standard repo,
it appears to live in Michaels Browns repo amongst others
https://github.com/the-snowwhite/CRAMPS2/blob/master/configs/ARM/Beaglebone/CRAMPS2-vel/thermistor_check.comp

There is also a peculiar error at the top of the screen, but don't know how that arose

@cdsteinkuehler 
> Yes, this is a problem with the thermistor_check component which is 
> not a built-in part of Machinekit.  Change the three bool types to 
> hal_bool and it should work. 

You may have a similar problem with the reset component.

ArcEye, 2018-08-01:
Retired on historical grounds, does not seem to be resulting in action and the main component is not in this repo.

closed 2018-08-01 by ArcEye